raspberrypi/libcamera #19: Temperature Measurement Support for the IMX477

meawoppl, 2022-05-18:
I am interested in adding this support. I have a few hints about the process, but what is the obvious starting point?

kbingham, 2022-05-19:
- [ ] A V4L2 control needs to be added to the linux kernel. This is a likely starting point : https://lore.kernel.org/linux-media/20220415111845.27130-3-benjamin.mugnier@foss.st.com/ will add V4L2_CID_TEMPERATURE.
- [ ] Sony needs to tell us the register to read the IMX477 temperature. (or someone with a datasheet needs to be allowed to do this, or tell us).
- [ ] This can then be implemented as a volatile control in which reads the register in a new .g_volatile_ctrl implementation extended from https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/media/i2c/imx477.c#L1415
- [ ] A control needs to be added to the libcamera control framework so that the camera can expose reading the temperature, or perhaps the libcamera camera sensor class should always read the temperature to provide it in the request completion metadata.

6by9, 2022-05-19:
Adding V4L2_CID_TEMPERATURE vs using the hwmon API is debated in that thread, so not a resolved question.

Reg 0x0138 bit 0 to enable(1)/disable(0).
Reg 0x013A as a signed 8 bit value for the temperature. Range -20 to +80.
Regs 0x0136/0x0137 need to be written with the correct clock setup before reading.
21ms is required after enabling via 0x0138 before reading if not streaming, or 8 frames if it is streaming.

It relies on correct calibration data to be set in the OTP. If not written then 0x013a will report a fixed value. I don't know if the modules we produce have appropriate calibration or not. The value I'm getting back of 0x1b-0x1d look vaguely plausible.

pinchartl, 2022-05-19:
@meawoppl What do you need the temperature for ? Is this something you would expect would be exposed to applications, or used internally by the algorithms to adapt image processing parameters ?

closed 2022-06-23 by naushir