raspberrypi/libcamera #22: Issue with gstreamer plugin and IMX519 higher resolutions

chemicstry, 2022-08-16:
I have an Arducam IMX519 camera on a Raspberry Pi 4, which seems to work fine with libcamera apps (libcamera-still), but fails on high resolutions with gstreamer plugin (`libcamerasrc`).

The following configuration works fine:
```bash
# I can see output in window
gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=3840,height=2400,framerate=5/1' ! glimagesink
# Can also record to a file
gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=3840,height=2400,framerate=5/1' ! jpegenc ! avimux ! filesink location=mjpeg.avi
```

However, when I increase resolution to a maximum of 4565x3496, this is what I get in glimagesink:

[image]

I can see pixels change when I move the camera, so it seems to be a buffer size/alignment/format problem.

What I found interesting is that all predefined resolutions work correctly, except for the `[min, max, step]` format, under which 4565x3496 falls. Here is output from `gst-device-monitor-1.0 Video` command:

```
pi@raspberrypi:~ $ gst-device-monitor-1.0 Video
Probing devices...

[125:22:23.295294941] [41496]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3701-f30ad033-dirty (2022-08-16T13:56:54+01:00)
[125:22:23.315788975] [41498]  WARN CameraSensorProperties camera_sensor_properties.cpp:174 No static properties available for 'imx519'
[125:22:23.315863214] [41498]  WARN CameraSensorProperties camera_sensor_properties.cpp:176 Please consider updating the camera sensor properties database
[125:22:23.438486087] [41498]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx519@1a to Unicam device /dev/media3 and ISP device /dev/media0

Device found:

        name  : /base/soc/i2c0mux/i2c@1/imx519@1a
        class : Source/Video
        caps  : video/x-raw, format=NV21, width=160, height=120
                video/x-raw, format=NV21, width=240, height=160
                video/x-raw, format=NV21, width=320, height=240
                video/x-raw, format=NV21, width=400, height=240
                video/x-raw, format=NV21, width=480, height=320
                video/x-raw, format=NV21, width=640, height=360
                video/x-raw, format=NV21, width=640, height=480
                video/x-raw, format=NV21, width=720, height=480
                video/x-raw, format=NV21, width=768, height=480
                video/x-raw, format=NV21, width=854, height=480
                video/x-raw, format=NV21, width=720, height=576
                video/x-raw, format=NV21, width=800, height=600
                video/x-raw, format=NV21, width=960, height=540
                video/x-raw, format=NV21, width=1024, height=576
                video/x-raw, format=NV21, width=960, height=640
                video/x-raw, format=NV21, width=1024, height=600
                video/x-raw, format=NV21, width=1024, height=768
                video/x-raw, format=NV21, width=1280, height=720
                video/x-raw, format=NV21, width=1152, height=864
                video/x-raw, format=NV21, width=1280, height=800
                video/x-raw, format=NV21, width=1360, height=768
                video/x-raw, format=NV21, width=1366, height=768
                video/x-raw, format=NV21, width=1440, height=900
                video/x-raw, format=NV21, width=1280, height=1024
                video/x-raw, format=NV21, width=1536, height=864
                video/x-raw, format=NV21, width=1280, height=1080
                video/x-raw, format=NV21, width=1600, height=900
                video/x-raw, format=NV21, width=1400, height=1050
                video/x-raw, format=NV21, width=1680, height=1050
                video/x-raw, format=NV21, width=1600, height=1200
                video/x-raw, format=NV21, width=1920, height=1080
                video/x-raw, format=NV21, width=2048, height=1080
                video/x-raw, format=NV21, width=1920, height=1200
                video/x-raw, format=NV21, width=2160, height=1080
                video/x-raw, format=NV21, width=2048, height=1152
                video/x-raw, format=NV21, width=2560, height=1080
                video/x-raw, format=NV21, width=2048, height=1536
                video/x-raw, format=NV21, width=2560, height=1440
                video/x-raw, format=NV21, width=2560, height=1600
                video/x-raw, format=NV21, width=3840, height=1080
                video/x-raw, format=NV21, width=2960, height=1440
                video/x-raw, format=NV21, width=3440, height=1440
                video/x-raw, format=NV21, width=2560, height=2048
                video/x-raw, format=NV21, width=3200, height=1800
                video/x-raw, format=NV21, width=3840, height=1600
                video/x-raw, format=NV21, width=3200, height=2048
                video/x-raw, format=NV21, width=3200, height=2400
                video/x-raw, format=NV21, width=3840, height=2160
                video/x-raw, format=NV21, width=4096, height=2160
                video/x-raw, format=NV21, width=3840, height=2400
                video/x-raw, format=NV21, width=[ 64, 4656, 2 ], height=[ 64, 3496, 2 ]
                video/x-raw, format=I420, width=160, height=120
                video/x-raw, format=I420, width=240, height=160
                video/x-raw, format=I420, width=320, height=240
                video/x-raw, format=I420, width=400, height=240
                video/x-raw, format=I420, width=480, height=320
                video/x-raw, format=I420, width=640, height=360
                video/x-raw, format=I420, width=640, height=480
                video/x-raw, format=I420, width=720, height=480
                video/x-raw, format=I420, width=768, height=480
                ***TRUNCATED***
```

I tried libcamera from Arducam binary distribution and compiling from git source, but the issue persists on both.

I'm not too familiar with libcamera stack, but I could try help debugging this if given some hints where to look.

kbingham, 2022-08-16:
Hi @chemicstry 

You referenced 4565x3496 a couple of times, while the gst-device monitor output shows a max of 4656x3496. Can you confirm if the correctly specified values work? (I expect it's the same, and the 4565 is just a typo, as the odd number should be corrected by the pipeline negotiation).

I think I recall that the RPi ISP can only do downscaling, not upscaling, so I wonder what the configuration was when you launched.
Could you post /attach a log of the output from:

`LIBCAMERA_LOG_LEVELS=RP*:0 gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=3840,height=2400,framerate=5/1' ! glimagesink`

(The RP* wildcard should match all RPi based logging internally, if this doesn't show the pipeline configurations, do LIBCAMERA_LOG_LEVELS=*:0)

chemicstry, 2022-08-16:
Sorry, the `4565x3496` was indeed a typo when writting the issue, I've updated the original comment.

Log of `LIBCAMERA_LOG_LEVELS=RP*:0 gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=3840,height=2400,framerate=5/1' ! glimagesink`:
<details>
  <summary>Collapsed</summary>

  ```js
Setting pipeline to PAUSED ...
[131:39:39.216858633] [43706] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:293 [0mlibcamera v0.0.0+3701-f30ad033-dirty (2022-08-16T13:56:54+01:00)
[131:39:39.237161749] [43723] [1;33m WARN [1;37mCameraSensorProperties [1;34mcamera_sensor_properties.cpp:174 [0mNo static properties available for 'imx519'
[131:39:39.237267359] [43723] [1;33m WARN [1;37mCameraSensorProperties [1;34mcamera_sensor_properties.cpp:176 [0mPlease consider updating the camera sensor properties database
[131:39:39.354405624] [43723] [1;36mDEBUG [1;37mRPiBlackLevel [1;34mblack_level.cpp:41 [0m Read black levels red 4096 green 4096 blue 4096
[131:39:39.356962810] [43723] [1;36mDEBUG [1;37mRPiAgc [1;34magc.cpp:191 [0mAgc
[131:39:39.357079845] [43723] [1;36mDEBUG [1;37mRPiAgc [1;34magc.cpp:144 [0mAgcConfig
[131:39:39.360565277] [43723] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cr calibration for ct 3000
[131:39:39.361366543] [43723] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cr calibration for ct 6000
[131:39:39.362079033] [43723] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cb calibration for ct 3000
[131:39:39.362796559] [43723] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cb calibration for ct 6000
[131:39:39.363697269] [43723] [1;36mDEBUG [1;37mRPiSharpen [1;34msharpen.cpp:45 [0mRead threshold 1 strength 1 limit 1
[131:39:39.366679152] [43723] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:1368 [0mRegistered camera /base/soc/i2c0mux/i2c@1/imx519@1a to Unicam device /dev/media3 and ISP device /dev/media0
[131:39:39.366855149] [43723] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:1154 [0mUnable to acquire a Unicam instance
Pipeline is live and does not need PREROLL ...
Got context from element 'sink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
[131:39:39.368445571] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 13471.2 (best 13471.2)
[131:39:39.368583106] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 11471.2 (best 11471.2)
[131:39:39.368624013] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 9152 (best 9152)
[131:39:39.368662383] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 5471.24 (best 5471.24)
[131:39:39.368698160] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 1000 (best 1000)
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
[131:39:39.368839195] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
[131:39:39.369012118] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
[131:39:39.369053081] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
[131:39:39.369088877] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
[131:39:39.369123617] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
[131:39:39.369412391] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:530 [0mTry color space JPEG
New clock: GstSystemClock
[131:39:39.374020436] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
[131:39:39.374168119] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
[131:39:39.374207637] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
[131:39:39.374244673] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
[131:39:39.374312635] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
[131:39:39.374683056] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:530 [0mTry color space JPEG
[131:39:39.376626509] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
[131:39:39.376767970] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
[131:39:39.376805988] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
[131:39:39.376841969] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
[131:39:39.376876469] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
[131:39:39.377189131] [43726] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:530 [0mTry color space JPEG
[... cut ...]
../ext/gl/gstglimagesink.c(1891): gst_glimage_sink_show_frame (): /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink
Execution ended after 0:00:08.950673303
Setting pipeline to NULL ...
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Internal data stream error.
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(311): processRequest (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
streaming stopped, reason error (-5)
Freeing pipeline ...
  ```
</details>

chemicstry, 2022-08-16:
Also a log of `LIBCAMERA_LOG_LEVELS=RP*:0 gst-launch-1.0 libcamerasrc ! 'video/x-raw,width=4656,height=3496,framerate=5/1' ! glimagesink`:
<details>
  <summary>Collapsed</summary>

  ```
Setting pipeline to PAUSED ...
[131:45:30.109280642] [43748] [1;32m INFO [1;37mCamera [1;34mcamera_manager.cpp:293 [0mlibcamera v0.0.0+3701-f30ad033-dirty (2022-08-16T13:56:54+01:00)
[131:45:30.129900473] [43764] [1;33m WARN [1;37mCameraSensorProperties [1;34mcamera_sensor_properties.cpp:174 [0mNo static properties available for 'imx519'
[131:45:30.130127858] [43764] [1;33m WARN [1;37mCameraSensorProperties [1;34mcamera_sensor_properties.cpp:176 [0mPlease consider updating the camera sensor properties database
[131:45:30.244298352] [43764] [1;36mDEBUG [1;37mRPiBlackLevel [1;34mblack_level.cpp:41 [0m Read black levels red 4096 green 4096 blue 4096
[131:45:30.246920425] [43764] [1;36mDEBUG [1;37mRPiAgc [1;34magc.cpp:191 [0mAgc
[131:45:30.247150829] [43764] [1;36mDEBUG [1;37mRPiAgc [1;34magc.cpp:144 [0mAgcConfig
[131:45:30.250768129] [43764] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cr calibration for ct 3000
[131:45:30.251730003] [43764] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cr calibration for ct 6000
[131:45:30.252597602] [43764] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cb calibration for ct 3000
[131:45:30.253417349] [43764] [1;36mDEBUG [1;37mRPiAlsc [1;34malsc.cpp:121 [0mRead calibrations_Cb calibration for ct 6000
[131:45:30.254386798] [43764] [1;36mDEBUG [1;37mRPiSharpen [1;34msharpen.cpp:45 [0mRead threshold 1 strength 1 limit 1
[131:45:30.257470087] [43764] [1;32m INFO [1;37mRPI [1;34mraspberrypi.cpp:1368 [0mRegistered camera /base/soc/i2c0mux/i2c@1/imx519@1a to Unicam device /dev/media3 and ISP device /dev/media0
[131:45:30.257783082] [43764] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:1154 [0mUnable to acquire a Unicam instance
Pipeline is live and does not need PREROLL ...
Got context from element 'sink': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayX11\)\ gldisplayx11-0";
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[131:45:30.259522742] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 13471.2 (best 13471.2)
[131:45:30.259661092] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 11471.2 (best 11471.2)
[131:45:30.259743610] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 9152 (best 9152)
[131:45:30.259811090] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 5471.24 (best 5471.24)
[131:45:30.259875941] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 1000 (best 1000)
[131:45:30.260133048] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
[131:45:30.260203344] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
[131:45:30.260266528] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
[131:45:30.260328842] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
[131:45:30.260389322] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
[131:45:30.260733521] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:530 [0mTry color space JPEG
[131:45:30.265304936] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
[131:45:30.265453268] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
[131:45:30.265492823] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
[131:45:30.265530748] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
[131:45:30.265649691] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
[131:45:30.266071333] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:530 [0mTry color space JPEG
[131:45:30.268078859] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
[131:45:30.268226653] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
[131:45:30.268267042] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
[131:45:30.268343633] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
[131:45:30.268387318] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:168 [0mFormat: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
[131:45:30.268765571] [43767] [1;36mDEBUG [1;37mRPI [1;34mraspberrypi.cpp:530 [0mTry color space JPEG
[... cut ...]
../ext/gl/gstglimagesink.c(1891): gst_glimage_sink_show_frame (): /GstPipeline:pipeline0/GstGLImageSinkBin:glimagesinkbin0/GstGLImageSink:sink
Execution ended after 0:00:10.967967205
Setting pipeline to NULL ...
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Internal data stream error.
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(311): processRequest (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
streaming stopped, reason error (-5)
Freeing pipeline ...
  ```
</details>

kbingham, 2022-08-16:
Ok - extracting the useful bits, from the 3840x2400 (working)

```
│ [131:39:39.377286018] [43726]  INFO Camera camera.cpp:1029 configuring streams: (0) 3840x2400-NV21 (1) 4656x3496-SRGGB10_CSI2P
│ [131:39:39.377840659] [43723] DEBUG RPI raspberrypi.cpp:168 Format: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
│ [131:39:39.377921417] [43723] DEBUG RPI raspberrypi.cpp:168 Format: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
│ [131:39:39.377959564] [43723] DEBUG RPI raspberrypi.cpp:168 Format: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
│ [131:39:39.377996175] [43723] DEBUG RPI raspberrypi.cpp:168 Format: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
│ [131:39:39.378033285] [43723] DEBUG RPI raspberrypi.cpp:168 Format: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
│ [131:39:39.378311467] [43723]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected sensor format: 4656x3496-SRGGB10_1X10 - Selected unicam format: 4656x3496-pRAA
│ [131:39:39.378625592] [43723] DEBUG RPI raspberrypi.cpp:791 Setting ISP Output0 to 3840x2400-NV21
│ [131:39:39.378815793] [43723] DEBUG RPI raspberrypi.cpp:805 Stream ISP Output0 has color space JPEG
│ [131:39:39.378861311] [43723] DEBUG RPI raspberrypi.cpp:867 Setting ISP Output1 (internal) to 1200x750-YU12
```

And on garbled:  

```
  [131:45:30.268895495] [43767]  INFO Camera camera.cpp:1029 configuring streams: (0) 4656x3496-NV21 (1) 4656x3496-SRGGB10_CSI2P
  [131:45:30.269483320] [43764] DEBUG RPI raspberrypi.cpp:168 Format: 1280x720 fmt SRGGB10 Score: 12471.2 (best 12471.2)
  [131:45:30.269654040] [43764] DEBUG RPI raspberrypi.cpp:168 Format: 1920x1080 fmt SRGGB10 Score: 10471.2 (best 10471.2)
  [131:45:30.269745594] [43764] DEBUG RPI raspberrypi.cpp:168 Format: 2328x1748 fmt SRGGB10 Score: 8152 (best 8152)
  [131:45:30.269832963] [43764] DEBUG RPI raspberrypi.cpp:168 Format: 3840x2160 fmt SRGGB10 Score: 4471.24 (best 4471.24)
  [131:45:30.269917499] [43764] DEBUG RPI raspberrypi.cpp:168 Format: 4656x3496 fmt SRGGB10 Score: 0 (best 0)
  [131:45:30.270251346] [43764]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected sensor format: 4656x3496-SRGGB10_1X10 - Selected unicam format: 4656x3496-pRAA
  [131:45:30.270617359] [43764] DEBUG RPI raspberrypi.cpp:791 Setting ISP Output0 to 4656x3496-NV21
  [131:45:30.270858671] [43764] DEBUG RPI raspberrypi.cpp:805 Stream ISP Output0 has color space JPEG
  [131:45:30.270953651] [43764] DEBUG RPI raspberrypi.cpp:867 Setting ISP Output1 (internal) to 1200x900-YU12
```

Both select the same sensor mode of 4656x3496-pRAA, and look as expected to me there.

The horizontal lines look like they're losing maybe 16 pixels per line perhaps ? So I suspect there is some stride/width issue around that indeed. I believe multiples of 16 pixels should be well aligned/supported - so can we find out where the fault starts?

Could you experiment around changing the width between 3840, and 4656, going up in multiples of either 16 or 32 until you find the point it starts to become garbled? I'm curious if the breakage starts around the 4096 mark in particular.

chemicstry, 2022-08-16:
Yep it seems to need an alignment of 32. Here are my test results:
4096x3496 - works
4112x3496 - does not work
4128x3496 - works
4144x3496 - does not work
4160x3496 - works

The last multiple of 32 at 4640x3496 also works.

Is this an issue with a camera width/height step value of `2`, that should have been `32`? Or is this a gstreamer alignment requirement?

kbingham, 2022-08-16:
I would suspect we have an issue conveying strides through from libcamera to the gstreamer element perhaps, but it needs more digging in the gstlibcamerasrc component.

6by9, 2022-08-17:
> I think I recall that the RPi ISP can only do downscaling, not upscaling, so I wonder what the configuration was when you launched.

The main output of the ISP will happily scale anything in either direction. The low resolution output must be unity or a downscale from the high res one.

The image certainly looks like a stride issue.
We can support any active image size that is a multiple of 2 in width & height (that's only to avoid the weirdness of chroma subsampling with an odd number of luma lines), but the overall buffer geometry has constraints.
The output strides are dictated by the hardware having to start each plane on a 32 byte boundary. For NV12/NV21 this means 32 byte alignment on ```bytesperline```, but 64 byte for I420/YV12 (otherwise there are resolutions where the last plane isn't aligned).
Vertical alignment isn't a requirement (other than a multiple of 2 lines as mentioned above).

naushir, 2022-11-04:
Closing due to inactivity - and given this is a striding issue with the gstreamer component.

closed 2022-11-04 by naushir

chemicstry, 2022-11-04:
> Closing due to inactivity - and given this is a striding issue with the gstreamer component.

It's still an issue. Should this be forwarded to upstream libcamera?

kbingham, 2022-11-04:
Yes, bugs.libcamera.org please.