raspberrypi/libcamera #23: disable_camera_led=1 in /boot/config.txt no longer works?

Cerlancism, 2022-08-22:
`disable_camera_led=1` in `/boot/config.txt` worked previously until a recent `sudo apt full-upgrade`, the camera led no longer able to turn off. 

Also tried the following to disable LED:
- Seems picamera no longer supported on the 64 bit OS release, tried `picamera2`
  ```py
  import picamera2

  with picamera2.Picamera2() as camera:
    camera.led = False
  ```
- GPIO
   ```py
   import RPi.GPIO as GPIO
   GPIO.setmode(GPIO.BCM)
   CAMLED = 32
   GPIO.setup(CAMLED, GPIO.OUT, initial=False) 
   GPIO.output(CAMLED,False)
   ```
    - Gets warning though `<stdin>:1: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.` 

My hardware and software info:
```
libcamera-apps build: 35266e84043f 07-07-2022 (17:59:45)
libcamera build: v0.0.0+3700-f30ad033
```
```
OS: Debian GNU/Linux 11 (bullseye) aarch64
Host: Raspberry Pi 3 Model B Plus Rev 1.3
Kernel: 5.15.56-v8+
Uptime: 22 mins
Packages: 1599 (dpkg)
Shell: bash 5.1.4
Terminal: /dev/pts/0
CPU: BCM2835 (4) @ 1.400GHz
Memory: 254MiB / 859MiB
```

6by9, 2022-08-22:
libcamera has NEVER supported the LED, therefore ```disable_camera_led=1``` has never done anything useful under libcamera. (Minor exception with the very original OV5647 driver which used GPIOs directly instead of via the regulator framework).

This is a firmware issue if using the basic start.elf, not startx or start_db. A fix has been created for review and release.

6by9, 2022-08-22:
https://forums.raspberrypi.com/viewtopic.php?t=338917 for the forum thread.

6by9, 2022-09-19:
Issue found in the firmware probe if start_x=1 is not set in config.txt (shouldn't be necessary for libcamera). It's been fixed in the latest firmware releases.

closed 2022-09-19 by 6by9