raspberrypi/libcamera #25: imx219 Rev.D instable behavior, init fails

torstenschenk, 2022-09-18:
**Describe the bug**
imx219 Rev.C (UC-350) working fine on raspberry4 64bit with libcamera interface
imx219 **Rev.D** (UC-350) not working with various issues with libcamera interface

**Bug report**
[bug-report.txt](https://github.com/raspberrypi/libcamera/files/9594050/bug-report.txt)

**Additional context**
Some imx219 Rev.D can run as secondary camera on multi camera boards. Several have issues on initialization with a direct connection to raspberry. It is unclear why there is variation in behavior the Rev.D camera have been ordered in one batch from the producer.

dmesg log is shown regardless if camera imx219 C or D is connected on boot:
[    7.055467] imx219 10-0010: Consider updating driver imx219 to match on endpoints

Updating OS, eeprom, rpi-update did not help.

6by9, 2022-09-19:
UC-350 is presumably an Arducam camera module. If there are issues with those modules that are not present with the Raspberry Pi IMX219 camera module, then you need to speak to Arducam about it. https://forum.arducam.com/ or https://www.arducam.com/contact-arducam/

naushir, 2022-11-04:
Is there anything further to look at here?  If not, I'll close this issue down shortly.

naushir, 2022-11-07:
Closing now...

closed 2022-11-07 by naushir