raspberrypi/libcamera #26: Cam utility is not listing the camera in RPI 3B+

savidha777, 2022-09-20:
Hi

In RDK-C RPI camera image, Added "dtoverlay=imx219" in /boot/config.txt file and rebooted the system.

Once system rebooted , tried to list out available camera in RPI system with 'cam -l' utility.

But it is not listing the camera. After the some prints binary is hanged.

Used latest libcamera revision : 74ab3f778c848b20cbf8fe299170756ff6ebab1a

Please find below log:
root@raspberrypi3-rdk-camera:/# LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[0:07:24.203442902] [2744] DEBUG IPAModule ipa_module.cpp:329 ipa_ipu3.so: IPA module /usr/lib/libcamera/ipa_ipu3.so is signed
[0:07:24.204257225] [2744] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_ipu3.so'
[0:07:24.204645246] [2744] DEBUG IPAModule ipa_module.cpp:329 ipa_rkisp1.so: IPA module /usr/lib/libcamera/ipa_rkisp1.so is signed
[0:07:24.204989882] [2744] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_rkisp1.so'
[0:07:24.205391288] [2744] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/lib/libcamera/ipa_rpi.so is signed
[0:07:24.205846496] [2744] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_rpi.so'
[0:07:24.206135559] [2744] DEBUG IPAModule ipa_module.cpp:329 ipa_vimc.so: IPA module /usr/lib/libcamera/ipa_vimc.so is signed
[0:07:24.206385871] [2744] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_vimc.so'
[0:07:24.206666548] [2744]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:07:24.207170871] [2745] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[0:07:24.221253736] [2745] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media2
[0:07:24.221469413] [2745] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media2 due to 3 missing dependencies
[0:07:24.225524621] [2745] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media2 found
[0:07:24.225669829] [2745] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-codec

Binary is hanged after this print.

#################################################################################################

But able to list out camera in RPI system with oldest libcamera revision

Used libcamera revision : 58d487b10e06ad823103957c1e6aa2201720fdae (30Oct2020)

Working log:
[Working_old_libcamera_log.txt](https://github.com/raspberrypi/libcamera/files/9606541/Working_old_libcamera_log.txt)

savidha777, 2022-09-20:
Currently we are using below kernel and boot loader version for RPI

Kernel:
SRC URI : git://github.com/raspberrypi/linux.git
Branch : rpi-5.4.y

Boot loader:
https://github.com/raspberrypi/firmware/archive/1eb8921ae7158249894f89e8bb6c4bfb04ec7377.tar.gz
SRC_REV : 1eb8921ae7158249894f89e8bb6c4bfb04ec7377 (19Nov2020)

Is this kernel and boot loader version is okay to check libcamera use case or else any specific RPI kernel and boot loader version required for libcamera.

kbingham, 2022-09-20:
Are you able to connect with GDB and get a backtrace of all threads to see where it is blocked perhaps?

The cam tool itself shouldn't hang ...

kbingham, 2022-09-20:
>  SRC_REV : 1eb8921ae7158249894f89e8bb6c4bfb04ec7377 (19Nov2020) 

This sounds quite old?

savidha777, 2022-09-20:
> Are you able to connect with GDB and get a backtrace of all threads to see where it is blocked perhaps?
> 
> The cam tool itself shouldn't hang ...

Okay, Will check and update

6by9, 2022-09-20:
> Currently we are using below kernel and boot loader version for RPI
>
> Kernel:
> SRC URI : git://github.com/raspberrypi/linux.git
> Branch : rpi-5.4.y

All the libcamera development is against rpi-5.15.y.
I wouldn't like to take any bets over which libcamera functionality works on rpi-5.10.y, let alone rpi-5.4.y.

savidha777, 2022-09-20:
Are you suggesting to use rpi-5.15.y for libcamera?

6by9, 2022-09-20:
> Are you suggesting to use rpi-5.15.y for libcamera?

Yes.

rpi-5.4.y was last updated from mainline in December 2020. Mainline 5.4 is still updated on https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/?h=linux-5.4.y if you had a desperate need to rebase it, but be aware that it goes EOL Dec 2025 (https://www.kernel.org/category/releases.html)

rpi-5.10.y is maintained as the kernel for Raspberry Pi Legacy OS (based on Debian Buster instead of Bullseye), but it will only gain updates from mainline, and not new features.

kbingham, 2022-11-04:
@savidha777 When you get a back trace of all threads, please attach it here (not the other issue).

savidha777, 2022-11-04:
Hi @kbingham 

Please find back trace of all thread 

(gdb) info threads
  Id   Target Id         Frame 
* 1    LWP 1556 "cam"    0x76b65a94 in ?? () from /lib/libpthread.so.0
  2    LWP 1693 "cam"    0x76b65a94 in ?? () from /lib/libpthread.so.0
(gdb) thread apply all bt

Thread 2 (LWP 1693):
#0  0x76b65a94 in ?? () from /lib/libpthread.so.0
#1  0x76b63d84 in ?? () from /lib/libpthread.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (LWP 1556):
#0  0x76b65a94 in ?? () from /lib/libpthread.so.0
#1  0x76b61376 in pthread_cond_wait () from /lib/libpthread.so.0
#2  0x76e3639e in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /usr/lib/libstdc++.so.6
#3  0x76f751f6 in std::condition_variable::wait<libcamera::CameraManager::Private::start()::<lambda()> > (__p=..., __lock=..., this=0x41588) at ../git/src/libcamera/camera_manager.cpp:90
#4  libcamera::ConditionVariable::wait<libcamera::CameraManager::Private::start()::<lambda()> > (stopWaiting=..., locker=..., this=0x41588) at ../git/include/libcamera/base/mutex.h:104
#5  libcamera::CameraManager::Private::start (this=0x41528) at ../git/src/libcamera/camera_manager.cpp:90
#6  0x76f7526c in libcamera::CameraManager::start (this=0x414e0) at /usr/include/c++/9.3.0/bits/unique_ptr.h:360
#7  0x00013fb6 in CamApp::init (argc=<optimized out>, argv=<optimized out>, this=0x7efffae8) at /usr/include/c++/9.3.0/bits/unique_ptr.h:360
#8  CamApp::init (this=0x7efffae8, argc=<optimized out>, argv=<optimized out>) at ../git/src/cam/main.cpp:69
#9  0x0000eb44 in main (argc=2, argv=0x7efffd34) at ../git/src/cam/main.cpp:346
(gdb)

kbingham, 2022-11-04:
This is a different backtrace location this time, and it shows that the CameraManager is still waiting for all the pipeline handlers to complete. Please *always* set LIBCAMERA_LOG_LEVELS=*:0 so we can see what libcamera is doing.

It could be worth running with strace too to see if that's showing more helpful information about where it is blocked.
- [ ]  `LIBCAMERA_LOG_LEVELS=*:0 strace -e ioctl -f -- cam -l`

Or for a really noisy log:
- [ ] `LIBCAMERA_LOG_LEVELS=*:0 strace -f -- cam -l`*

Something very odd going on with your system. The fact that thread2 is not providing a stack trace is perhaps worrying too.

- [ ] Another thing to try would be running with valgrind or something that adds memory validation checks?
- [ ] Are you able to reproduce this using Raspberry Pi OS?
      Or is this an issue potentially caused by some differences in the build of your filesystem?
- [ ] Can you confirm that you're on an updated firmware now? (That's still important).

- [ ] Can you run the libcamera test suite (meson test)? That could check for things like the core threads and internal functionality - it seems like something very suspicious is going on in your environment.

It would also be helpful/interesting to see the results of the meson configuration stage to see how you are building libcamera.

naushir, 2022-11-04:
Would you be able to run the ``camera-bug-report`` command, and share the output file generated please?

savidha777, 2022-11-07:
Hi @naushir 

No, We don't have this camera-bug-report utility in RDK-C RPI image.

root@raspberrypi3-rdk-camera:~# camera-bug-report
-sh: camera-bug-report: command not found
root@raspberrypi3-rdk-camera:~#

naushir, 2022-11-07:
You can find the tool in the ``utils/`` directory of the libcamera-apps repo (https://github.com/raspberrypi/libcamera-apps/tree/main/utils).

savidha777, 2022-11-10:
Hi @naushir 

Please find attached camera-bug-report 
[bug-report.txt](https://github.com/raspberrypi/libcamera/files/9981711/bug-report.txt)

savidha777, 2022-11-10:
Hi @kbingham 

> * LIBCAMERA_LOG_LEVELS=*:0 strace -f -- cam -l

Please find this attached strace log  
[strace.txt](https://github.com/raspberrypi/libcamera/files/9981756/strace.txt)

> * Are you able to reproduce this using Raspberry Pi OS?
**cam -l** utility is working fine with Raspberry Pi OS
pi@raspberrypi:/ $ cam -l
[0:22:46.949737158] [6417]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+54-d528119f
[0:22:47.017153802] [6418]  WARN RPI raspberrypi.cpp:1297 Mismatch between Unicam and CamHelper for embedded data usage!
[0:22:47.018680878] [6418]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
**Available cameras:
1: 'imx219' (/base/soc/i2c0mux/i2c@1/imx219@10)**

> * Can you confirm that you're on an updated firmware now? (That's still important).
Using below revision of firmware,kernel and libcamera.
Raspberrypi Firmware:
------------------------------
https://github.com/raspberrypi/firmware/archive/${SRCREV}.tar.gz
SRCREV : ab37ef59f67f8edc4cf1383a94e39edee03bcf00 (10-Oct-2022)

Rpi Kernel
---------------
git://github.com/raspberrypi/linux.git
SRCREV : dbd073e4028580a09b6ee507e0c137441cb52650 (17-Oct-2022)
Linux_Version : 5.15.74

libcamera:
--------------
git://linuxtv.org/libcamera.git
SRCREV : 685da13d02c01ed0a2bb7080f95b7ac56e80f221 (21-Sep-2022)

naushir, 2022-11-11:
It's going to be very hard to debug your issue here given you are running a custom OS environment.  I would suggest you first try with a Raspberry Pi OS distribution and see if everything works for you.  If it does, you may want to raise an issue with the folks who package your OS about how they build/include the libcamera libraries.

kbingham, 2022-11-11:
@savidha777 - could we perhaps re-clarify what issue you are really trying to solve here ?

In the log you've sent - it includes:

** libcamera-hello --list-cameras **
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

This shows that the camera *is* correctly being identified, found and enumerated.

savidha777, 2022-11-11:
Hi @kbingham 

libcamera-hello --list-cameras -> This utility is listing the camera, This is provided by libcamera-apps repo

we are getting hang issue with **"cam -l"** utility.
pls find this strace log file 
[strace.txt](https://github.com/raspberrypi/libcamera/files/9990395/strace.txt)

kbingham, 2022-11-11:
The fact that it's hanging with cam, but not libcamera-hello is *very* odd.

How have you got libcamera-hello on the target, is it different to how you've got cam on the target?

Could you build and test 'simple-cam' please?
https://git.libcamera.org/libcamera/simple-cam.git/

What about qcam - or the gstreamer element - are there other libcamera enabled applications that you are able to test?

Are you running the packages installed ? or building them and running from the build directory on the target?

savidha777, 2022-11-17:
Hi @kbingham 

We have used yocto recipe for the libcamera compilation. Used same steps only for both cam and libcamera-hello bring-up.

libcamerasrc gstreamer plugin is working fine with upgraded 5.15 kernel, but cam utility only is hanged. 

hang issue observed libcamera revision : 685da13d02c01ed0a2bb7080f95b7ac56e80f221 (21-09-2022)

NOTE:
Now cam utility hang issue got resolved with a8113fb3a89984cc65d51436480cee45b60543e8 (25-10-2022) revision.

kbingham, 2022-12-05:
Is this fully resolved now?

savidha777, 2022-12-05:
Yes, It's resolved.

savidha777, 2022-12-05:
Cam utility hang issue got resolved with latest libcamera revision a8113fb3a89984cc65d51436480cee45b60543e8

closed 2022-12-05 by savidha777