raspberrypi/libcamera #27: After update libcamera Cannot allocate memory

dzyla, 2022-09-26:
Hi,

I am using RPi 3B and had a running script using picamera2. However, after the update of picamera2, I had to update the system, and now, I cannot use the Pi HQ camera on Pi3B, but the problem seems to be related to libcamera itself, not picamera2. The error I am getting is:

```
pi@raspberrypi:~ $ libcamera-jpeg -o test.jpg --nopreview
[0:00:31.443247607] [1056]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3866-0c55e522
[0:00:31.588132350] [1057]  INFO RPI raspberrypi.cpp:1374 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:00:31.590505922] [1056]  INFO Camera camera.cpp:1035 configuring streams: (0) 2028x1520-YUV420
[0:00:31.591771616] [1057]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:00:37.311628603] [1056]  INFO Camera camera.cpp:1035 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:00:37.314459826] [1057]  INFO RPI raspberrypi.cpp:761 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[0:00:37.318026212] [1057] ERROR V4L2 v4l2_videodevice.cpp:1234 /dev/video14[15:cap]: Unable to request 1 buffers: Cannot allocate memory
ERROR: *** failed to allocate capture buffers ***
```

I tried RPi camera 1.3, and it works well. The details of my system:
`Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux`

I have activated KMS GL driver and glamor.

My boot.txt:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
#dtoverlay=imx477 (does not work)
#dtoverlay=vc4-fkms-v3d (does not work)
#max_framebuffers=2 (also does not work)

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtparam=act_led_trigger=actpwr
#gpu_mem=76
disable_camera_led=1
```
If someone could guide me and show me how to fix that issue, that would be great. 
Thanks,
Dawid

dzyla, 2022-09-26:
Ok, after an update to the boot.txt, I added:

```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
dtoverlay=imx477
max_framebuffers=6
```
And it seems to be working. Before the update, RPi was automatically detecting the camera sensor. 

In addition, updated picamera2 works as well.

naushir, 2022-11-04:
Do not change the default value of ``max_framebuffers`` unless you need to.  This parameter is **not** use for camera frame buffer allocation.

naushir, 2022-11-04:
If there is nothing else pending for this issue, I'll resolve it.  Feel free to reopen if you are still having problems.