raspberrypi/libcamera #31: Documentation claims this can be built with Meson 0.56, but 0.58 is required

Newbytee, 2022-11-04:
Hello! :wave: 

Currently, your fork of libcamera advertises itself as only requiring Meson 0.56. However, this is not actually the case as your meson.build uses the replace() function for strings, which was introduced in Meson 0.58: https://mesonbuild.com/Release-notes-for-0-58-0.html#string-replace

To resolve this you can cherry-pick this commit from upstream: https://git.libcamera.org/libcamera/libcamera.git/commit/meson.build?id=02518e598e8f057b76a08940e1bc8a99d879fad6

Or just rebase.

kbingham, 2022-11-04:
Fetching/mirroring the tags here would also prevent the issue from being hit. The code path is only executed when the 0.0.1 tag isn't available in the tree, which can happen if you clone from a tree without the tags.

kbingham, 2022-11-04:
(fetching tags from the upstream repo is another workaround if you hit the bug locally)

naushir, 2022-11-04:
This will be updated in our next libcamera release.

naushir, 2022-11-09:
The latest libcamera release on apt has the change to remove the string replace().

closed 2022-11-09 by naushir