raspberrypi/libcamera #33: Getting Started fails

ktorimaru, 2022-11-19:
ninja -C build install
ninja: Entering directory `build'
[1/2] Installing files.
Installation failed due to insufficient permissions.
Attempting to use polkit to gain elevated privileges...
==== AUTHENTICATING FOR org.freedesktop.policykit.exec ===
Authentication is needed to run `/usr/bin/python3' as the super user
Authenticating as: root
Password: 
polkit-agent-helper-1: error response to PolicyKit daemon: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: No session for cookie
==== AUTHENTICATION FAILED ===
Error executing command as another user: Not authorized

This incident has been reported.
FAILED: meson-internal__install 
/home/pi/.local/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.

kbingham, 2022-11-19:
Try calling 'sudo ninja -C build install'
 root permission is required to install in the host system

ktorimaru, 2022-11-21:
Thanks for the suggestion. I have tried that:

sudo ninja -C build install
ninja: Entering directory `build'
[1/2] Installing files.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mesonbuild/mesonmain.py", line 140, in run
    return options.run_func(options)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 554, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 359, in do_install
    d = self.check_installdata(pickle.load(ifile))
AttributeError: Can't get attribute 'OctalInt' on <module 'mesonbuild.coredata' from '/usr/lib/python3/dist-packages/mesonbuild/coredata.py'>
FAILED: meson-internal__install 
/home/pi/.local/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.

Also tried:
sudo apt upgrade meson
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
meson is already the newest version (0.56.2-1).
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

ktorimaru, 2022-11-21:
Although I know I ran:
   pip3 install --user --upgrade meson

I ran it again and this time:
  sudo ninja -C build install

Worked!

Thanks!

closed 2022-11-21 by ktorimaru