raspberrypi/libcamera #35: How should a high-quality still stream and a fast video-stream work together?

pietermarsman, 2022-12-06:
Hiya! 

Thanks for creating this amazingly well written package!

I'm working on a usecase where we want to capture a video and capture a still at somewhere during the video. The video should have a decent fps, and the still should be of reasonable quality. I understand that these two requirements don't go well together, but I do need to work my way around it best as I can. 

I want to know what the recommended solution in cases like this is. I can think of the following general strategies, but I don't know what will work best.
- Stop the camera, clear all associated resources, load a new configuration, and start it again from scratch. 
    - This approach is taken by the libcamera-still app from libcamera-apps. But it takes roughly 0.5 seconds and in those seconds there are no frames recorded for the video. 
- Stop the camera, not clear all the resources, load the new configuration, and do a warm start.
    - I've tried this, but could not get it to work yet. I dont understand if I can keep the `Stream`, `Request`, `FrameBuffer` and `Allocator` objects or not. 
- Use a single stream and update the configuration while the camera is running. I don't think this is possible because the `Camera` needs to be in the `Acquired` or `Configured` state to be able to configure it again. 
- Temporary disable a stream. I don't think this can be done, but happy to be surprised :) 
- Do a `Request` for a subset of the configured streams. I don't know if this is at all possible. 

Plenty of ideas, but I don't have a clue on what is the recommend approach. Happy to add more detail if needed.

naushir, 2022-12-07:
Hi @pietermarsman, thank you for the report.  To start with, what sensor are you using, and what resolution still image capture are you looking for?  Is it going to be full sensor resolution, or will something like 1080p (i.e. same as the video encode) be good enough? Because of the limited throughput of the VC4 ISP, we are unfortunately not able to pass through large (i.e. > 5Mpix) resolutions through the ISP at full video rates, so there will be certain limitations to work around.

Although libcamera apps can be bent to do what you want here, it might be better to look at picamera2 for such tasks.  In fact, one of our examples here does pretty much [this](https://github.com/raspberrypi/picamera2/blob/main/examples/still_during_video.py)!  I'm sure @davidplowman can provide more details on this script if you need.

naushir, 2022-12-07:
Any dynamic re-configuration is going to cause interruptions in the video stream, this is unavoidable.  We can look to ways of minimising this if needed.  However to start with, I would look at the picamera2 script, setup a video/capture resolution of 1920x1080, and evaluate the quality to see if it is acceptable.

pietermarsman, 2022-12-07:
I'll definitely do that. Thanks for the help so far!

pietermarsman, 2022-12-13:
I've checked out the script. I think it works like this:
- The `picam2.create_video_configuration` configures the camera for high framerate. E.g. it uses fast denoising.
- The original capture from the camera is half its sensor resolution, this is probably fast because it can use 2x2 binning. 
- The video is recorded in 640x480. The image is captured in half the sensor resolution. 

I can see that this works, and why it works. The camera is configured with just one configuration with multiple streams. 

Since the overall framerate is determined by the slowest stream this does not fit my usecase. I want to run the video at high framerates 99% of the time. And very occasionally capture a high-quality image. E.g. an image that would have been captured with the `picam2.create_still_configuration` settings. I'm still wondering what the best way to do this is. And what you would recommend in this scenario. 

(Thanks for your help so-far, I'm still reading through the docs and trying things out :smile: )

pietermarsman, 2022-12-13:
The `picam2.switch_mode_and_capture_file` looks pretty interesting to me. From what I can see that switches to another mode and captures a high-quality image in about 0.5 seconds. So 0.5 seconds of frame drop.

davidplowman, 2022-12-14:
Yes, unfortunately with the hardware we have there really is no way round this. You can have 12MP, but you will max out at 10fps. Or you can have 30fps but that will max out at 3MP. And switching between camera modes involves stopping the sensor, re-programming and re-starting it which is relatively slow.

naushir, 2022-12-14:
As @davidplowman noted, the solution comes down to a choice of either:
1) A lower resolution stills capture to avoid any glitches in the video stream.
2) A high resolution stills capture if approx. 1s of glitches in the video stream are acceptable.

naushir, 2022-12-14:
There is another option (3) that perhaps you might want to consider:

3. Use the camera v2 instead of the HQ camera.  This allows you to use a 8MPix 20fps mode.  Based on some quick calculations,  it ought to be possible for the ISP to achieve the 8*20 = 160 MPix/s.  Then you might get a glitch-free video encode stream, and a 8Mpix still capture periodically.  You may have to try this out and tuning framerates and output resolutions to see what exactly is achievable.

naushir, 2023-01-04:
Is there anything more to look at here?  If not, I'll close this issue down.

naushir, 2023-01-04:
Closing now.  Feel free to re-open if you still require more information.

closed 2023-01-04 by naushir