raspberrypi/libcamera #36: Rotate 90 Deg

mrfoxie, 2022-12-13:
is there any way to rotate **Pi Camera Module 2 NoIR** to 90 Deg it's just only rotate 0 & 180 Deg

naushir, 2022-12-13:
Unfortunately it is not possible to perform transpose based operations through the hardware.  Your best approach would be to do this in software after the request has completed.

closed 2022-12-13 by naushir

6by9, 2022-12-13:
>  Your best approach would be to do this in software after the request has completed.

If capturing JPEG you can store the rotation in the EXIF ORIENTATION tag, and for MP4 files the composition matrix can do a similar thing. The viewer will then do the rotation.
Similarly if displaying through X then it should be able to do a rotation for the preview.
This is the approach that has been taken by most mobile phones for at least the last decade.

mrfoxie, 2022-12-13:
I have smartpi case and in that i have option to add came but it's 90 so when i use online its on 0 ot 180 i used this command in `sudo nano /etc/rc.local`
```
# above exit 0
v4l2-ctl --set-ctrl=rotate=90
```
but didn't work