raspberrypi/libcamera #37: How are '--awb custom' and '--awbgains' working?

lxne, 2022-12-20:
Hi, I am using a Raspberry Pi HighQuality Camera with a Sony IMX477 sensor. Playing around with libcamera on a Raspberry Pi 4 running Bullseye.

There are two things I do not understand and I couldn't find any enlightening information:

1. What does '--awb custom' do? The other presets (tungten, indoor, ...) are clear to me, but with custom I would assume one would have to give some more arguments like --awb custom 5600 (for Kelvin). But that is not working.
2. How does '--awbgains 0,0' work? It should control red-green and blue-yellow shift, right? I do not understand the systematics behind it. For example I tried '--awbgains 1.0,1.0' and everything turned green. '--awbgains 1.0,0.0' on the other hand did nothing to the image.

I would appreciate some help.

Best wishes,
Alex

naushir, 2023-01-03:
Hi Alex!

> * What does '--awb custom' do? The other presets (tungsten, indoor, ...) are clear to me, but with 'custom' I would assume one would have to give some more arguments like '--awb custom 5600' (for Kelvin). But that is not working.

If you look at one of the sensor tuning files, under the section "rpi.awb.modes" (e.g. [here](https://github.com/raspberrypi/libcamera/blob/0684c3735eb5392d79b9919b62d8b04ac9b50cc1/src/ipa/raspberrypi/data/imx477.json#L76)), you will see a list of modes that specify a custom CT range to search for.  ``--awb custom`` chooses the "custom" mode, if you have specified one in the tuning file under the "rpi.awb.modes.custom" key.  It's essentially an easy way of specifying a custom CT range without having to adjust one of the other modes. 

> * How does '--awbgains 0,0' work? It should control red-green and blue-yellow shift, right? I do not understand the systematics behind it. For example I tried '--awbgains 1.0,1.0' and everything turned green. '--awbgains 1.0,0.0' on the other hand did nothing to the image.

``--awbgains`` sets explicit red and blue gain values to colour balance the image.  So for example, '--awbgains 1.1,2.0' sets the red gain to 1.1 and blue gain to 2.0.

Hope this helps!

lxne, 2023-01-03:
Hi naushir,

thank you very much for your answer.
I didn't know about the tuning file. This off course makes sense now.

In principal `--awbgains`  is understandable, too. But I still don't understand why `--awbgains 1.0,0.0` is giving me the exact same image as without any gains, `--awbgains 1.0,1.0` on the other hand is turning everything green. Gain is a Faktor, right? I thought '1.0,1.0' would do nothing to the image and smaller or higher values would change the corresponding balance. What am I missing?

Best wishes!

naushir, 2023-01-03:
> In principal `--awbgains` is understandable, too. But I still don't understand why `--awbgains 1.0,0.0` is giving me the exact same image as without any gains, `--awbgains 1.0,1.0` on the other hand [...]

Correct, gain is a linear scaling factor on the red/blue colour channels.  '1.0,1.0' apply a 1x gain, and so will not modify any channel values, which means that the image is not white-balance corrected and will look very green.

One thing I am unsure of is what happens if you use gain values of less than 1.0 - theoretically this should be fine, but I have not tested it myself.  For now, I suggest use any value > 1.0 for either gain value.

lxne, 2023-01-03:
Thank you naushir, your answers are really helpful.

Now that I do not have to guess that much any more I will be playing around with my newly acquired knowledge.

closed 2023-01-03 by lxne

lxne, 2023-01-23:
> In principal `--awbgains` is understandable, too. But I still don't understand why `--awbgains 1.0,0.0` is giving me the exact same image as without any gains, `--awbgains 1.0,1.0` on the other hand is turning everything green.
Found out that only values greater than 0 work. If one of the values is 0 both values for awbgains are ignored.
Example:
0.0,5 is not working (awbgains argument is being ignored)
0.001,5 is working