raspberrypi/libcamera #39: Raspberry PI 4 B 8Gb + USB camera: gstreamer plugin doesn't work

fghoussen, 2023-01-05:
Running Raspberry PI 4 B 8Gb with fresh and updated install of raspbian.

```
pi@raspberrypi:~/Workspaces/libcamera $ sudo apt-get update && sudo apt-get upgrade && sudo apt-get full-upgrade && sudo apt-get dist-upgrade
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@raspberrypi:~/Workspaces/libcamera $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date (delete /boot/.firmware_revision to force an update anyway)

pi@raspberrypi:~/Workspaces/libcamera $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

Installing and testing gstreamer which does work: the command line below runs as expected

```
pi@raspberrypi:~/Workspaces/libcamera $ gst-launch-1.0 videotestsrc ! ximagesink
```

Plugging USB camera (video0 and video1 - vlc enable to capture /dev/video0: the images are displayed by vlc as expected)
```
pi@raspberrypi:~/Workspaces/libcamera $ ll /dev/video*
crw-rw----+ 1 root video 81, 14 Jan  5 14:17 /dev/video0
crw-rw----+ 1 root video 81, 15 Jan  5 14:17 /dev/video1
crw-rw----+ 1 root video 81,  2 Jan  5 14:17 /dev/video10
crw-rw----+ 1 root video 81,  4 Jan  5 14:17 /dev/video11
crw-rw----+ 1 root video 81,  6 Jan  5 14:17 /dev/video12
crw-rw----+ 1 root video 81,  1 Jan  5 14:17 /dev/video13
crw-rw----+ 1 root video 81,  3 Jan  5 14:17 /dev/video14
crw-rw----+ 1 root video 81,  5 Jan  5 14:17 /dev/video15
crw-rw----+ 1 root video 81,  7 Jan  5 14:17 /dev/video16
crw-rw----+ 1 root video 81,  8 Jan  5 14:17 /dev/video18
crw-rw----+ 1 root video 81,  0 Jan  5 14:17 /dev/video19
crw-rw----+ 1 root video 81, 10 Jan  5 14:17 /dev/video20
crw-rw----+ 1 root video 81, 11 Jan  5 14:17 /dev/video21
crw-rw----+ 1 root video 81, 12 Jan  5 14:17 /dev/video22
crw-rw----+ 1 root video 81, 13 Jan  5 14:17 /dev/video23
crw-rw----+ 1 root video 81,  9 Jan  5 14:17 /dev/video31
```

Now, compiling from source https://github.com/raspberrypi/libcamera (with `meson --prefix=/path/to/libcamera/local build`), 

```
pi@raspberrypi:~/Workspaces/libcamera $ gst-device-monitor-1.0 Video
Probing devices...

[0:57:37.249909054] [4654]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373

Device found:

	name  : /base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366
	class : Source/Video
	caps  : image/jpeg, width=160, height=120
	        image/jpeg, width=320, height=240
	        image/jpeg, width=640, height=480
	        image/jpeg, width=800, height=600
	        image/jpeg, width=1280, height=720
	        image/jpeg, width=1280, height=960
	        image/jpeg, width=1600, height=1200
	        video/x-raw, format=YUY2, width=800, height=600
	        video/x-raw, format=YUY2, width=1280, height=720
	        video/x-raw, format=YUY2, width=1600, height=1200
	gst-launch-1.0 libcamerasrc camera-name="/base/scb/pcie\@7d500000/pci\@0\,0/usb\@0\,0-1.1:1.0-0c45:6366" ! ...

pi@raspberrypi:~/Workspaces/libcamera $ GST_PLUGIN_PATH="/home/sbg/Workspaces/libcamera/local/lib/aarch64-linux-gnu/gstreamer-1.0" gst-launch-1.0 libcamerasrc camera-name="/base/scb/pcie\@7d500000/pci\@0\,0/usb\@0\,0-1.1:1.0-0c45:6366" ! ximagesink
Setting pipeline to PAUSED ...
[0:58:03.345601554] [4658]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0: Internal data stream error.
Additional debug info:
../src/gstreamer/gstlibcamerasrc.cpp(593): gst_libcamera_src_task_enter (): /GstPipeline:pipeline0/GstLibcameraSrc:libcamerasrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.016944832
Setting pipeline to NULL ...
Freeing pipeline ...

```

Looks like `gst-launch-1.0 libcamerasrc` does not work with USB camera?!... Is this expected or what did I miss?

I am trying to understand if the compiled-from-source `libcamera` can get live images from USB cameras on Raspberry PI 4 B 8Gb: is this supported?

Note: `cam` seems to run but no way to get images from it.

```
pi@raspberrypi:~/Workspaces/libcamera $ ./local/bin/cam --list
[1:00:45.891999956] [4669]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Available cameras:
1: 'Arducam OV2311 USB Camera: Ardu' (/base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366)

pi@raspberrypi:~/Workspaces/libcamera $ ./local/bin/cam --camera 1 --capture
[1:01:33.762594353] [4674]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Using camera /base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366 as cam0
[1:01:33.921108221] [4674]  INFO Camera camera.cpp:1026 configuring streams: (0) 1600x1200-MJPEG
cam0: Capture until user interrupts by SIGINT
3694.555635 (0.00 fps) cam0-stream0 seq: 000000 bytesused: 49784
3694.727667 (5.81 fps) cam0-stream0 seq: 000001 bytesused: 49880
3694.759673 (31.24 fps) cam0-stream0 seq: 000002 bytesused: 50712
^CExiting

pi@raspberrypi:~/Workspaces/libcamera $ ./local/bin/cam --camera 1 --capture -D
[1:01:38.033678349] [4676]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Using camera /base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366 as cam0
[1:01:38.115038485] [4676]  INFO Camera camera.cpp:1026 configuring streams: (0) 1600x1200-MJPEG
Failed to open /dev/dri/ directory: No such file or directory
Failed to open any DRM/KMS device: No such file or directory
Failed to configure frame sink
Failed to start camera session

```

Note: first, I tried to build from source `libcamera-apps` but get
```
pi@raspberrypi:~/Workspaces/libcamera-apps $ ./local/bin/libcamera-hello 
libEGL warning: DRI2: failed to authenticate
Made X/EGL preview window
[1:06:53.916145384] [4689]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
ERROR: *** no cameras available ***
```
Narrowing down the problem, I realized that gstreamer plugin from `libcamera` didn't work and that I was not sure if `cam` from `libcamera` was able to fetch live images from USB camera.

kbingham, 2023-01-05:
Can you try with `gst-launch-1.0 libcamerasrc ! xvimagesink` instead? or `gst-launch-1.0 libcamerasrc ! glimagesink` please?

It should work.

fghoussen, 2023-01-05:
@kbingham: can't use `glimagesink` (from googling which is supposed to be installed in `gstreamer1.0-plugins-base-apps` which is indeed installed)

```
pi@raspberrypi:~/Workspaces/libcamera $ GST_PLUGIN_PATH="/home/sbg/Workspaces/libcamera/local/lib/aarch64-linux-gnu/gstreamer-1.0" gst-launch-1.0 libcamerasrc camera-name="/base/scb/pcie\@7d500000/pci\@0\,0/usb\@0\,0-1.1:1.0-0c45:6366" ! glimagesink
WARNING: erroneous pipeline: no element "glimagesink"

pi@raspberrypi:~/Workspaces/libcamera $ sudo apt-get install apt-file gstreamer1.0-tools gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
apt-file is already the newest version (3.2.2).
gstreamer1.0-plugins-bad is already the newest version (1.18.4-3).
gstreamer1.0-plugins-base-apps is already the newest version (1.18.4-2).
gstreamer1.0-tools is already the newest version (1.18.4-2.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

fghoussen, 2023-01-05:
Adding boot config file in case this may help:

```
pi@raspberrypi:~ $ cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#dtoverlay=arducam-pivariety
#dtoverlay=ov9281
dtoverlay=camera-mux-2port,cam0-ov9281,cam1-ov9281
force_turbo=1
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtparam=i2c_vc=on

```

fghoussen, 2023-01-06:
Any clue? Help? Ideas to test?

Back here, USB camera plugged (I see `/dev/video[01]`), looks like I have no much solution:

- VLC is able to capture stream from `/dev/video0`

- a simple python script using `opencv` works as expected

```
pi@raspberrypi:~/Workspaces/testUSBCam $ cat testUSBCam.py 
import cv2

cam = cv2.VideoCapture(0)
while True:
	ret, image = cam.read()
	cv2.imshow('testUSBCam', image)
	k = cv2.waitKey(1)
	if k != -1:
		break
cam.release()
cv2.destroyAllWindows()

```

- `cam` from `libcamera` (only available when built from source - seems not present in debian native package) is able to get images from USB camera: it writes them on disk but no way to get them in a live GUI

- gstreamer plugin from `libcamera` doesn't work

- `libcamera-hello` from `libcamera` does not work

```
pi@raspberrypi:~/Workspaces $ ./libcamera-apps/local/bin/libcamera-hello --list-cameras
No cameras available!

pi@raspberrypi:~/Workspaces $ ./libcamera-apps/local/bin/libcamera-hello
Made X/EGL preview window
[0:37:32.742973124] [2498]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
ERROR: *** no cameras available ***

```

jmondi, 2023-01-06:
> * `cam` from `libcamera` (only available when built from source - seems not present in debian native package) is able to get images from USB camera: it writes them on disk but no way to get them in a live GUI

cam is just  a test application, so it's not expected to have a GUI interface. It can however display frames in an SDL2 window (-S option) if libsdl2-dev and libsdl2_image-dev are installed when building libcamera. 

> * gstreamer plugin from `libcamera` doesn't work

Can you provide a tad more debug info ?
Try running with LIBCAMERA_LOG_LEVELS=0 and enable gst debug with the usual GST_DEBUG=..  (careful it might get rather verbose)

naushir, 2023-01-06:
Raspberry Pi's libcamera-apps suite are currently only able to work with Bayer sensors, see [here](https://github.com/raspberrypi/libcamera-apps/blob/4fea2eed68300dcc88e89aa30da6079d10dce822/core/libcamera_app.cpp#L104).

kbingham, 2023-01-06:
@fghoussen Could you also perhaps explain a bit more about what you're actually trying to accomplish overall?

As stated by @naushir the libcamera-apps provided by Raspberry Pi do not accept USB cameras.

For a USB camera you do not need to use libcamera, but you should be able to (excluding libcamera-apps provided by Raspberry Pi). Looking at your command above when trying to use gstreamer, I wonder if setting the GST_PLUGIN_PATH is affecting it from finding the existing plugins. Please make sure that gstreamer can find the desired components with `gst-inspect-1.0 glimagesink` and `gst-inspect-1.0 xvimagesink` or such.

As you are working with a USB camera, you could also use v4l2src from gstreamer instead of the libcamerasrc.

fghoussen, 2023-01-06:
To sum up USB are not supported, and, CSI-MIPI can not be detected... The problem is that no CSI-MIPI cameras have never been detected (I realise that just now...) on my raspberry pi: is there some patch I could apply (guided by you) to check why this occurs?...

Do I need to change the OS? If yes which one? Any other clue?

> libcamera-apps provided by Raspberry Pi do not accept USB cameras.

Really?!.. If so this should be in top position in the doc... Didn't see it anywhere I looked. USB is a first entry point for almost everyone?!... I engage you to document that in top position in README.md

> @fghoussen Could you also perhaps explain a bit more about what you're actually trying to accomplish overall?

I have one USB camera, two CSI-MIPI camera, a camarray hat (multiplexer that enables using 2 CSI cameras in stereo) and a fresh raspberry pi 4 8Gb: in each case I am just trying to understand how to get live images from cameras in different configuration (mono and/or stereo). This is just the basics.

As there is not way to get CSI-MIPI cams to work neither using gstreamer from `libcamera` nor using `libcamera-hello`, I started to step back and check if I could get to work USB cam using gstreamer from `libcamera` or `libcamera-hello`

> Can you provide a tad more debug info ?Try running with LIBCAMERA_LOG_LEVELS=0 

Seems no... I didn't know `libcamera` and `cam` where _not_ supposed to support USB cams... In https://github.com/raspberrypi/libcamera/issues/39#issue-1520946121, OV2311 was USB... Which means I have indeed _never_ been able to see/list a CSI-MIPI camera... 

```
pi@raspberrypi:~/Workspaces/libcamera $ ./local/bin/cam --list
[1:00:45.891999956] [4669]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Available cameras:
1: 'Arducam OV2311 USB Camera: Ardu' (/base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366)

pi@raspberrypi:~/Workspaces/libcamera $ ./local/bin/cam --camera 1 --capture --file
[1:01:33.762594353] [4674]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Using camera /base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366 as cam0
[1:01:33.921108221] [4674]  INFO Camera camera.cpp:1026 configuring streams: (0) 1600x1200-MJPEG
cam0: Capture until user interrupts by SIGINT
3694.555635 (0.00 fps) cam0-stream0 seq: 000000 bytesused: 49784
3694.727667 (5.81 fps) cam0-stream0 seq: 000001 bytesused: 49880
3694.759673 (31.24 fps) cam0-stream0 seq: 000002 bytesused: 50712
^CExiting
```

Running the only one USB camera I have ever been able to detect

```
pi@raspberrypi:~/Workspaces/libcamera $ ./local/bin/cam --camera 1 -S
[0:22:58.799681705] [2014]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+47-0684c373
Using camera /base/scb/pcie@7d500000/pci@0,0/usb@0,0-1.1:1.0-0c45:6366 as cam0

```

Which sounds expected if USB is not supported...

> Please make sure that gstreamer can find the desired components 

Not used to gstreamer but looks fine to me. The problem may be, I realize this just now, is that the camera used with gstreamer is USB (checkout log up there) as the only one cam I have ever able to detect is USB

naushir, 2023-01-06:
> To sum up USB are not supported, and, CSI-MIPI can not be detected... The problem is that no CSI-MIPI cameras have never been detected (I realise that just now...) on my raspberry pi: is there some patch I could apply (guided by you) to check why this occurs?...

If you start from a freshly install [Raspberry Pi OS image](https://www.raspberrypi.com/software/), that should include everything needed to get any of the official Raspberry Pi cameras working out-of-the-box.

> > libcamera-apps provided by Raspberry Pi do not accept USB cameras.
> 
> Really?!.. If so this should be in top position in the doc... Didn't see it anywhere I looked. USB is a first entry point for almost everyone?!..

The expectations from the vast majority of our users is to seamlessly use one of our official Raspberry Pi cameras with libcamera-apps.

naushir, 2023-01-13:
I'm unclear if there is more to do for this issue on my end?  If not, I'll close this down shortly.

fghoussen, 2023-01-13:
No solution. Can be closed.

closed 2023-01-13 by fghoussen

kbingham, 2023-01-13:
Just to clarify.

Uvc is not supported by raspberry pi's "libcamera-apps" therefore libcamera-vid / libcamera-hello ...

But uvc is supported by cam/qcam/gstreamer.

The gstreamer plugin should work with uvc cameras and Raspberry pi cameras. If there is a fault with the gstreamer plugin gstlibcamerasrc, please raise that at bugs.libcamera.org.