raspberrypi/libcamera #40: installation guide that works

MazrimT, 2023-01-06:
I accidentally updated my pi to bullseye and thought, why not try the new libcamera instead of turning legacy back on that now says it's being depricated.

before all I had to do to code in python, which is the raspberry pi's go to language:

apt install ffmpeg
create virtual environment  (because that's how python should be done)
pip install ffmpeg-python
(other packages exist to do similar things like open-cv, but this was from my experience the easiest and most pythonic package)

and then I have everything usable in python for whatever I want to do.

Now I've spent a full day trying to figure out how to get basic "record a video" functionallity up and running.
You've made something simple into someting seemingly impossible and made it feel very much not like raspberry pi.

- libcamera can't be installed or used from a virtual environment
- picamera2 can't be installed with pip without first installing strange python libraries and then it cant be used in a virutal environment because it libcamera is not installed there.

This "upgrade" is a hot mess and I hope this will be something usable in the future, but for now I'm going back to legacy as basically every guide on the internet says to do instead of trying to use libcamera.

naushir, 2023-01-06:
Here are a couple of useful documents to look at if you re-consider using libcamera:

https://www.raspberrypi.com/documentation/computers/camera_software.html
https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf

closed 2023-01-06 by naushir

MazrimT, 2023-01-06:
Problem still exists, libcamera can not be used in a virtual environment by picamera2 since libcamera does not exist there.
libcamera can not be installed in a virtual environment since it's not a pip package.

https://www.raspberrypi.com/documentation/computers/camera_software.html litterally states that:
```
Python bindings for libcamera are currently in development.
Full working examples will be provided here.
```

On the whole it was a misstake for raspberry to switch to something so unfinnished.
Why not have this as a "you can switch to this library if you want as a beta" instead of force switching it with the new version of raspberry pi os?

tomba, 2023-01-07:
Not that it invalidates the point of this bug report, but I just want to mention that comparing ffmpeg-python to libcamera is not quite fair, as ffmpeg-python is "just" a python wrapper to the ffmpeg command, whereas libcamera bindings are "real" bindings to a C++ library.