raspberrypi/libcamera #41: no element 'libcamerasrc'

brianabouchard, 2023-02-13:
I'm running Ubuntu 22.04 and have successfully built libcamera and libcamera-apps from source. Both work flawlessly, except I'm unable to use gstreamer, which appears to be impacting my ability to use open-cv as well. The error I get whenever I run any variation of: `gst-launch-1.0 libcamerasrc`

Is: `WARNING: erroneous pipeline: no element "libcamerasrc"`

I confirmed in the meson build report that gstreamer was successfully recognized.

naushir, 2023-02-16:
When you do a ``ninja install``, can you confirm that the libcamerasrc element has been copied correctly to a system library path?

naushir, 2023-02-27:
Any updates on this one?  If not, I'll close this issue down for now.

naushir, 2023-02-28:
Closing this down now, please do re-open if this is still an issue.

closed 2023-02-28 by naushir

EwingKang, 2024-09-30:
I have the same issue. With the latest debian bookworm, I'm also unable to run the gstreamer example in https://www.raspberrypi.com/documentation/computers/camera_software.html

naushir, 2024-09-30:
I'm afraid we don't control Ubuntu's packaging flow - they create their our libcamera and accompanying packages.  Please speak to their support teams to help with this.

EwingKang, 2024-10-01:
It turns out that I have to explicitly installed the libcamera-gstreamer plugin through `sudo apt install gstreamer1.0-libcamera`
This is not mentioned in the [official document](https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-hello), but is mentioned in [this forum post](https://forums.raspberrypi.com/viewtopic.php?t=361253)
After the installation the command works properly now 
```bash
gst-launch-1.0 libcamerasrc camera-name="/base/axi/pcie@120000/rp1/i2c@80000/imx219@10" ! video/x-raw,colorimetry=bt709,format=NV12,width=720,height=480,framerate=30/1 ! jpegenc ! multipartmux ! tcpserversink host=0.0.0.0 port=8554
```
    
Also, the V4L2 elements `v4l2convert` and `v4l2h264enc` in the [official document](https://www.raspberrypi.com/documentation/computers/camera_software.html#libcamerasrc-gstreamer-element) does not exist for RPi 5.
```
WARNING: erroneous pipeline: no element "v4l2convert"
```
I believe it is because that RPi5 does not have a hardware encoder as mentioned in [this gstreamer issue](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/3157)
I'm now working on using x264enc for that capability.