raspberrypi/libcamera #42: Camera connected, but not available in libcamera on RPi4B running Ubuntu 22.04

lm782, 2023-02-15:
Not sure if I posted this in the right spot originally (libcamera-org/libcamera). Posting here as well. Will close whichever gets assigned or makes progress. 

Just got libcamera working on an RPi4B running Ubuntu 22.04 LTS, but "cam -l" does not list a camera available. (Confirmed the camera and hardware work with RPi OS and libcamera-hello prior to the build.)
From preliminary troubleshooting suspect it has something to do with permissions.

The initial error codes indicate libcamera (or the OS) recognized the camera (imx477), but had problems with dmaHeap.
"ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device", and
"ERROR RPI raspberrypi.cpp:1278 Failed to register camera imx477 10-001a: -12"

From reading about dmaHeap issues online I added my username to the 'video' group and no longer get those errors, but still no camera is available. Not sure what to make of the debug output below, other than things generally look good (no errors).

Any recommendations on what else to look into?
```
luis@RPi4-Ubuntu:~$ LIBCAMERA_LOG_LEVELS=*:DEBUG cam -l
[0:10:29.031297600] [3202] DEBUG IPAModule ipa_module.cpp:329 ipa_rkisp1.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so is signed
[0:10:29.031920457] [3202] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so'
[0:10:29.033633179] [3202] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so is signed
[0:10:29.034040170] [3202] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so'
[0:10:29.034515567] [3202]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+17-3f8bcc1a
[0:10:29.034992705] [3205] DEBUG Camera camera_manager.cpp:108 Starting camera manager
[0:10:29.052254924] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media2
[0:10:29.052679674] [3205] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media2 due to 5 missing dependencies
[0:10:29.058059409] [3205] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media2 found
[0:10:29.058322440] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-codec
[0:10:29.060125198] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media0
[0:10:29.060326675] [3205] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media0 due to 4 missing dependencies
[0:10:29.061321116] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media3
[0:10:29.061479539] [3205] DEBUG DeviceEnumerator device_enumerator_udev.cpp:95 Defer media device /dev/media3 due to 4 missing dependencies
[0:10:29.065747705] [3205] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media0 found
[0:10:29.065879906] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: bcm2835-isp
[0:10:29.069546567] [3205] DEBUG DeviceEnumerator device_enumerator_udev.cpp:320 All dependencies for media device /dev/media3 found
[0:10:29.069689675] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media3: bcm2835-isp
[0:10:29.070569564] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media1
[0:10:29.071008221] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: unicam
[0:10:29.072645574] [3205] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerISI'
[0:10:29.073274987] [3205] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerRPi'
[0:10:29.073481056] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "unicam"
[0:10:29.073627035] [3205] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "bcm2835-isp"
[0:10:29.073813771] [3205] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerRkISP1'
[0:10:29.074557088] [3205] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'SimplePipelineHandler'
[0:10:29.074890711] [3205] DEBUG Camera camera_manager.cpp:151 Found registered pipeline handler 'PipelineHandlerUVC'
Available cameras:
luis@RPi4-Ubuntu:~$ 
```

naushir, 2023-02-16:
You will need to speak to the Ubuntu support team about this.  They build and package up the libcamera libraries on their OS.  On the Raspberry Pi OS, we do the packaging, so all the dependences are available.

closed 2023-02-16 by naushir

lm782, 2023-02-17:
Thanks, naushir. In case anyone comes across this in the future, I could not find an Ubuntu libcamera page on GitHub and posted this question on launchpad.net.