raspberrypi/libcamera #45: Linking error during building for armv6l (RPi 1)

strec007, 2023-03-03:
Hi,
Since this library is missing in Void linux, I'm trying to package it. It builds for all supported platforms (arm64, armv7) including the musl variants, except for armv6l. Is there anything special with RPi 1 and armv6l which I need to consider? Thank you. The error message follows.

```
arm-linux-gnueabihf-c++  -o src/libcamera/libcamera.so.0.0.4 src/libcamera/libcamera.so.0.0.4.p/meson-generated_.._proxy_raspberrypi_ipa_proxy.cpp.o src/libcamera/libcamera.so.0.0.4.p/meson-generated_.._proxy_rkisp1_ipa_proxy.cpp.o src/libcamera/libcamera.so.0.0.4.p/meson-generated_.._control_ids.cpp.o src/libcamera/libcamera.so.0.0.4.p/meson-generated_.._property_ids.cpp.o src/libcamera/libcamera.so.0.0.4.p/meson-generated_.._version.cpp.o src/libcamera/libcamera.so.0.0.4.p/meson-generated_.._ipa_pub_key.cpp.o src/libcamera/libcamera.so.0.0.4.p/bayer_format.cpp.o src/libcamera/libcamera.so.0.0.4.p/byte_stream_buffer.cpp.o src/libcamera/libcamera.so.0.0.4.p/camera.cpp.o src/libcamera/libcamera.so.0.0.4.p/camera_controls.cpp.o src/libcamera/libcamera.so.0.0.4.p/camera_lens.cpp.o src/libcamera/libcamera.so.0.0.4.p/camera_manager.cpp.o src/libcamera/libcamera.so.0.0.4.p/camera_sensor.cpp.o src/libcamera/libcamera.so.0.0.4.p/camera_sensor_properties.cpp.o src/libcamera/libcamera.so.0.0.4.p/color_space.cpp.o src/libcamera/libcamera.so.0.0.4.p/controls.cpp.o src/libcamera/libcamera.so.0.0.4.p/control_serializer.cpp.o src/libcamera/libcamera.so.0.0.4.p/control_validator.cpp.o src/libcamera/libcamera.so.0.0.4.p/converter.cpp.o src/libcamera/libcamera.so.0.0.4.p/delayed_controls.cpp.o src/libcamera/libcamera.so.0.0.4.p/device_enumerator.cpp.o src/libcamera/libcamera.so.0.0.4.p/device_enumerator_sysfs.cpp.o src/libcamera/libcamera.so.0.0.4.p/fence.cpp.o src/libcamera/libcamera.so.0.0.4.p/formats.cpp.o src/libcamera/libcamera.so.0.0.4.p/framebuffer.cpp.o src/libcamera/libcamera.so.0.0.4.p/framebuffer_allocator.cpp.o src/libcamera/libcamera.so.0.0.4.p/geometry.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipa_controls.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipa_data_serializer.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipa_interface.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipa_manager.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipa_module.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipa_proxy.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipc_pipe.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipc_pipe_unixsocket.cpp.o src/libcamera/libcamera.so.0.0.4.p/ipc_unixsocket.cpp.o src/libcamera/libcamera.so.0.0.4.p/mapped_framebuffer.cpp.o src/libcamera/libcamera.so.0.0.4.p/media_device.cpp.o src/libcamera/libcamera.so.0.0.4.p/media_object.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_handler.cpp.o src/libcamera/libcamera.so.0.0.4.p/pixel_format.cpp.o src/libcamera/libcamera.so.0.0.4.p/process.cpp.o src/libcamera/libcamera.so.0.0.4.p/pub_key.cpp.o src/libcamera/libcamera.so.0.0.4.p/request.cpp.o src/libcamera/libcamera.so.0.0.4.p/source_paths.cpp.o src/libcamera/libcamera.so.0.0.4.p/stream.cpp.o src/libcamera/libcamera.so.0.0.4.p/sysfs.cpp.o src/libcamera/libcamera.so.0.0.4.p/transform.cpp.o src/libcamera/libcamera.so.0.0.4.p/v4l2_device.cpp.o src/libcamera/libcamera.so.0.0.4.p/v4l2_pixelformat.cpp.o src/libcamera/libcamera.so.0.0.4.p/v4l2_subdevice.cpp.o src/libcamera/libcamera.so.0.0.4.p/v4l2_videodevice.cpp.o src/libcamera/libcamera.so.0.0.4.p/yaml_parser.cpp.o src/libcamera/libcamera.so.0.0.4.p/converter_converter_v4l2_m2m.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_imx8-isi_imx8-isi.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_raspberrypi_delayed_controls.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_raspberrypi_dma_heaps.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_raspberrypi_raspberrypi.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_raspberrypi_rpi_stream.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_rkisp1_rkisp1.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_rkisp1_rkisp1_path.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_simple_simple.cpp.o src/libcamera/libcamera.so.0.0.4.p/pipeline_uvcvideo_uvcvideo.cpp.o -L/usr/arm-linux-gnueabihf/usr/lib -flto -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libcamera.so.0.0.4 -Wno-psabi -Wshadow -include /builddir/rpi-libcamera-20230303/build/config.h -Wl,-z,relro -Wl,-z,now -Wl,--as-needed '-Wl,-rpath,$ORIGIN/base:/usr/arm-linux-gnueabihf/usr/lib32:/' -Wl,-rpath-link,/builddir/rpi-libcamera-20230303/build/src/libcamera/base -Wl,-rpath-link,/usr/arm-linux-gnueabihf/usr/lib32 src/libcamera/base/libcamera-base.so.0.0.4 /usr/arm-linux-gnueabihf/usr/lib32/libgnutls.so /usr/arm-linux-gnueabihf/usr/lib/libyaml.so -Wl,--end-group
/usr/lib/gcc/arm-linux-gnueabihf/12.2.0/../../../../arm-linux-gnueabihf/bin/ld: /tmp/ccmihFB1.ltrans9.ltrans.o: in function `libcamera::V4L2BufferCache::V4L2BufferCache(std::vector<std::unique_ptr<libcamera::FrameBuffer, std::default_delete<libcamera::FrameBuffer> >, std::allocator<std::unique_ptr<libcamera::FrameBuffer, std::default_delete<libcamera::FrameBuffer> > > > const&)':
<artificial>:(.text+0x2da0): undefined reference to `__atomic_fetch_add_8'
/usr/lib/gcc/arm-linux-gnueabihf/12.2.0/../../../../arm-linux-gnueabihf/bin/ld: /tmp/ccmihFB1.ltrans9.ltrans.o: in function `libcamera::V4L2BufferCache::get(libcamera::FrameBuffer const&)':
<artificial>:(.text+0x30c8): undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
[161/182] Compiling C++ object src/ipa/rkisp1/ipa_rkisp1.so.p/algorithms_gsl.cpp.o
[162/182] Compiling C++ object src/ipa/rkisp1/ipa_rkisp1.so.p/rkisp1.cpp.o
[163/182] Compiling C++ object src/apps/common/libapps.a.p/image.cpp.o
[164/182] Compiling C++ object src/ipa/rkisp1/ipa_rkisp1.so.p/algorithms_filter.cpp.o
[165/182] Compiling C++ object src/ipa/rkisp1/ipa_rkisp1.so.p/algorithms_lsc.cpp.o
[166/182] Compiling C++ object src/apps/common/libapps.a.p/stream_options.cpp.o
[167/182] Compiling C++ object src/apps/common/libapps.a.p/options.cpp.o
ninja: build stopped: subcommand failed.
```

kbingham, 2023-03-03:
Could be related to needing or not linking against libatomic? But I though we already handled that in libcamera.

strec007, 2023-03-04:
> Could be related to needing or not linking against libatomic? But I though we already handled that in libcamera.

You are absolutely right, sir. All builds work now. Thank you.

closed 2023-03-04 by strec007