raspberrypi/libcamera #48: ctt tool unable to read exif tags

tewarid, 2023-03-13:
ctt tool is unable to read exif tags from raw image created using `libcamera-still -o filename.dng`...

```log
Traceback (most recent call last):
  File "/home/dkt/.local/lib/python3.9/site-packages/pyexiv2/metadata.py", line 195, in _get_exif_tag
    return self._tags['exif'][key]
KeyError: 'Exif.SubImage1.ImageWidth'
```

naushir, 2023-03-14:
Adding @davidplowman 

> ctt tool is unable to read exif tags from raw image created using `libcamera-still -o filename.dng`...

Can you clarify the exact command used?  The above command line will capture a JPEG file with the name `filename.dng`, not an actual DNG file.  To get a DNG file, you need to run with the following command:

```
`libcamera-still -o filename.jpg-r`
```
This will generate `filename.jpg` and `filename.dng`.

naushir, 2023-03-14:
``libcamera-raw`` will not write dng files.  You should continue using `libcamera-still`.  I'll close this issue down now.  If you have any further problems, feel free to re-open.

closed 2023-03-14 by naushir