raspberrypi/libcamera #49: Error running ctt tool

tewarid, 2023-03-16:
I am trying out ctt tool using off-the-shelf v1 camera, but am unable to run it to completion due to error `ValueError: On entry to DLASCL parameter number 4 had an illegal value`

```log
$ python3 ../libcamera/utils/raspberrypi/ctt/ctt.py  -c ctt_config.json -i ov5647 -o ov5647.json

Loading images from ov5647/
Files found: 6

Loading image: 6150K_130l.dng
Locating macbeth chart
Macbeth chart located successfully
Confidence: 0.834
Image loaded successfully!

Loading image: 6150K_130l.jpg

ERROR:
No Broadcom header found
DISCARDED

Loading image: alsc_6150K_1.dng
Image loaded successfully!

Loading image: alsc_6150K_1.jpg

ERROR:
No Broadcom header found

DISCARDED

Loading image: alsc_6150K_1u.dng
Image loaded successfully!

Loading image: alsc_6150K_1u.jpg

ERROR:
No Broadcom header found

DISCARDED

Finished loading images

STARTING CALIBRATIONS

Starting ALSC calibration
Finished ALSC calibrations

Starting GEQ calibrations
Finished GEQ calibrations

Starting LUX calibrations
Finished LUX calibrations

Starting NOISE calibrations
Finished NOISE calibrations

Starting AWB calibration
Traceback (most recent call last):
  File "/home/dkt/ctt/../libcamera/utils/raspberrypi/ctt/ctt.py", line 837, in <module>
    run_ctt(json_output, directory, config, log_output)
  File "/home/dkt/ctt/../libcamera/utils/raspberrypi/ctt/ctt.py", line 801, in run_ctt
    Cam.awb_cal(greyworld, do_alsc_colour)
  File "/home/dkt/ctt/../libcamera/utils/raspberrypi/ctt/ctt.py", line 309, in awb_cal
    awb_out = awb(self, cal_cr_list, cal_cb_list, plot)
  File "/home/dkt/libcamera/utils/raspberrypi/ctt/ctt_awb.py", line 95, in awb
    a, b, c = np.polyfit(rbs_hat[0], rbs_hat[1], 2)
  File "<__array_function__ internals>", line 5, in polyfit
  File "/usr/lib/python3/dist-packages/numpy/lib/polynomial.py", line 629, in polyfit
    c, resids, rank, s = lstsq(lhs, rhs, rcond)
  File "<__array_function__ internals>", line 5, in lstsq
  File "/usr/lib/python3/dist-packages/numpy/linalg/linalg.py", line 2306, in lstsq
    x, resids, rank, s = gufunc(a, b, rcond, signature=signature, extobj=extobj)
ValueError: On entry to DLASCL parameter number 4 had an illegal value
```

davidplowman, 2023-03-20:
Hmm, I have to say I don't think I've seen that before. 

@twarid I thought maybe it was falling over because there appears to be only a single non-ALSC file, but that seemed to work OK for me. 

Can I assume you've downloaded the latest version of all the software? It seems to be that call to `np.polyfit` that's going wrong, would you be able to print out the input arguments just beforehand (`rbs_hat[0]` and `rbs_hat[1]`)? Maybe also double-check your numpy version - mine claims to be 1.24.2.

tewarid, 2023-03-20:
I think I've got the latest version of most packages - I did the following and don't see anything significant to update

```log
$ sudo apt update
...
$ apt list --upgradable
Listing... Done
linux-libc-dev/stable 1:1.20230317-1 arm64 [upgradable from: 1:1.20230306-1]
raspberrypi-bootloader/stable 1:1.20230317-1 arm64 [upgradable from: 1:1.20230306-1]
raspberrypi-kernel-headers/stable 1:1.20230317-1 arm64 [upgradable from: 1:1.20230306-1]
raspberrypi-kernel/stable 1:1.20230317-1 arm64 [upgradable from: 1:1.20230306-1]
vcdbg/stable 1:1.20230317-1 armhf [upgradable from: 1:1.20230306-1]
xserver-common/stable,stable 2:1.20.11-1+rpt2+deb11u5 all [upgradable from: 2:1.20.11-1+rpt1+deb11u5]
xserver-xorg-core/stable 2:1.20.11-1+rpt2+deb11u5 arm64 [upgradable from: 2:1.20.11-1+rpt1+deb11u5]
xwayland/stable 2:1.20.11-1+rpt2+deb11u5 arm64 [upgradable from: 2:1.20.11-1+rpt1+deb11u5]
```

Curiously, numpy version listed here is 1.19.5

```log
$ apt info python3-numpy
Package: python3-numpy
Version: 1:1.19.5-1
```

Maybe I should try to getting numpy using pip3 instead?

davidplowman, 2023-03-20:
Those NaNs are definitely still wrong. When I run it (even with a single non-ALSC file), I have proper numbers there. Though note that ultimately you probably will need more than one image (even though it doesn't seem like the actual problem).

So I guess we need to find out why that 6150K_130l.dng file is causing trouble. Would you be able to post a link so that I can download it and try it? Thanks.

davidplowman, 2023-03-20:
Can you say anything about how you got that DNG file? It's claiming a black level of 64 which is incorrect for the ov5647. It should have a black level of 16 (out of the full 10-bit range). Let's start by trying to fix that and see if things improve. Thanks!

tewarid, 2023-03-20:
My bad - I should be using 1024. I will try with that.

davidplowman, 2023-03-20:
You should be able to use the standard ov5647.json - all you need is the correct black level and reasonable exposure, which you should get. All the other stuff (to do with colours, lens shading etc.) has no effect on the DNG file.

tewarid, 2023-03-20:
That did it! Thanks!

closed 2023-03-20 by tewarid