raspberrypi/libcamera #50: IPC/IPA error while running gst-launch pipeline

KunalAivero, 2023-03-20:
Hi,
I have attached Rpi-Hq-Camera to R.Pi4 board running Ubuntu 22.04.1.
The dmesg shows this
```
[    8.007776] imx477 10-001a: Device found is imx477
[    8.008300] imx477 10-001a: Consider updating driver imx477 to match on endpoints
```
And When I try to run a gst-launch pipeline to view the data from libcamerasrc I get this error
```
->gst-launch-1.0 libcamerasrc ! videoscale ! videoconvert ! ximagesinkSetting pipeline to PAUSED ...
[0:03:13.831499342] [6843]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2
[0:03:13.885655931] [6845]  INFO RPI raspberrypi.cpp:1386 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[0:03:14.105665133] [6849]  INFO Camera camera.cpp:1026 configuring streams: (0) 1280x1080-XRGB8888
[0:03:14.106529363] [6845]  INFO RPI raspberrypi.cpp:766 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:03:16.169349543] [6845] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout!
[0:03:16.169469521] [6845] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[0:03:16.169521334] [6845] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:238 Failed to call start
[0:03:16.841559096] [6845] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Connection refused
[0:03:16.841782348] [6845] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
[0:03:16.841884844] [6845] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:555 Failed to call signalQueueRequest
[0:03:16.842072116] [6845] ERROR IPCUnixSocket ipc_unixsocket.cpp:191 Failed to send: Transport endpoint is not connected
[0:03:16.842175446] [6845] ERROR IPCPipe ipc_pipe_unixsocket.cpp:80 Failed to call async
[0:03:16.842252406] [6845] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:593 Failed to call signalIspPrepare
```
At this point the pipeline is fully stalled and I do a Ctrl+C to end it.

Can you suggest what may be causing the above IPC/IPA errors.

naushir, 2023-03-20:
Just to confirm, you are using Ubuntu 22.04 OS environment?  If so, you will have to get in touch with the Ubuntu support team to help with this.  They package libcamera separately from what we do for Raspberry Pi OS.

KunalAivero, 2023-03-20:
hi @naushir , thanks for your quick reply. I forgot to point out that I have built libcamera from source from this repo.

naushir, 2023-03-21:
In which case, I expect it's a problem with your build environment.  Specifically, you need to install the following packages:
```
sudo apt install -y libgnutls28-dev openssl
```

then redo the meson configure to ensure your IPA is not running in isolated mode.

naushir, 2023-03-24:
Unfortunately my knowledge of gstreamer is mostly non-existent.  I would suggest posting to the libcamera developers mailing list (libcamera-devel@lists.libcamera.org) to see if somebody there can help you.

jmondi, 2023-03-24:
Hello

> Hi @naushir I have done some further tests
> 
> (1) The openssl was already installed, and meson was configured not to run IPA in isolated mode. But anyway I installed gnutls as well. It does not matter, as build system checks for openssl first.
> 
> (2) I am able to capture binary images with `cam` application. So as such libcamera installations seems ok.
> 
> (3) I have also built libcamera-apps and tested the image capture with libcamera-jpeg and video capture with libcamera-vid is ok. So, libcamera installation is ok, and libcamera-apps work ok.
> 
> (4) so problem relates to running gstreamer pipelines only. Is there any special configuration I may require to get libcamera working with gstreamer support on raspberry pi?
>

Could you please re-run with LIBCAMERA_LOG_LEVELS=0 to increase the debug logs ? With additional debug could you please check if the IPA module signature verification succeeds or not ? You should be able to see:

`IPA module ... signature is valid/not valid`

 
> (5) Note that I have also built libcamera on x86 machine with gstreamer and gst-launch pipelines are working ok. Only difference I see is that on x86 machine IPA IPU3 is used.
> 
> If you can provide some suggestions I can try them out. I am running out of ideas as I am not well versed with libcamera IPA functionality. I would appreciate if you could give me some pointers to try.
> 
> If there is any other forum where this question may be more relevant. Note that the gstreamer support is working ok on x86 machine with ubuntu and it only fails on raspberry Pi ubuntu.

KunalAivero, 2023-03-28:
hi @jmondi 
Thanks for your inputs. Signal check indeed fails.
When I run `cam -l` with `LIBCAMERA_LOG_LEVELS=*:DEBUG", first all signed binaries are reported
```
0:58:16.467801343] [7881] DEBUG IPAModule ipa_module.cpp:329 ipa_rkisp1.so: IPA module /home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_rkisp1.so is signed
[0:58:16.469146747] [7881] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_rkisp1.so'
[0:58:16.470087583] [7881] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_rpi.so is signed
[0:58:16.470749752] [7881] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_rpi.so'
[0:58:16.472083788] [7881] DEBUG IPAModule ipa_module.cpp:329 ipa_vimc.so: IPA module /home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_vimc.so is signed
[0:58:16.472793727] [7881] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_vimc.so'
```
but later the signature is reported to be `not valid`
```
[0:58:16.505046754] [7882] DEBUG IPAManager ipa_manager.cpp:303 IPA module /home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_rpi.so signature is not valid
[0:58:16.505308454] [7882] DEBUG IPAProxy raspberrypi_ipa_proxy.cpp:45 initializing raspberrypi proxy: loading IPA from /home/kunal/.conan/data/gst-libcamera/0.0.4/_/_/package/b4d2cb36ad454869cf57844e2325abef18b7dc73/lib/libcamera/ipa_rpi.so
```
I am using conan to build and package the libcamera. I am investigating the problem. If you have any further suggestions please let me know.

jmondi, 2023-03-28:
I have no idea how conan interfaces to meson, but be aware that the install phase in meson strips the binaries, rendering the signatures unusable. There's a re-sign step visible at the end of src/ipa/meson.build which is run at `ninja install` time. You might want to check if anything like this is required/doable with conan

kbingham, 2023-03-28:
If conan is creating a package for libcamera, then indeed it is likely to be stripping the built objects and causing this issue.
@KunalAivero Can you point to the build scripts that you are using for this package please?

https://salsa.debian.org/multimedia-team/libcamera/-/commit/7871e6737d82e76942aaa3b6f56a2f03f28b1d91 might be a useful reference for you.

KunalAivero, 2023-03-29:
Hi @kbingham thanks for you suggestions. I am out of office this week. I will try this out next week and get back to you.

kbingham, 2023-04-19:
@KunalAivero How did you get on here, did you resolve your issue ?

dgsiegel, 2023-04-19:
I've just ran into a similar issue on a RPi Zero built with buildroot. From what I could figure out was that the GnuTLS  signature path is broken and you need **both** GnuTLS **and** libcrypto/openssl. By adding both dependencies at build time, the signature will be found and valid.

KunalAivero, 2023-04-19:
@kbingham I haven't yet managed to fix the conan build file, I am planning a few more tests tomorrow. will update you tomorrow.

KunalAivero, 2023-04-20:
Closing this issue as it has been fixed. Problem was caused by stripping the signed binaries. Resolution is to re-sign the binaries after stripping.

closed 2023-04-20 by KunalAivero

kbingham, 2023-04-20:
@KunalAivero Could you share any reference to your Conan package scripts/build please? Are they going to be merged 'into' Conan package manager at all?

KunalAivero, 2023-04-21:
@kbingham our devops team had added customised post_package hooks which stripped the debug information from binaries. Without the customised hooks everything should work fine. Now we have modified the customised post_package hooks to check for presence of .pem and .sign files and if those exist, we re-sign the stripped binaries