raspberrypi/libcamera #52: Libcamera failed to get frames from OV9281 camera

tamano123, 2023-03-21:
Hello, 
I have an OV9281 camera from Inno-maker. I used it on my Raspberry Pi 4 Model B Rev 1.5 with the latest Raspberry OS (Release date: February 21st 2023, System: 64-bit).

In order to get high framerate i use the inno-maker driver :
1. I could check the camera

```
pi@raspberrypi:~ $ ls /dev/video0
/dev/video0
```

2. I could check the framerate with v4l2-ctl 
```
pi@raspberrypi:~ $ cat /sys/module/inno_mipi_ov9281/parameters/sensor_mode
9
pi@raspberrypi:~ $ v4l2-ctl --stream-mmap --stream-count=-1 -d /dev/video0 --stream-to=/dev/null
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 253.10 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 253.10 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 253.10 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<^C
```

3. I could use VLC to visualize the camera stream
4.  But i failed when using Libcamera :

```
pi@raspberrypi:~ $ libcamera-hello -t 0
Made X/EGL preview window
[0:00:53.575375948] [1267]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:00:53.597454207] [1268] ERROR CameraSensor camera_sensor.cpp:125 'inno_mipi_ov9281 10-0060': No image format found
[0:00:53.597697522] [1268] ERROR RPI raspberrypi.cpp:1277 Failed to register camera inno_mipi_ov9281 10-0060: -22
ERROR: *** no cameras available ***
```

Can you help me please ?
Thank you

6by9, 2023-03-21:
> In order to get high framerate i use the inno-maker driver :

Sorry, we only support the driver shipped as part of our kernel.

What high frame rate mode are you after? The standard kernel driver supports
- 10 bit output at 1280x800@120fps, 1280x720@144fps, 640x400@260fps
- 8 bit output at 1280x800@150fps, 1280x720@180fps, and 640x400@320fps.
Those are well up into H264 level 4.2, so performance is not guaranteed.

tamano123, 2023-03-21:
Thank you for your answser.

1. I understand that you don't support innomaker's driver but, since VLC could get the camera stream by using ATSC 8-VSB (with innomaker's driver), Don't you think that libcamera (or Picamera2) could do it too ?
2. 
> What high frame rate mode are you after? The standard kernel driver supports
> 
>     10 bit output at 1280x800@120fps, 1280x720@144fps, 640x400@260fps
>     8 bit output at 1280x800@150fps, 1280x720@180fps, and 640x400@320fps.
>     Those are well up into H264 level 4.2, so performance is not guaranteed.

Since i am newbie to vision development, do you have samples using Picamera2 in order to get 10bit output 640x400@260fps or 8bit 640x400@320fps ?
Thank you

6by9, 2023-03-21:
If v4l2-ctl and VLC are getting frames without manual media-ctl commands, then the Media Controller API is not enabled for the driver. libcamera requires Media Controller - https://forums.raspberrypi.com/viewtopic.php?t=322076

@davidplowman can you advise on Picamera2
I'm using ```libcamera-vid --width 640 --height 400 --framerate 240 --denoise off -o foo.h264 --save-pts foo.pts``` 
For some reason 250fps stalls after a few seconds. 259fps drops into repeated timeouts. 260fps just stalls. Using v4l2-ctl it seems to stream fine at those rates.

davidplowman, 2023-03-21:
Picamera2 works if and only if libcamera works. So you will need a compatible driver that supports these framerates. Note that these are quite high framerates so you would have to experiment to find out what is achievable in practice.

tamano123, 2023-03-21:
Thank you for your advices

closed 2023-03-21 by tamano123