raspberrypi/libcamera #53: libcamera module compilation for Python version 3.11

hugokernel, 2023-03-22:
Is libcamera compatible with Python 3.11 ?

The Python version 3.11 is installed on a Raspberry with `altinstall` flag, so, both Python 3.9 and 3.11 works together.

I can create virtual env, install picamera2 from pip (`pip3.11`), everything is ok, but:

```shell
Python 3.11.2 (main, Mar 16 2023, 15:19:49) [GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import picamera2
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/tmp/env311/lib/python3.11/site-packages/picamera2/__init__.py", line 1, in <module>
    import libcamera
ModuleNotFoundError: No module named 'libcamera'
```

The error is normal, and it is the subject of my question: how can I specify to libcamera to compile for a specific Python version, 3.11 in this case ?

Thx

naushir, 2023-03-23:
This is a known problem, as the libcamera library files are not bundled with pip.  These have to be installed separately.  I believe the libcamera team are aware of this and working on a solution.

In the meantime, you will have to run picamera2 from outside of a virtual environment.

closed 2023-03-23 by naushir

hugokernel, 2023-03-23:
Thanks for your answer but my problem is not about virtual environments.

> This is a known problem, as the libcamera library files are not bundled with pip. These have to be installed separately. I believe the libcamera team are aware of this and working on a solution.
> 
> In the meantime, you will have to run picamera2 from outside of a virtual environment.

`libcamera` / `picamera2` works in a virtual environment if it is created with the `--system-site-packages` argument 
(I know it, that's how I use it with the version of Python 3.9), in this case, if a module is not in the venv, it will look for it in the system packages (this is the case with `libcamera`).
However, the module must be installed for the version of Python you want and that's the subject of this ticket.

I can compile `libcamera` for the default version of Python installed on the system but I would like to do it for a specific version (here 3.11) and I didn't find any information about it.

How can I compile libcamera for a specific Python version, 3.11 in this case ?

naushir, 2023-03-23:
@davidplowman might be able to help more here.

> How can I compile libcamera for a specific Python version, 3.11 in this case ?

I don't know if it is possible to do what you want here.  The libcamera library meson configuration only really specifics if pycamera is to be built or not.  There are no further configuration options for the python bindings.

davidplowman, 2023-03-23:
I'm afraid I'm not too sure. Is it possible to point your python and python3 executables at your new version? I don't know how meson decides what version of Python to use, but I would guess if you can make your new one the default then it might "just work"? Sorry not to be more help.

hugokernel, 2023-03-29:
> Is it possible to point your python and python3 executables at your new version?

I have tried but without success. libcamera always compile for Python 3.9