raspberrypi/libcamera #54: build error

neolu, 2023-03-28:
..............................................................................................................................+++++
[42/185] Generating src/libcamera/ipa_pub_key_cpp with a custom command
writing RSA key
[184/185] Installing files.
Installation failed due to insufficient permissions.
Attempting to use polkit to gain elevated privileges...
==== AUTHENTICATING FOR org.freedesktop.policykit.exec ===
Authentication is needed to run `/usr/bin/python3' as the super user
Authenticating as: ,,, (dev)
Password:
polkit-agent-helper-1: error response to PolicyKit daemon: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: No session for cookie
==== AUTHENTICATION FAILED ===
Error executing command as another user: Not authorized

This incident has been reported.
FAILED: meson-internal__install
/home/dev/.local/bin/meson install --no-rebuild

naushir, 2023-03-28:
You are going to have to provide **much** more context as to what you are doing in order for us to help you.  Please also follow the build instructions posted [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#building-libcamera-and-libcamera-apps).

naushir, 2023-03-31:
Any update on this issue?  If not, I'll close this shortly.

naushir, 2023-04-03:
Closing now.  Feel free to re-open if you are still having problems.

closed 2023-04-03 by naushir

pinchartl, 2023-04-04:
> A quick glance I would say you're trying to call "ninja install" without root permissions.
> 
> Try 'sudo ninja install'
> 
> Make sure you run 'sudo ldconfig' after installing as well.

meson (called by `ninja install`) is smart enough and uses polkit to get elevated privileges. That's the officially supported way to install to system directories. Running the installation command with `sudo` is *not* recommended, as it can lead to all kind of issues, especially when the user and system versions of meson are different.