raspberrypi/libcamera #56: Timeout setting

edward-ardu, 2023-05-12:
I noticed on https://github.com/kbingham/libcamera/tree/master/src/libcamera/pipeline/rpi/common that there have been many updates to the code. 
There are some meaningful changes that could potentially allow me to update the timeout setting. Would it be possible to incorporate these changes?
[image]

I'd like to explain why this function is important to me. I'm working on a project involving external triggering, and I need the timeout to be unlimited. Otherwise, the camera will keep restarting, which would significantly impact the project's progress. This function is critical to the success of my project, and I would greatly appreciate any support you can offer.

naushir, 2023-05-13:
If you are looking to increase the timeout, you do have the option of setting a custom value in the config file.  You cannot disable it entirely, but setting it to a very large values (e.g. hours). Would that be sufficient for your needs?

naushir, 2023-05-16:
You can provide a config file to libcamera, and in that file you can set the timeout through the ``camera_timeout_value_ms`` variable.  An example file is located at ``/usr/share/libcamera/pipeline/rpi/vc4/example.yaml`` (or possibly ``/usr/local/share/libcamera/pipeline/rpi/vc4/example.yaml`` if you did your own build/install package).

To use the config file, set the ``LIBCAMERA_RPI_CONFIG_FILE`` environment variable to the location of the file.

naushir, 2023-05-19:
Is there anything further to look at here?  If not, I'll close this down shortly.

closed 2023-05-22 by naushir