raspberrypi/libcamera #57: how to correctly generate python bindings?

MattBlack85, 2023-05-18:
Hi,

I am trying to generate the python bindings (I read those are experimental), first I changed `meson_options.txt` enabling the bindings, then I run the build accordingly with the README and I see something has been generated, I have a `py` folder in the `src` dir which structure is the following

```
.
├── cam
│   ├── cam.py
│   ├── cam_kms.py
│   ├── cam_null.py
│   ├── cam_qt.py
│   ├── cam_qtgl.py
│   ├── gl_helpers.py
│   └── helpers.py
├── examples
│   ├── simple-cam.py
│   ├── simple-capture.py
│   └── simple-continuous-capture.py
├── libcamera
│   ├── __init__.py
│   ├── build
│   │   ├── meson-info
│   │   ├── meson-logs
│   │   │   └── meson-log.txt
│   │   └── meson-private
│   │       └── meson.lock
│   ├── gen-py-controls.py
│   ├── gen-py-formats.py
│   ├── meson.build
│   ├── py_camera_manager.cpp
│   ├── py_camera_manager.h
│   ├── py_controls_generated.cpp.in
│   ├── py_enums.cpp
│   ├── py_formats_generated.cpp.in
│   ├── py_geometry.cpp
│   ├── py_helpers.cpp
│   ├── py_helpers.h
│   ├── py_main.cpp
│   ├── py_main.h
│   ├── py_properties_generated.cpp.in
│   └── utils
│       ├── MappedFrameBuffer.py
│       └── __init__.py
└── meson.build

```

I tried to cpy over the `libcamera` folder into my `sites-packages` one but when importing libcamera I have an error which I believe it comes for the `__init__.py` from the bindings.
`__init__.py` looks like 
```
# SPDX-License-Identifier: LGPL-2.1-or-later
# Copyright (C) 2022, Tomi Valkeinen <tomi.valkeinen@ideasonboard.com>

from ._libcamera import *

```

What am I missing here?

naushir, 2023-05-19:
I'm not an expert on the python setup, but what I do is install libcamera after building (``sudo meson install``), and set my ``PYTHONPATH`` env variable to point to the right location for the libcamera packages (``PYTHONPATH=/usr/local/lib/aarch64-linux-gnu/python3.9/site-packages/`` in my case).  This seems to work correctly for me.

@davidplowman might be able add more details.

MattBlack85, 2023-05-19:
hi @naushir and thanks for the tip, in which folder you run `sudo meson install`?

naushir, 2023-05-19:
I run that from by libcamera build directory.  Alternatively you can do ``sudo meson install -C <build_direcotory>``.

naushir, 2023-05-19:
When you do the ``meson install``, you will get a log output and you can see exactly where the script has install the python libraries for your ``PYTHONPATH`` env variable.

Adityasrinivas24, 2023-05-20:
This worked for me. First install all dependencies mentioned for python bindings in README.rst.

Enter the directory
cd libcamera

meson setup build -Dpycamera=enabled 

sudo ninja -C build install 

This should build the python bindings.

naushir, 2023-05-24:
Closing this now, if you have any further issue, fell free to reopen.

closed 2023-05-24 by naushir