raspberrypi/libcamera #58: gst-libcamerasrc : basic parameters missing

Consti10, 2023-05-28:
Hello,
We've now had a lot of users complaining that they cannot
Flip / rotate and or change the brightness, usw when using
gst-libcamerasrc (functionalities that were available in legacy
gst-rpicamsrc (mmal). Is there any plans to add support for changing
those parameters ?
If no, can you hint me at possible problems that might be hard to
overcome and hinder an implementation using either caps or gst pipeline
Parameters ?
(Second one obviously would be nicer, since it would allow
Changing the brightness for example without a restart)

6by9, 2023-05-28:
gst-libcamerasrc is part of the main [libcamera project](https://libcamera.org/). This repo is only the Raspberry Pi fork so that we can handle any parts that haven't been or can't be merged into the main project.

Already logged in the upstream bug tracker - https://bugs.libcamera.org/show_bug.cgi?id=111
libcamera is all open source, so contributions are always welcome!

closed 2023-05-28 by 6by9

Consti10, 2023-05-30:
While I am quite disappointed of how less responsibility the rpi foundation takes here (Providing the same feature set with gst-libcamerasrc as there was in gst-rpicamsrc (mmal) would have been nice) I can say that I started working on this (not hard to do, but annoying time consuming stuff).

6by9, 2023-05-30:
Raspberry Pi didn't write gst-rpicamsrc - it was a third party GST element that was then upstreamed into the main GStreamer project https://github.com/thaytan/gst-rpicamsrc
Why should Raspberry Pi take responsibility for replicating all third party libraries?

Nicolas Dufresne is one of the main GStreamer contributors and is the original author of gst-libcamerasrc. There is ongoing development of it on the libcamera mailing list (linked from https://libcamera.org/contributing.html), including work from Nicolas such as sensor mode selection https://lists.libcamera.org/pipermail/libcamera-devel/2023-March/037277.html

Consti10, 2023-05-30:
In my opinion, if someone like rpi foundation makes the decision to deprecate something (mmal / gst-rpicamsrc) you'd expect similar features and performance to be available by the replacement already or come in the future.

naushir, 2023-05-30:
@Consti10 as @6by9 mentioned, [gst-rpicamsrc](https://github.com/thaytan/gst-rpicamsrc) is a third party library that Raspberry Pi Ltd has no involvement with.  You have been provided a link to the libcamerasrc gstreamer element issue tracker, please post your requests to that.

There is nothing more to say about this issue. Locking now.