raspberrypi/libcamera #59: Fatal usage in IPA module

kbingham, 2023-06-05:
There's a user report of setting controls causing the whole process to crash:

https://github.com/ayufan/camera-streamer/issues/67:
```
Getting this error and service restart when try to set various options (not all, but too few)

Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: RESCALLER:STREAM: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: RESCALLER:VIDEO: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: SNAPSHOT: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: STREAM: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: device/v4l2/device_options.c: VIDEO: The 'AeExposureMode=2' was failed to find.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: util/http/http.c: HTTP8080/6: Client disconnected 192.168.2.6.
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: [518:41:50.377648261] [44019] FATAL RPiAgc agc.cpp:540 No exposure profile long
Jun 02 13:21:17 wgrpinlcam01 camera-streamer[44004]: Backtrace:
Jun 02 13:21:17 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Main process exited, code=killed, status=6/ABRT
Jun 02 13:21:17 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Failed with result 'signal'.
Jun 02 13:21:17 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Consumed 1min 32.532s CPU time.
Jun 02 13:21:27 wgrpinlcam01 systemd[1]: camera-streamer-raspi-v3-12MP.service: Scheduled restart job, restart counter is at 5.
```

The Fatals in void Agc::housekeepConfig() seems to be a bit painful. Is there anything we can do to reduce this error level? Or is this really a fatal error, and an issue in the users system configuration or such?

naushir, 2023-06-06:
This error message suggests to me that the user is using a custom camera tuning file, and that file is malformed/incomplete.  I don't think demoting the error message would be the right thing here.  It should be quite easy to fix the tuning file to pass the validation though.

kbingham, 2023-06-08:
Fixes merged at https://git.libcamera.org/libcamera/libcamera.git/commit/?id=0ee9339331c648232e87d2de2ccd5a92cc61cab2