raspberrypi/libcamera #60: ArduCam IMX219 extreme delay and low FPS on Yocto

landswellsong, 2023-06-09:
I'm building a RPi-based system on Yocto and I'm getting extremely bad performance with the ArduCam IMX219 sensor, but at the same time the exact same setup works flawlessly in RaspiOS images. I thought it could be the version issue, but neither upgrading the kernel, libcamera and RPI firmware helped. In fact I managed to make it work in RaspiOS with a lower kernel than in my Yocto. The `config.txt` is also virtually identical, the md5's of the relevant overlays and the Broadcom blobs also match. I ran `cam` with a lot of debug information like this: `LIBCAMERA_LOG_LEVELS=*:DEBUG ./cam -c 1 --capture=300 --stream width=640,height=480,role=video > rpios-640x480-stdout.log 2> rpios-640x480-stderr.log` (NOTE: I had to compile the `cam` from source at `v0.0.4` tag for RPiOS because they didn't include it in the `libcamera-tools` package, but I do believe that aside from complaining about the paths not being installed it works pretty much the same). The logging also slowed down the operation considerably, it's a RPi0 v1.1 we're talking about, so I also included an stdout log without `LIBCAMERA_LOG_LEVELS` env set for the reference performance.

My immediate observations as a libcamera neophyte:
1. Both builds seem to be missing some frames and thus end up gathering more than is actually requested, but in order to have 300 good frames RPiOS captured 349 frames, whereas Yocto had to capture 527. This is consistent with the FPS being around half the needed one give or take.
2. Some frames in the Yocto build (31 out of 527) are being used for AWB algorithm and at least the last one (frame 527) takes absurdly long 400 ms to run from the "frame started" to "CT found"

I'd be glad if I'm given any pointers on how to fix it or debug further. Both builds are on separate µSDs, so I can boot them and run any further diagnostic too. Kernels are 6.1+ both, libcameras are 0.0.4

[rpios-640x480-stderr.log](https://github.com/raspberrypi/libcamera/files/11699526/rpios-640x480-stderr.log)
[rpios-640x480-stdout.log](https://github.com/raspberrypi/libcamera/files/11699527/rpios-640x480-stdout.log)
[rpios-640x480-notrace.log](https://github.com/raspberrypi/libcamera/files/11699528/rpios-640x480-notrace.log)
[yocto-640x480-stdout.log](https://github.com/raspberrypi/libcamera/files/11699529/yocto-640x480-stdout.log)
[yocto-640x480-stderr.log](https://github.com/raspberrypi/libcamera/files/11699530/yocto-640x480-stderr.log)
[yocto-640x480-notrace.log](https://github.com/raspberrypi/libcamera/files/11699531/yocto-640x480-notrace.log)

6by9, 2023-06-09:
I'd suggest you also eliminate a kernel difference.

After you've run whatever libcamera command to set up the mode, use ```v4l2-ctl --list-ctrls -d /dev/v4l-subdev0``` to note all the control settings in each case, and ```v4l2-ctl -V``` to confirm that format selected. You can then use ```v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null``` to stream just the sensor without ISP or control loops, and it'll report the received frame rate.

landswellsong, 2023-06-09:
@naushir 
1. My reasoning is that `cam` should stop at 300 frames which the `stdout` log confirms, taking into account that the first 6 frames are dropped as expected. When we look at `stderr` log which is from within libcamera code we can see that there were much more frames captured to serve 300 frames to the application using libcamera. I may be wrong, but the fact that on RPiOS the number is very close to 300 and on Yocto it's off by a factor or 2 aligns with the observations. 
2. Yes, I can repro on Rpi4, but tomorrow. Not sure if I have an official Pi camera, but I can check.
3. I honestly don't know what the cam's target is. From the logs it seems that they are targeting 15 fps. In my actual application using gstreamer I'm realiably getting 30 to 40 in RpiOS for the whole pipeline, so 30 is good enough. But the biggest issue is delay since it's half a second (again consistent with huge 400 ms gaps in the log)
4. I had to build `cam` from source on RPiOS so it's not intentional. 

As to exposure I think it relates to `cam` because there's no way to request FPS I could find. Here are the logs from `libcamera-vid`, they are pretty similar except for the correct 33 ms exposure for 30 fps and the fact it's even more frames for Yocto now. In both cases the log was being written to `tmpfs` to remove any disk IO factor. 
[yocto-libcamvid-stderr.log](https://github.com/raspberrypi/libcamera/files/11705374/yocto-libcamvid-stderr.log)
[rpios-libcamvid-stderr.log](https://github.com/raspberrypi/libcamera/files/11705375/rpios-libcamvid-stderr.log)

@6by9 I ran it and in both cases 30 fps were acheived. I'm thinking the kernel behaves the same?
RPiOS for the commands provided:
```
pi@raspberrypi:~ $ uname -a 
Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
pi@raspberrypi:~ $ v4l2-ctl --list-ctrls -d /dev/v4l-subdev0

User Controls

                       exposure 0x00980911 (int)    : min=4 max=1759 step=1 default=1600 value=1759
                horizontal_flip 0x00980914 (bool)   : default=0 value=1 flags=modify-layout
                  vertical_flip 0x00980915 (bool)   : default=0 value=1 flags=modify-layout

Camera Controls

             camera_orientation 0x009a0922 (menu)   : min=0 max=2 default=2 value=2 flags=read-only
         camera_sensor_rotation 0x009a0923 (int)    : min=180 max=180 step=1 default=180 value=180 flags=read-only

Image Source Controls

              vertical_blanking 0x009e0901 (int)    : min=32 max=65055 step=1 default=1283 value=1283
            horizontal_blanking 0x009e0902 (int)    : min=2808 max=32112 step=1 default=2808 value=2808
                  analogue_gain 0x009e0903 (int)    : min=0 max=232 step=1 default=0 value=217
                red_pixel_value 0x009e0904 (int)    : min=0 max=1023 step=1 default=1023 value=1023
          green_red_pixel_value 0x009e0905 (int)    : min=0 max=1023 step=1 default=1023 value=1023
               blue_pixel_value 0x009e0906 (int)    : min=0 max=1023 step=1 default=1023 value=1023
         green_blue_pixel_value 0x009e0907 (int)    : min=0 max=1023 step=1 default=1023 value=1023

Image Processing Controls

                 link_frequency 0x009f0901 (intmenu): min=0 max=0 default=0 value=0 flags=read-only
                     pixel_rate 0x009f0902 (int64)  : min=182400000 max=182400000 step=1 default=182400000 value=182400000 flags=read-only
                   test_pattern 0x009f0903 (menu)   : min=0 max=4 default=0 value=0
                   digital_gain 0x009f0905 (int)    : min=256 max=4095 step=1 default=256 value=256
pi@raspberrypi:~ $ v4l2-ctl -V
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'pBAA' (10-bit Bayer BGBG/GRGR Packed)
        Field             : None
        Bytes per Line    : 800
        Size Image        : 384000
        Colorspace        : Raw
        Transfer Function : Default (maps to None)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             : 
pi@raspberrypi:~ $ v4l2-ctl --stream-mmap=3 --stream-count=1000 --stream-to=/dev/null
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
[... 79 lines cut ...]
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<
```

naushir, 2023-06-09:
> In both cases the log was being written to `tmpfs` to remove any disk IO factor.

Could you capture these logs from libcamera-vid again with the following env setup ``LIBCAMERA_LOG_NO_COLOR=1 LIBCAMERA_LOG_LEVELS=RPI:0,IPARPI:0`` and post the output please?

landswellsong, 2023-06-09:
Neat, I was looking for a "no colors" option ^_^
[yocto-libcamvid-rpi-stderr.log](https://github.com/raspberrypi/libcamera/files/11705851/yocto-libcamvid-rpi-stderr.log)
[rpios-libcamvid-rpi-stderr.log](https://github.com/raspberrypi/libcamera/files/11705852/rpios-libcamvid-rpi-stderr.log)

naushir, 2023-06-09:
Thanks for that.  From these logs, both RPi OS and yocto processed the exact same number of frames - 308.  This sounds like it corresponds with the ``--frames 300`` that you have (probably) set.  So it *feels* like everything is working as expected given you have fixed the framerate to 30 in libcamera-vid?

landswellsong, 2023-06-09:
I don't think so, the timing is dramatically different:
```
$ grep 'Drop' rpios-libcamvid-rpi-stderr.log 
[0:02:03.689507361] [671] DEBUG IPARPI raspberrypi.cpp:336 Drop 7 frames on startup
$ tail -n 1 rpios-libcamvid-rpi-stderr.log 
[0:02:14.008300444] [665] DEBUG RPISTREAM rpi_stream.cpp:238 Queuing buffer 0 for ISP Stats
$ grep 'Drop' yocto-libcamvid-rpi-stderr.log 
[0:53:19.983466830] [270] DEBUG IPARPI raspberrypi.cpp:334 Drop 7 frames on startup
$ tail -n 1 yocto-libcamvid-rpi-stderr.log 
[0:53:42.994356530] [270] DEBUG IPARPI raspberrypi.cpp:1394 Applying WB R: 1.07356 B: 2.76785
```
This gives 29.07317 fps for RPiOS average since the "Drop 7 frames" message
But only 13.03731 fps for Yocto between the same logging points, again consistent with what I'm getting with `libcamerasrc` pipeline element in acutal application.

Moreover, for full logging cases, the number of "frame X started" messages differs as noted above:
```
$ sed -ne 's/^.*frame \([0-9]*\) started/\1/p' rpios-libcamvid-stderr.log | xargs
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 256 257 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 277 278 279 280 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 300 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 320 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 340 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 360 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 380 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 400 401 402 403 404 405 406 407 408 409 410 411 412 413 414 415 416 417 418 419 420 421 422 423 424 425 426 427 428 429 430 431 432 433 434 435 436 437 438 439 440 441 442 443 444 445 446 447 448 449 450 451 452 453 454 455 456 457 458 459 460 461 462 463 464 465 466 467 468 469 470 471 472 473 474 475 476 477 478 479 480 481 482 483 484 485 486 487 488 489 490 491 492 493 494
$ sed -ne 's/^.*frame \([0-9]*\) started/\1/p' yocto-libcamvid-stderr.log | xargs
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175 176 177 178 179 180 181 182 183 184 185 186 187 188 189 190 191 192 193 194 195 196 197 198 199 200 201 202 203 204 205 206 207 208 209 210 211 212 213 214 215 216 217 218 219 220 221 222 223 224 225 226 227 228 229 230 231 232 233 234 235 236 237 238 239 240 241 242 243 244 245 246 247 248 249 250 251 252 253 254 255 256 257 258 259 260 261 262 263 264 265 266 267 268 269 270 271 272 273 274 275 276 277 278 279 280 281 282 283 284 285 286 287 288 289 290 291 292 293 294 295 296 297 298 299 300 301 302 303 304 305 306 307 308 309 310 311 312 313 314 315 316 317 318 319 320 321 322 323 324 325 326 327 328 329 330 331 332 333 334 335 336 337 338 339 340 341 342 343 344 345 346 347 348 349 350 351 352 353 354 355 356 357 358 359 360 361 362 363 364 365 366 367 368 369 370 371 372 373 374 375 376 377 378 379 380 381 382 383 384 385 386 387 388 389 390 391 392 393 394 395 396 397 398 399 400 401 402 403 404 405 406 407 408 409 410 411 412 413 414 415 416 417 418 419 420 421 422 423 424 425 426 427 428 429 430 431 432 433 434 435 436 437 438 439 440 441 442 443 444 445 446 447 448 449 450 451 452 453 454 455 456 457 458 459 460 461 462 463 464 465 466 467 468 469 470 471 472 473 474 475 476 477 478 479 480 481 482 483 484 485 486 487 488 489 490 491 492 493 494 495 496 497 498 499 500 501 502 503 504 505 506 507 508 509 510 511 512 513 514 515 516 517 518 519 520 521 522 523 524 525 526 527 528 529 530 531 532 533 534 535 536 537 538 539 540 541 542 543 544 545 546 547 548 549 550 551 552 553 554 555 556 557 558 559 560 561 562 563 564 565 566 567 568 569 570 571 572 573 574 575 576 577 578 579 580 581 582 583 584 585 586 587 588 589 590 591 592 593 594 595 596 597 598 599 600 601 602 603 604 605 606 607 608 609 610 611 612 613 614 615 616 617 618 619 620 621 622 623 624 625 626 627 628 629 630 631 632 633 634 635 636 637 638 639 640 641 642 643 644 645 646 647 648 649 650 651 652 653 654 655 657 660 66 [... cut ...]  906 907 908 909 910 911 912 913 914 915 916 917 918 919 920 921 922 923 924 925 926 927 928 929 930 931 932 933 934 935 936 937 938 939 940 941 942 943 944 945 946 947 948 949 950 951 952 953 954 955 956 957 958 959 960 961 962 963 964 965 966 967 968 969 970 971 972 973 974 975 976 977 978 979 980 981 982 983 984 985 986 987 988 989 990 991 992 993 994 995 996 997 998 999 1000 1001 1002 1003 1004 1005 1006 1007 1008 1009 1010 1011 1012 1013 1014 1015 1016 1017 1018 1019 1020 1021 1022 1023
```

naushir, 2023-06-09:
> 
> Moreover, for full logging cases, the number of "frame X started" messages differs as noted above:
> 

The last logs you sent don't have any "frame X started messages" in either of the files.  Am I looking at the right logs?

Additionally, the yocto log shows local changes might be present in the build (annotated by the dirty flag in the version string):
```
[0:53:18.626140573] [263]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+dirty (2023-06-08T01:03:49+00:00)
```
Would you be able to revert all local changes just to be sure?

landswellsong, 2023-06-09:
I think this message comes with a tag we didn't select for:
```
[0:01:51.799576660] [624] DEBUG RPiDelayedControls delayed_controls.cpp:240 frame 494 started
```
Do you want me to rerun and activate `RPiDelayedControls` logs too?

As to changes, it's the openembedded patchset: https://github.com/openembedded/meta-openembedded/tree/mickledore/meta-multimedia/recipes-multimedia/libcamera/libcamera

As a matter of fact one single small patch. Should I revert it and try?

naushir, 2023-06-09:
I'm looking for any messages in the pipeline handler (under the ``RPI`` tag).  I don't seem to find any in your logs after the camera starts up.  In particular, I want to track this: https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/pipeline/rpi/common/pipeline_base.cpp#n1351
(note that your tree is older than upsteam, so the equivalent line is in raspberrypi.cpp in your builds)

The patch is only for the cam application, so should be inconsequential.  No need to remove it.

landswellsong, 2023-06-09:
Also the build config is here: https://github.com/openembedded/meta-openembedded/blob/mickledore/meta-multimedia/recipes-multimedia/libcamera/libcamera_0.0.4.bb#L30
It's ammended by RPi layer like this: https://github.com/agherzan/meta-raspberrypi/blob/mickledore/dynamic-layers/multimedia-layer/recipes-multimedia/libcamera/libcamera_%25.bbappend

landswellsong, 2023-06-09:
There's notably not a lot under `RPI` tag at all:
```
$ grep 'DEBUG RPI[^A-Z]' rpios-libcamvid-rpi-stderr.log
[0:02:03.543035870] [665] DEBUG RPI raspberrypi.cpp:1253 Unable to acquire a Unicam instance
[0:02:03.543914831] [664] DEBUG RPI raspberrypi.cpp:170 Format: 640x480 fmt SRGGB10 Score: 10248.8 (best 10248.8)
[0:02:03.544328813] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1640x1232 fmt SRGGB10 Score: 6744 (best 6744)
[0:02:03.544633799] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1920x1080 fmt SRGGB10 Score: 6655.48 (best 6655.48)
[0:02:03.544935786] [664] DEBUG RPI raspberrypi.cpp:170 Format: 3280x2464 fmt SRGGB10 Score: 1000 (best 1000)
[0:02:03.545242772] [664] DEBUG RPI raspberrypi.cpp:170 Format: 640x480 fmt SRGGB8 Score: 11248.8 (best 1000)
[0:02:03.545541759] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1640x1232 fmt SRGGB8 Score: 7744 (best 1000)
[0:02:03.546027737] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1920x1080 fmt SRGGB8 Score: 7655.48 (best 1000)
[0:02:03.546357722] [664] DEBUG RPI raspberrypi.cpp:170 Format: 3280x2464 fmt SRGGB8 Score: 2000 (best 1000)
[0:02:03.546801703] [664] DEBUG RPI raspberrypi.cpp:170 Format: 640x480 fmt SRGGB10 Score: 9248.81 (best 9248.81)
[0:02:03.547169686] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1640x1232 fmt SRGGB10 Score: 5744 (best 5744)
[0:02:03.547476673] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1920x1080 fmt SRGGB10 Score: 5655.48 (best 5655.48)
[0:02:03.547858656] [664] DEBUG RPI raspberrypi.cpp:170 Format: 3280x2464 fmt SRGGB10 Score: 0 (best 0)
[0:02:03.548174642] [664] DEBUG RPI raspberrypi.cpp:170 Format: 640x480 fmt SRGGB8 Score: 10248.8 (best 0)
[0:02:03.548471628] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1640x1232 fmt SRGGB8 Score: 6744 (best 0)
[0:02:03.548763616] [664] DEBUG RPI raspberrypi.cpp:170 Format: 1920x1080 fmt SRGGB8 Score: 6655.48 (best 0)
[0:02:03.549052603] [664] DEBUG RPI raspberrypi.cpp:170 Format: 3280x2464 fmt SRGGB8 Score: 1000 (best 0)
```

I'm very much at loss why is the message you are after not showing up, however the very next line apparantly goes here: https://github.com/raspberrypi/libcamera/blob/v0.0.4/src/libcamera/pipeline/raspberrypi/delayed_controls.cpp#L238
And this message does get shown and it is the one I mentioned before when counting frames.

naushir, 2023-06-09:
Unfortunately nothing from the logs gives any clues as to what's going on.  From what I can tell, the frames are coming out of the sensor at an unexpected rate - both logs show they are programmed with the same vertical blanking value, so they really ought to be framing identically.  And the ``v4l2-ctl`` streaming test seems to confirm this.  So I'm really stumped...

It would be useful to get the pipeline handler logging available if you can get that working.

landswellsong, 2023-06-09:
Thank you for your guidance. I'll try to hack the pipeline logging over the weekend since it's more than I can affort to be distracted from work at the moment and will report back.

landswellsong, 2023-06-13:
Didn't have time for a full debugging session, but upgraded libcamera to `v0.0.5` just in hopes that RPiOS might have a slightly fresher build, but to no avail.

Also tried the `--awbgains` option that disables the AWB algorithm and it shortened the 300 frames run from 42.350 sec (7.08 fps average) to 29.49 sec (10.172 fps average)

landswellsong, 2023-06-13:
I realized that RPiOS uses this distro to build while Yocto defaults to official libcamera
I retargeted to this repo and 923f5d707bb760bd3e724b3373568fa88c68454f commit 

Now they are on the same commit exactly (not sure why it's marked as dirty, the build directory is clean w/o patches):
```
[0:02:22.377654000] [536]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:01:28.742257546] [242]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70-dirty (2023-06-13T16:30:58+00:00)
```

And there is no significant change at all compared to the official repo, still bad performance

landswellsong, 2023-06-13:
I've changed the severity of the pipeline message asked above to `INFO` and it started popping up in the logs. I honestly don't know why doesn't it do so otherwise at `DEBUG` level, it defies my superficial understanding of the code. But here are the 2 logs, 300 frames, one is with `--awbgain ` option added, another is without it. As you can see we do end up capturing a lot more than 300 frames for some reason.

[yocto_pipeline_info_no_awb.log](https://github.com/raspberrypi/libcamera/files/11738403/yocto_pipeline_info_no_awb.log)
[yocto_pipeline_info.log](https://github.com/raspberrypi/libcamera/files/11738404/yocto_pipeline_info.log)

landswellsong, 2023-06-13:
Another bunch of observations:
1. My build was not linking `libdw` and `libunwind`. Added those, no effect. (Just in case)
2. Frame to frame comparison based on "frame started" log entries show that the frames _on average_ are processed at a similar speed:
[image]
But then the yocto one takes way more frames for the same command.

landswellsong, 2023-06-13:
Another difference spotted. Reference RPiOS:
```
[0:02:38.343264000] [543] DEBUG IPARPI raspberrypi.cpp:1503 Applying AGC Exposure: 25217.28us (Shutter lines: 1334, AGC requested 25228.05us) Gain: 2 (Gain Code: 128)
[0:02:38.347062000] [543] DEBUG IPARPI raspberrypi.cpp:1364 Metadata - Exposure: 25217.28us Frame length: 1763 Line length: 18.90us Gain: 2
```

Yocto:
```
[0:02:10.995561424] [249] DEBUG IPARPI raspberrypi.cpp:1503 Applying AGC Exposure: 26275.88us (Shutter lines: 2780, AGC requested 26281.02us) Gain: 2 (Gain Code: 128)
[0:02:11.023308394] [249] DEBUG IPARPI raspberrypi.cpp:1364 Metadata - Exposure: 26275.88us Frame length: 3526 Line length: 9.45us Gain: 2
```

Shutter lines are double in yocto, frame length is also double, and line length is half compared to reference. Unsure what any of that means honestly, but it's an apple to apple comparison of the same commit, same camera, same Pi, same environment and same command.

landswellsong, 2023-06-13:
More insight, the IPA log entries contain 309 groups in both cases what I assume to be frames. 

[image]
For reference, the mean delay is about 50 ms (I assume there's an overhead for logging, it's a RPi0 after all)

[image]
for yocto the mean delay is about 130 ms and there are distinct peaks at roughly the "normal" 50 as well as the 130-150 ms mark

[image]
interestingly w/o AWB algo running (`--awbgains` set) the lower peak disappears, but the mean is shifted to 96 ms

So in my layman perspective:
1. The sensor is pumping data at the same rate in both cases
2. The Pipeline (?) part is accepting frames roughly at the same rate on average, but with much bigger variance at edge cases with distinct classes of delays
3. The IPA part is significantly slower in Yocto case, turning off bits of it (e.g. AWB) has a tangible effect
4. The delays at IPA cause the pipeline to be working for much longer, resulting in capturing more frames than needed since not all of them can be processed in time due to slow IPA
5. Possibly(?) the holdups in the IPA are affecting the pipeline performance too because it's a single core device after all

I will re-test on RPi4 at the weekend or so to verify the (5) bit as having extra cores would allow the libcamera threads not to have much of an impact on each other. Aside from that I have completely no idea how to profile the IPA further.

naushir, 2023-06-14:
So from your updated logs, the 

> Another difference spotted. Reference RPiOS:
>
> ```
> [0:02:38.343264000] [543] DEBUG IPARPI raspberrypi.cpp:1503 Applying AGC Exposure: 25217.28us (Shutter lines: 1334, AGC requested 25228.05us) Gain: [...]

This looks to be the difference right here!  For some reason, your Yocto build kernel driver is programming the sensor frame duration to an unexpected value.  In the RPi case, the line length is reported by the sensor driver to be 18.90us but in Yocto it's 9.45us.  The IPA uses this value to compute the required frame length (visible + blanking) and exposure time.  If the values reported by the sensor driver are wrong, the calculations go wrong and you get an unexpected framing time from the sensor.

Now as to why the Yocto kernel driver is reporting the wrong value... I'm not sure.  Can you check you have identical kernel driver source files from the RPi downstream kernel?  Also worth checking your imx219 overlay files are identical as well.  Given that this is an Arducam module, you want to also check the XCLK crystal frequency and ensure the value is correct in the imx219 overlay file.  The RPi overlay sets this to 24MHz.

landswellsong, 2023-06-14:
The overlays were identical to the md5sum last time I checked. I'll try checking the source.

landswellsong, 2023-06-14:
Note to self, libcamera gets the values from here: https://github.com/raspberrypi/libcamera/blob/main/src/ipa/raspberrypi/cam_helper_imx219.cpp#L31

The kernel module sets the values here:
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/media/i2c/imx219.c#L769

Frame Length is same register as `IMX219_REG_VTS` (V-blank)
Line Length is same register as `IMX219_REG_HTS` (H-blank)

Does it make sense? Probably does, just making sure.

The kernel doesn't set it to anything in particular within imx219 driver per se and just exposes as a V4L2 control. I'll check the value at bootup in both cases and see if it's anyhow different.

naushir, 2023-06-15:
> Note to self, libcamera gets the values from here: https://github.com/raspberrypi/libcamera/blob/main/src/ipa/raspberrypi/cam_helper_imx219.cpp#L31

Just to clarify, these values are only used to parse the metadata buffer sent by the sensor.  In the case of imx219, we don't use the metadata buffer at all, so these values are never actually used.  All blanking and timing details come directly from the sensor driver.

naushir, 2023-06-15:
> Frame Length is same register as `IMX219_REG_VTS` (V-blank) Line Length is same register as `IMX219_REG_HTS` (H-blank)
> 
> Does it make sense? Probably does, just making sure.

I'm pretty confident this works as intended at least in the RPi OS case.  If there were any errors, you would see a discrepancy with the framerate the sensor is generating frames at, and what the user/application requested (as seems to be the case for the Yocto build).

landswellsong, 2023-06-17:
Huh, now I am getting:
```
[0:08:02.201179327] [664] DEBUG IPARPI raspberrypi.cpp:1362 Metadata - Exposure: 9999.96us Frame length: 3526 Line length: 9.45us Gain: 1.00787
``` 

in RPiOS as well, argh

landswellsong, 2023-06-17:
What's more puzzling, here the blanking seems identical. I've recorded some i2c trace to analyze.

landswellsong, 2023-06-17:
Additional info:
```
root@raspberrypi:/home/pi# v4l2-ctl -C vertical_blanking -d /dev/v4l-subdev0
vertical_blanking: 3046
root@raspberrypi:/home/pi# v4l2-ctl -c vertical_blanking=1283 -d /dev/v4l-subdev0
root@raspberrypi:/home/pi# v4l2-ctl -C vertical_blanking -d /dev/v4l-subdev0
vertical_blanking: 1283
root@raspberrypi:/home/pi# libcamera-vid --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
root@raspberrypi:/home/pi# v4l2-ctl -C vertical_blanking -d /dev/v4l-subdev0
vertical_blanking: 32
root@raspberrypi:/home/pi# libcamera-vid 
Made DRM preview window
[0:37:25.834117199] [964]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:37:26.149641474] [965]  WARN RPI raspberrypi.cpp:1308 Mismatch between Unicam and CamHelper for embedded data usage!
[0:37:26.165811386] [965]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:37:26.184024286] [964]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420
[0:37:26.188880259] [965]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
Halting: reached timeout of 5000 milliseconds.
root@raspberrypi:/home/pi# v4l2-ctl -C vertical_blanking -d /dev/v4l-subdev0
vertical_blanking: 3046
```

Seems like someone does mess with this value.

landswellsong, 2023-06-17:
Moreover while gstreamer is running, it's constantly updated too:
```
root@raspberrypi:/home/pi# systemctl start camera@default
root@raspberrypi:/home/pi# v4l2-ctl -C vertical_blanking -d /dev/v4l-subdev0
vertical_blanking: 3046
root@raspberrypi:/home/pi# v4l2-ctl -c vertical_blanking=1283 -d /dev/v4l-subdev0
root@raspberrypi:/home/pi# v4l2-ctl -C vertical_blanking -d /dev/v4l-subdev0
vertical_blanking: 3046
```

naushir, 2023-06-26:
I'm a bit unsure how to help with debugging this issue further unfortunately.  @landswellsong do you have any further updates on this?

naushir, 2023-07-05:
Is there any further updates to this issue?  If not, I'll close this shortly.

closed 2023-07-07 by naushir

Raymchk, 2025-01-26:
Excuse me, what are the necessary packages to include to the yocto built to use correctly the camera ?