raspberrypi/libcamera #61: Pi Camera Module v3 PDAF Issues

RyanMurdochWillcom, 2023-06-12:
I urgently need assistance as this issue is holding up multiple projects I have running.

I'm having issues with multiple Pi Camera Module 3's where I'm getting the following errors:

`ERROR IPARPI cam_helper_imx708.cpp:247 PDAF data in unsupported format
ERROR IPARPI cam_helper_imx708.cpp:311 Lin/Log histogram mismatch`

I've tried different ribbon cables, connecting and reconnecting them but have no success. I'm using the latest picamera2 library in Python.

The script actually doesn't break but rather reports the errors via console as per the attached screenshot. These errors seem to start after initialising the camera `Picamera2().start()` and keeps printing these errors frequently without even taking any pictures.

[cam_bug_report.txt](https://github.com/raspberrypi/libcamera/files/11721222/cam_bug_report.txt)

[image]

Find the `camera-bug-report` output also attached.

naushir, 2023-06-12:
Pinging @njhollinghurst.  This is definitely not expected, it possibly indicates a transmission error for the metadata buffer over the CSI-2 bus. You software environment seems to all be in order.

A few questions to help narrow things down:

1) How long is the ribbon connector cable you are using here?
2) Have you tried another Pi4 device?
3) Are you able to share a minimum picamera2 script that triggers this behaviour?
4) Could you try one of the applications from [``libcamera-apps``](https://github.com/raspberrypi/libcamera-apps) to see if you get the same error messages?

RyanMurdochWillcom, 2023-06-13:
@naushir The ribbon cable is the one that ships with the PiCamera v3 (15cm length). I have tried it on other Pi4 devices and also tried different PiCamera 3's. A basic script that also gives the same error is as follows:

```
from picamera2 import Picamera2
import time

picam2 = Picamera2()

def main():
    init_Camera()

    for i in range(5):
        take_photo()
        time.sleep(5)

def init_Camera():

    picam2.stop()

    still_image_size = {'size':[4608, 2592]}

    camera_config = picam2.create_still_configuration(main=still_image_size)

    picam2.set_controls({
        'ExposureTime': 1000,
        'AfMode':2,
    })
    
    picam2.configure(camera_config)
    picam2.start()

def take_photo():
    captImage = picam2.capture_image()
    captImage.save('image.jpg', format='JPEG')
    

if __name__ == '__main__':
    try:
       main()
    except Exception as e:
        print(f"{e}", flush=True)
```

Could you please point me to a specific app which I should try and how/where should I set the environment variable?

naushir, 2023-06-13:
There are a bunch of AF related example scripts and apps in the picamera2 repo that you could try out.  A good one to start with would be [this](https://github.com/raspberrypi/picamera2/blob/main/apps/app_full.py).

Would you be able to provide some further details based on my earlier comment when you get a chance? Additionally, can you set a fixed framerate in your script above to see if it makes a difference:

```python
    picam2.set_controls({
        'ExposureTime': 1000,
        'AfMode':2,
        'FrameDurationLimits': (33333, 33333),
    })
```

naushir, 2023-06-13:
> Is there another app I can try since I don't have a GUI and only a remote SSH connection to the Pi?

If you add a ``-n`` to the command line arguments to ``libcamera-hello/libcamera-vid`` that will disable the preview window and you can run that over ssh.

RyanMurdochWillcom, 2023-06-13:
Okay thanks I've tried that now, but still no luck it seems:

```
root@pi:/Apps# libcamera-hello -t 0 -n
[20:33:20.269795864] [52498]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[20:33:20.387262963] [52499]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
[20:33:20.388217131] [52498]  INFO Camera camera.cpp:1028 configuring streams: (0) 2304x1296-YUV420
[20:33:20.388581150] [52499]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[20:33:21.622971680] [52503] ERROR IPARPI cam_helper_imx708.cpp:247 PDAF data in unsupported format
[20:33:24.755253558] [52503] ERROR IPARPI cam_helper_imx708.cpp:247 PDAF data in unsupported format

root@pi:/Apps# libcamera-vid -t 0 -n
[20:33:44.662016779] [52574]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[20:33:44.792967204] [52575]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media4 and ISP device /dev/media0
[20:33:44.794219279] [52574]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420
[20:33:44.795087724] [52575]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
[20:33:52.999860524] [52581] ERROR IPARPI cam_helper_imx708.cpp:247 PDAF data in unsupported format
[20:33:59.030518435] [52581] ERROR IPARPI cam_helper_imx708.cpp:247 PDAF data in unsupported format
```

naushir, 2023-06-13:
Unfortunately, this is starting to sound like a hardware fault on the camera device.  You mentioned you tried other camera boards, how many showed the fault? Did you have any that didn't show this problem?  Also what kind of environment is the device running in?  Would there be possibility of any sort of electrical or RF interference?

As a last resort, would we like to arrange a swap replacement for the faulty sensors if that is possible with you.  Could you email info@raspberrypi.com and quote this issue and we can start the process.

RyanMurdochWillcom, 2023-06-13:
Thank you I will send an email to Raspberry Pi. I've actually encountered this on all 5 camera modules I have and on various Pi 4's using different ribbon cables. The only thing that does change between the modules is the frequency at which I get those errors. I've had the device running indoors and outdoors (in an IP rated box) and in different locations with the same issue occurring.

naushir, 2023-06-26:
@RyanMurdochWillcom just checking in again, did you get in touch with us through the above email regarding this issue?  If so, I can chase this from our end.

RyanMurdochWillcom, 2023-06-26:
I forgot to touch base with Raspberry Pi but have sent them an email just now. If you’re able to chase it down from your end I will appreciate it.

naushir, 2023-07-24:
@RyanMurdochWillcom did you manage to resolve this through our email channels?

RyanMurdochWillcom, 2023-07-24:
I have emailed them but had no response unfortunately.

naushir, 2023-07-24:
Sorry about that, let me chase this up.

Sebastian-ehrig, 2023-10-09:
Any updates on this issue? I am getting a similar error but couldn't find a solution yet. In my case the camera captures frames but sporadically throws the same error until the camera freezes and eventually stops capturing frames:
"ERROR IPARPI cam_helper_imx708.cpp:273 PDAF data in unsupported format".

I have 7 identical setups with the same camera modules and configurations but only 2 of the devices show this error.

I also emailed the Raspberry support but didn't get a response yet.

naushir, 2023-10-10:
@Sebastian-ehrig this may be a hardware problem with your devices.  I would suggest get in touch with the store/reseller and ask for a replacement part.

naushir, 2023-10-10:
Closing this now, as there is nothing we can do in software about this.

closed 2023-10-10 by naushir

habeebmuraina, 2024-07-09:
I encountered the same problem now. all I did was switch the internet from wireless to using a cable and the error ceased to show. Try that

Sebastian-ehrig, 2024-07-09:
In my case the camera cable was very close to the Mainboard (specifically the power management IC) which caused some interference. Due to the compact design there was no other way of storing the cable. 

Wrapping the CSI cable with copper-tape and then scotch-tape successfully shields the EM-field and solved the issue for all of my 10 devices.