raspberrypi/libcamera #64: Assertion '!this->empty()' failed in libcamera::utils::Duration - libcamera-tools 0.0.5-1

goverp, 2023-06-24:
Raspberry Pi 4 running Arch linux armv7h.
I have the pi camera, and autoconfiguration is selecting the imx219 device overlay.

cam -l says:
```

[0:20:10.571704116] [522]  INFO Camera camera_manager.cpp:298 libcamera v0.0.5
[0:20:10.615908781] [523]  WARN RPI raspberrypi.cpp:1368 Mismatch between Unicam and CamHelper for embedded data usage!
[0:20:10.616795199] [523]  INFO RPI raspberrypi.cpp:1487 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media1
Available cameras:
1: 'imx219' (/base/soc/i2c0mux/i2c@1/imx219@10)
```
but
cam --camera=1 --capture=1 --file
produces:
```
[0:22:03.543965567] [534]  INFO Camera camera_manager.cpp:298 libcamera v0.0.5
[0:22:03.586931637] [535]  WARN RPI raspberrypi.cpp:1368 Mismatch between Unicam and CamHelper for embedded data usage!
[0:22:03.587817592] [535]  INFO RPI raspberrypi.cpp:1487 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media1
Using camera /base/soc/i2c0mux/i2c@1/imx219@10 as cam0
[0:22:03.588456271] [534]  INFO Camera camera.cpp:1028 configuring streams: (0) 800x600-NV12
[0:22:03.588748935] [535]  INFO RPI raspberrypi.cpp:857 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
/usr/include/c++/12.1.0/bits/stl_deque.h:1576: void std::deque<_Tp, _Alloc>::pop_front() [with _Tp = libcamera::utils::Duration; _Alloc = std::allocator<libcamera::utils::Duration>]: Assertion '!this->empty()' failed.
Aborted (core dumped)
```
Suggestions gratefully received!

naushir, 2023-06-26:
That command seems to work for me, but that's not very helpful in debugging this issue.
I'm not too familiar with cam,  are you able to run this under gdb and get a backtrace out when the crash occurred?

goverp, 2023-06-26:
On Monday, 26 June 2023 08:44:55 BST naushir wrote:
> That command seems to work for me, but that's not very helpful in debugging
> this issue. I'm not too familiar with cam,  are you able to run this under
> gdb and get a backtrace out when the crash occurred?
I will have a go.  IIUC, the Arch ARM libcamera and libcamera-tools packages 
will need rebuilding with debug symbols.  This will probably take me a day or 
so; I've not done it before on Arch.

naushir, 2023-07-05:
Any further updates on this issue?

goverp, 2023-07-05:
On Wednesday, 5 July 2023 17:56:14 BST naushir wrote:
> Any further updates on this issue?

Thanks for checking back.  Unfortunately, real life intruded.  Ignore it until 
I either post some useful data, or announce that it's working in version "n".

naushir, 2023-07-24:
Any updates on this issue?  If not, I'll close this down shortly.

goverp, 2023-07-24:
On Monday, 24 July 2023 08:27:16 BST naushir wrote:
> Any updates on this issue?  If not, I'll close this down shortly.
No, sorry, please close it.  If I get some diagnostics, I'll reopen it, but 
with luck the problem with vanish with the tide - or at least some Arch linux 
update!

closed 2023-07-24 by naushir

Torxed, 2023-08-13:
Just to give an update, this issue still exists.
I suspect not many Arch ARM users use the picam module(?) hence feedback will be slow.

A full debug traceback:
```
[alarm@alarmpi ~]$ sudo LIBCAMERA_LOG_LEVELS=*:DEBUG cam --camera 1 --capture=10 --file=./ --stream height=1080,width=2028,role=still
[sudo] password for alarm:
[0:02:06.920768515] [384] DEBUG IPAModule ipa_module.cpp:334 ipa_rkisp1.so: IPA module /usr/lib/libcamera/ipa_rkisp1.so is signed
[0:02:06.926426602] [384] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_rkisp1.so'
[0:02:06.935339196] [384] DEBUG IPAModule ipa_module.cpp:334 ipa_rpi_vc4.so: IPA module /usr/lib/libcamera/ipa_rpi_vc4.so is signed
[0:02:06.935534151] [384] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_rpi_vc4.so'
[0:02:06.938585074] [384] DEBUG IPAModule ipa_module.cpp:334 ipa_vimc.so: IPA module /usr/lib/libcamera/ipa_vimc.so is signed
[0:02:06.938723605] [384] DEBUG IPAManager ipa_manager.cpp:245 Loaded IPA module '/usr/lib/libcamera/ipa_vimc.so'
[0:02:06.942368928] [384] ERROR IPAModule ipa_module.cpp:172 Symbol ipaModuleInfo not found
[0:02:06.942464405] [384] ERROR IPAModule ipa_module.cpp:292 v4l2-compat.so: IPA module has no valid info
[0:02:06.942947884] [384]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0
[0:02:06.943637317] [385] DEBUG Camera camera_manager.cpp:69 Starting camera manager
[0:02:06.977987138] [385] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "rpivid" created from /dev/media3
[0:02:06.978113059] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media3 due to 1 missing dependencies
[0:02:06.978877415] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media3 found
[0:02:06.978956356] [385] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media3: rpivid
[0:02:06.979459426] [385] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media4
[0:02:06.979543960] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media4 due to 5 missing dependencies
[0:02:06.982376522] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media4 found
[0:02:06.982420612] [385] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media4: bcm2835-codec
[0:02:06.982746728] [385] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media0
[0:02:06.982785226] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media0 due to 4 missing dependencies
[0:02:06.983120397] [385] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media1
[0:02:06.983157080] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media1 due to 4 missing dependencies
[0:02:06.985310894] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media0 found
[0:02:06.985347800] [385] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: bcm2835-isp
[0:02:06.987469171] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media1 found
[0:02:06.987504002] [385] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: bcm2835-isp
[0:02:06.988038461] [385] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media2
[0:02:06.988437758] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:96 Defer media device /dev/media2 due to 2 missing dependencies
[0:02:06.989490027] [385] DEBUG DeviceEnumerator device_enumerator_udev.cpp:322 All dependencies for media device /dev/media2 found
[0:02:06.989523211] [385] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: unicam
[0:02:06.989942452] [385] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerISI'
[0:02:06.990368396] [385] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerRkISP1'
[0:02:06.994665173] [385] DEBUG Camera camera_manager.cpp:113 Found registered pipeline handler 'PipelineHandlerVc4'
[0:02:06.997095956] [385] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "unicam"
[0:02:06.997157175] [385] DEBUG DeviceEnumerator device_enumerator.cpp:312 Successful match for media device "bcm2835-isp"
[0:02:06.999962127] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Exposure (0x00980911)
[0:02:07.000046846] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Horizontal Flip (0x00980914)
[0:02:07.000083437] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Vertical Flip (0x00980915)
[0:02:07.000120324] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Camera Orientation (0x009a0922)
[0:02:07.000179433] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Camera Sensor Rotation (0x009a0923)
[0:02:07.000263355] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Vertical Blanking (0x009e0901)
[0:02:07.000338852] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Horizontal Blanking (0x009e0902)
[0:02:07.000411275] [385] DEBUG V4L2 v4l2_device.cpp:636 'imx477 10-001a': Control: Analogue Gain (0x009e0903)
[... cut ...]
#13 0xb6ec89fc in libcamera::CameraManager::Private::run() () from /usr/lib/libcamera.so.0.1
#14 0xb68e6a88 in std::execute_native_thread_routine (__p=0x46e138) at /usr/src/debug/gcc/libstdc++-v3/src/c++11/thread.cc:82
#15 0xb66dc140 in ?? () from /usr/lib/libc.so.6
#16 0xb675dd1c in ?? () from /usr/lib/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

I'll try to compile it manually with debug symbols so that the traceback becomes more useful. Not sure how that works as it's the first time doing it.

Torxed, 2023-08-13:
It's a packaging issue for sure, most likely in Arch ARM. I've posted a forum thread about it: https://archlinuxarm.org/forum/viewtopic.php?f=65&t=16563

Manually building `libcamera` (not using the PKGBUILD) works after some symlinks to the correct paths, or setting PATH to include the default build location for debian libs.

```
$ git clonemake_clickable_callback(MAGIC_URL_FULL, ' ', 'https://git.libcamera.org/libcamera/libcamera.git', '', ' class="postlink"')
$ cd libcamera
$ meson setup build
# ninja -C build install
# ln -s /usr/local/share/libcamera /usr/lib/libcamera
# ln -s /usr/local/lib/* /usr/lib/
# pacman -S sdl2
# LIBCAMERA_LOG_LEVELS=*:DEBUG cam --camera 1 --capture=10 --file=./ --stream height=1080,width=2028,role=still
```

karol57, 2023-09-16:
Hi all,

I investigated the issue. During first call to `IpaBase::applyAGC` function the `IpaBase::frameLengths_` container is empty. This can be a problem as calling `pop_front` on empty container is UB ([cppreference](https://en.cppreference.com/w/cpp/container/deque/pop_front)). Probably STL handles this situation fine, but I think that this should be corrected.

It believe that ArchARM package was compiled with `_GLIBCXX_ASSERTIONS` enabled and this situation is detected.

kbingham, 2023-09-17:
Excellent - thanks for the diagnosis.

Perhaps this can be updated easily to something like:
```
if (!frameLengths_.empty())
        frameLengths_.pop_front();
```

I would suspect that Arch ARM should not compile with those assertions enabled for packaging, but that /we/ should have them enabled in non-release builds.

naushir, 2023-09-17:
I doubt this is the problem. The queue is populated on camera start (with 0.0s), so should never be empty.

naushir, 2023-09-17:
Having said that, I wonder is applyAgc is called before that point? In which case we should initialise the queue at an earlier point.

kbingham, 2023-09-17:
It looks like IpaBase::start() initialises it at : https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/common/ipa_base.cpp#L276, but this is being called during libcamera::ipa::RPi::IpaBase::configure() according to the backtraces above.

naushir, 2023-09-18:
Perhaps this is the fix we are looking for:

```diff
diff --git a/src/ipa/rpi/common/ipa_base.cpp b/src/ipa/rpi/common/ipa_base.cpp
index a47ae3a9e5cb..7cdab5d49db7 100644
--- a/src/ipa/rpi/common/ipa_base.cpp
+++ b/src/ipa/rpi/common/ipa_base.cpp
@@ -203,6 +203,11 @@ int32_t IpaBase::configure(const IPACameraSensorInfo &sensorInfo, const ConfigPa
        /* The pipeline handler passes out the mode's sensitivity. */
        result->modeSensitivity = mode_.sensitivity;

+       /* Reset the frame lengths queue state. */
+       lastTimeout_ = 0s;
+       frameLengths_.clear();
+       frameLengths_.resize(FrameLengthsQueueSize, 0s);
+
        if (firstStart_) {
                /* Supply initial values for frame durations. */
                applyFrameDurations(defaultMinFrameDuration, defaultMaxFrameDuration);
@@ -277,11 +282,6 @@ void IpaBase::start(const ControlList &controls, StartResult *result)

        controller_.switchMode(mode_, &metadata);

-       /* Reset the frame lengths queue state. */
-       lastTimeout_ = 0s;
-       frameLengths_.clear();
-       frameLengths_.resize(FrameLengthsQueueSize, 0s);
-
        /* SwitchMode may supply updated exposure/gain values to use. */
        AgcStatus agcStatus;
        agcStatus.shutterTime = 0.0s
```

kbingham, 2023-09-18:
Aha - I saw the 'unedited' version in my email with the removal of the re-clear / init at start() So I was going to say "should we keep the reinitialisation there for start/stop/start sequences ... "

But it looks like the edit removes that and just adds the initialisation at construct which sounds reasonable so far.
Are there any other missing fields that should be initialised at construct too ?

naushir, 2023-09-18:
> Are there any other missing fields that should be initialised at construct too ?

None that I can tell, but we will find out :-)