raspberrypi/libcamera #67: Invalid IPA module name 'rpi/vc4'

bfree-github, 2023-07-07:
I'm able to use libcamera-vid on a Raspberry pi4b running Raspberry Pi OS, but get an IPA error when I attempt to do it running on the same device using Ubuntu 22.04.2 LTS

I'm using the same dtoverlay settings in Raspberry Pi OS and Ubuntu.

On Ubuntu, I've built libcamera and libcamera-apps from github source.

When running on Ubuntu I'm able to open an X/EGL preview window, but get the following error:
ERROR IPAModule ipa_module.cpp:312 ipa_rpi_vc4.so: Invalid IPA module name 'rpi/vc4'

I've verified that this file exists:
/usr/local/lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so

When I output libcamera-jpeg to a file on Ubuntu, the image is garbled.

What am I missing? - TIA

naushir, 2023-07-07:
Can you run your application with ``LIBCAMERA_LOG_LEVELS=*:0`` set as an environment variable and post the output please?

bfree-github, 2023-07-07:
Thanks for your rapid response!

My sensor is an Arducam B0425 IMX327
I'm using an imx290 dtoverlay - as instructed, although this may be causing the image format issues.

I'm using: libcamera-jpeg -o test.jpg 2&>test.log

There's nothing in the preview.  The resulting test.jpg looks like a format/dimension/interlacing issue.
[image]

The test.log indicates:
* Invalid IPA module name 'rpi/vc4'
[test.log](https://github.com/raspberrypi/libcamera/files/11986386/test.log)

Thanks!

bfree-github, 2023-07-07:
These are from the same Raspberry pi4B and same Arducam camera B0425

libcamera-jpeg --version > libcamera.txt

Ubuntu (built from source):
libcamera-apps build: d932d3132a24-intree 07-07-2023 (04:14:30)
libcamera build: v0.0.4+23-06bbff9c

rpios (installed)
libcamera-apps build: 54a781dffdd1 01-03-2023 (17:53:39)
libcamera build: v0.0.4+22-923f5d70

rpios test.jpeg:
The camera is upside down and the IR filter off
[image]

rpios test.log:
Made X/EGL preview window
[0:06:04.106156807] [2045]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:06:04.159278651] [2049]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx290@1a to Unicam device /dev/media0 and ISP device /dev/media2
[0:06:04.160289889] [2045]  INFO Camera camera.cpp:1028 configuring streams: (0) 960x540-YUV420
[0:06:04.160707716] [2049]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format: 1280x720-SRGGB12_1X12 - Selected unicam format: 1280x720-pRCC
[0:06:09.454368856] [2045]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUV420 (1) 1920x1080-SRGGB12_CSI2P
[0:06:09.455640178] [2049]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx290@1a - Selected sensor format: 1920x1080-SRGGB12_1X12 - Selected unicam format: 1920x1080-pRCC
Still capture image received

bfree-github, 2023-07-07:
I note that the configured streams dimension is different...
ubuntu: 968x548-YUV420 (not on a 64/32/16 byte boundary)
rpios: 960x540-YUV420

ubuntu selects a camera once:
sensor format: 1280x720-SRGGB12_1X12, unicam format: 1280x720-pRCC

rpios does it twice, switching to a higher res:
sensor format: 1280x720-SRGGB12_1X12 - unicam format: 1280x720-pRCC
sensor format: 1920x1080-SRGGB12_1X12 - unicam format: 1920x1080-pRCC

The ubuntu streams configuration dimensions seem suspicious.

Guess I'm going to have to learn about compiling dtoveray files, and dive into libcamera code ;^)

Any help is much appreciated!

bfree-github, 2023-07-08:
@naushir - after digging through code, I've found that the raspberrypi libcamera-apps is not compatible with either the raspberrypi or upstream libcamera.

raspberrypi libcamera-apps uses rpi/cv4; raspberrypi libcamera used raspberrypi/vc4.
I tried using the upstream libcamera (which uses rpi/vc4), but it lacks support for the Streams method, which is used by raspberrypi's libcamera-apps.

I attempted to update the raspberrypi libcamera-apps/core/libcamera_app.cpp to replace rpi with raspberrypi, but that didn't seem to make a difference.  I'm guess it's being provided in some other IPA library.

If you can point me where else I need to update code, I'll give it a go - although I suspect it'll be faster/easier for you to do it.  Is there a previous commit that I can roll back to where these work together?

Thanks for your assistance in this!

FacuM, 2023-07-10:
> @naushir - after digging through code, I've found that the raspberrypi libcamera-apps is not compatible with either the raspberrypi or upstream libcamera.
>
> raspberrypi libcamera-apps uses rpi/cv4; [...]

I think you could switch to the latest version (v4.0.0) by locking your cloning step to the commit related to that tag. I've been experiencing the same issue (in my case, it doesn't even build now).

bfree-github, 2023-07-18:
@naushir - switching to the 3d9ac10 commit did the trick - it's working now on Ubuntu - yay!

bfree-github, 2023-07-18:
Thanks for the 3d9ac10 update!

closed 2023-07-18 by bfree-github