raspberrypi/libcamera #69: Unable to disable auto white balance

bfree-github, 2023-07-18:
I need to disable awb for some cv work that I'm doing.

I've attempted to set options in the cameralib-apps and in config.txt to no avail.  If I put my hand in front of the camera, the white balance changes the background colors completely. Same effect if I set "--awb indoor" or any other "fixed" awb setting.  I see numerous postings that indicate the same experience.

I'm using an imx327 (same as imx290 w/ a switchable IR-cut filter).

I'd be happy to work on the code myself, if someone can point me in the right direction - thx.

naushir, 2023-07-18:
If you set  AWB to fixed and specify manual gains, you should not see the colour gains changing at all.  Can you provide the command line you are using?

However, I suspect your issue might actually be related to auto lens shading correction.  If you open up the imx327.json config file, under the key ``rpi.alsc``, there is a ``n_iter`` pram that you can set to ``0`` to disable the auto correction.  See if that makes a difference to your setup.

naushir, 2023-07-24:
Any updates on this issue?  If not, I'll close this down shortly.

bfree-github, 2023-07-24:
I tried your suggestions without effect.  I'm continuing to try other options and will post if I find a solution.

bfree-github, 2023-07-25:
Modifying n_iter had no effect.

Initially it did not appear that awb-gains disabled awb... however after some experimentation I found that it did disable awb, but the background continued to change due to autoexposure.

I eventually set "--awb-gains 1.8,1.8 --gains 4.0" and the background remained fixed.

I'll need to tune those values, but at least I've now confirmed that I'm able to disable 3A and do my own image processing.

Thanks!

closed 2023-07-25 by bfree-github