raspberrypi/libcamera #71: RAW 16-bit Bayer fourcc codes in drm_fourcc.h don't match those in include/linux/videodev2.h

octopus-russell, 2023-07-21:
Some of the Bayer format fourcc codes drm_fourcc.h (line 422) don't match those in videodev2.h (line 677)
All the Bayer codes in drm_fourcc.h appear to be added by libcamera and are not present in mainline Linux 'drm_fourcc.h' so I'm inclined to think the ones added in drm_fourcc.h should to be changed to match videodev2.h

This appears to be the cause of 16-bit Bayer formats causing this error:
[2:20:09.584212451] [47461]  WARN V4L2 v4l2_pixelformat.cpp:312 Unsupported V4L2 pixel format <INVALID>
[2:20:09.584385901] [47461]  WARN Formats formats.cpp:838 Unsupported pixel format 0x00000000
when validating a 16-bit Bayer format

6by9, 2023-07-21:
The only one I see that doesn't match is
```
#define DRM_FORMAT_SRGGB16	fourcc_code('R', 'G', 'B', '6')
```
vs 
```
#define V4L2_PIX_FMT_SRGGB16 v4l2_fourcc('R', 'G', '1', '6') /* 16  RGRG.. GBGB.. */
```

However AIUI the V4L2 pixel format fourcc should be independent of the DRM (and hence libcamera) one. It implies that something is searching either for the wrong value, or using the wrong search function.

Your line numbers don't match up with the current repo
```WARN V4L2 v4l2_pixelformat.cpp:312 Unsupported V4L2 pixel format ```
https://github.com/raspberrypi/libcamera/blob/main/src/libcamera/v4l2_pixelformat.cpp#L328
```WARN Formats formats.cpp:838 Unsupported pixel format 0x00000000```
https://github.com/raspberrypi/libcamera/blob/main/src/libcamera/formats.cpp#L918
Please either report which version of the libcamera source you are modifying, or work with top of tree.
At a guess you're missing at least https://github.com/raspberrypi/libcamera/commit/e0a51061bc69568f6b6b8f956b327f850f094430, so from before April.

Or better yet would be to be using the mainline libcamera repo (https://git.linuxtv.org/libcamera.git/) as this is largely a staging branch for Raspberry Pi OS releases. Any fix should be submitted to the main libcamera project first, and then ripples down.

octopus-russell, 2023-07-21:
Ah yes it is just the one 'typo' (they're listed in different orders so it seemed like more than one of them was wrong).
We're on a libcamera fork to add support for a new Sony sensor which can do 16-bit RAW output, its public: https://github.com/octopuscinema/libcamera
I think its not actually a typo, I think its intentionally different because there is already an RG16 fourcc in that drm_fourcc.h (DRM_FORMAT_RGB565).

Investigated a little further, there is some jiggery-pokery as part of the libcamera build process that generates a <libcamera/formats.h> file. This is generated from /src/libcamera/formats.yaml which contains the DRM_FORMAT based fourccs. These fourccs are what is used to create all the statically defined 'constexpr PixelFormat SRGGB...' in that generated file. Every libcamera based app will be using the autogenerated formats.h and internally libcamera uses 'PixelFormat' as if they match V4L2 fourccs, but they don't!

I'm not entire sure the best way to fix this, looks like it needs a redesign :/ And annoyingly its only 16-bit RAW (what I'm working on) which is broken because its the only one that doesn't match. I'll figure out a quick hacky solution so I can test 16-bit RAW data.

kbingham, 2023-07-21:
libcamera explicitly uses DRM style formats - not V4L2. Applications should not consider libcamera formats to be direct matches of V4L2 formats. We convert from DRM formats to v4l2 formats inside libcamera.

There's no redesign required - this is already an existing and *very specific* decision.

kbingham, 2023-07-21:
If you're working directly on libcamera development, please feel free to join https://matrix.to/#/#_oftc_#libcamera:matrix.org and/or the mailing list for direct discussions with the core team.

kbingham, 2023-07-21:
Excellent, I see you're using Soho cameras - Are you working to get those supported upstream?

octopus-russell, 2023-07-22:
Apologies, this might have been a red herring for us trying to debug getting 16-bit bayer RAW working. We are looking to support some newish Sony sensors produced by Soho yes. It looks like the 16-bit issue might be a lack of support in the raspberry pi unicam driver.

closed 2023-07-22 by octopus-russell