raspberrypi/libcamera #77: Noir V3 Wide V4L2 issues?

MattGeale, 2023-09-23:
Hi all,
I'm stumped and have no idea where to go from here.

I have a PI 4B 8gb, Noir v3 wide camera and a fresh install of Full Raspian 64bit. Whenever I try and run libcamera-still -o test.jpg or something along those lines, libcamera complains and throws out errors such as:
```
[0:18:18.676643661] [1827] ERROR V4L2 v4l2_device.cpp:353 'dw9807 10-000c': Unable to set controls: Remote I/O error
```
[multiple errors in sequence vs just a snippet](https://pastebin.com/gmvc1rXX) - ran on ssh so know it doesn't find an interface as indicated in the top.

I've done full upgrade, apt update, and also verified the camera is being acknowledged as when running the libcamera-hello --list-cameras I get: 
```
pi@raspberrypi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx708_wide_noir [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]

pi@raspberrypi:~ $
``` 
Libcamera-hello -t 0 when running the command via on the pi terminal (instead of ssh), shows the camera window briefly too per design of libcamera-hello, but those errors are still present.

If anyone could let me know what im doing wrong or if I'm missing some crucial part of configuring libcamera to play nice.

Thanks in advance!

MattGeale, 2023-09-25:
> This suggests that there is an I2C communication error trying to talk to the lens actuator device.  A few questions:
> 
> 
> 
> - Is this on a fresh installed Raspberry Pi OS image without any modifications?  Your description suggests yes, but just wanted to confirm.
> 
> - Has this only recently started happening, i.e. did the camera module ever work successfully in the past?
> 
> - Where did you purchase this module from?
> 
> - 

Hey,

In response to your questions:
1) yep. Fresh install. If by modifications you're meaning installs, the only things I've installed were PiJuice_CLI, Docker to run Restremer and then Raspystream (what is actually running the camera). I broke it down into various installs to ensure it wasn't a conflicting packages and just running Raspystream, and it still didn't work. The above was opened just with running Raspystream.
2) no it's never worked that I can tell.
3) From LittleBird Electronics in Australia however I have ordered another one thinking what I have could be a DOA.

naushir, 2023-09-25:
> 3\. From LittleBird Electronics in Australia however I have ordered another one thinking what I have could be a DOA.

This sounds like the most likely scenario.  I would suggest returning the camera board for a refund as it is most likely faulty.

MattGeale, 2023-09-25:
> > 3\. From LittleBird Electronics in Australia however I have ordered another one thinking what I have could be a DOA.
> 
> 
> 
> This sounds like the most likely scenario.  I would suggest returning the camera board for a refund as it is most likely faulty.

Copy thanks. 

In the interim to test my theory, is there a way to disable the autofocus? I want to text everything and double check my entire project works :P

naushir, 2023-09-25:
If you add

```
dtoverlay=imx708,vcm=0
```
to the bottom of ``/boot/config.txt`` and reboot, that should disable the actuator and avoid these errors.

MattGeale, 2023-09-25:
Hey, 

When I what you suggested at the bottom of config.txt, now the camera isn't even being detected...

Just to triple check everything and make sure there weren't any conflicting packages, I did another full reinstall of Rasbian.

```
pi@raspberrypi:~ $ libcamera-hello
Preview window unavailable
[0:00:45.451248212] [1123]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:00:45.470361712] [1124] ERROR V4L2 v4l2_device.cpp:93 'imx708_wide_noir': Failed to open V4L2 device '': No such file or directory
[0:00:45.470456453] [1124] ERROR RPI vc4.cpp:219 Failed to register camera imx708_wide_noir: -22
ERROR: *** no cameras available ***
pi@raspberrypi:~ $ libcamera-hello --list-cameras
No cameras available!
pi@raspberrypi:~ $
``` 

What do you think from here?

naushir, 2023-09-25:
Can you paste your output of the ``dmesg`` command?

6by9, 2023-09-25:
You'll need to comment out ```camera_auto_detect=1```, or set it to 0. Otherwise the firmware will be applying the overlay as well as your ```dtoverlay=``` line.

MattGeale, 2023-09-25:
> Can you paste your output of the `dmesg` command straight after a reboot?

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[    0.000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:45:4A:FE vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=fa4aa405-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7471352K/8245248K available (11776K kernel code, 2106K rwdata, 3688K rodata, 4160K init, 1077K bss, 249608K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 39744 entries in 156 pages
[    0.000000] ftrace: allocated 156 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000320] Console: colour dummy device 80x25
[    0.000384] printk: console [tty1] enabled
[    0.000451] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000477] pid_max: default: 32768 minimum: 301
[    0.000599] LSM: Security Framework initializing
[    0.000866] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[... 350 lines cut ...]
[    7.670507] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.835217] imx708 10-001a: camera module ID 0x0382
[    7.838350] imx708 10-001a: Consider updating driver imx708 to match on endpoints
[    8.382250] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   13.480481] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[... cut ...]
[   15.548313] Bluetooth: HCI UART protocol Broadcom registered
[   15.839443] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.839463] Bluetooth: BNEP filters: protocol multicast
[   15.839477] Bluetooth: BNEP socket layer initialized
[   15.845397] Bluetooth: MGMT ver 1.22
[   15.863518] NET: Registered PF_ALG protocol family
[   15.994658] Bluetooth: RFCOMM TTY layer initialized
[   15.994689] Bluetooth: RFCOMM socket layer initialized
[   15.994709] Bluetooth: RFCOMM ver 1.11
pi@raspberrypi:~ $
```

naushir, 2023-09-25:
> You'll need to comment out `camera_auto_detect=1`, or set it to 0. Otherwise the firmware will be applying the overlay as well as your `dtoverlay=` line.

Good catch, forgot about that!

MattGeale, 2023-09-25:
yep no errors now. Thanks guys. 

Will see how I go with regards to a replacement cam.

WIll keep this thread updated, but thanks for illustrating I'm not a complete idiot :P

🤝

naushir, 2023-10-05:
Closing now...

closed 2023-10-05 by naushir