raspberrypi/libcamera #80: Libcamera does time out when used with external Trigger Camera

DaniMambo, 2023-10-04:
Dear Libcamera Team,

I am quite new to the libcamera framework and am trying to make it work, for a camera we are trying to set up with the raspberry pi. The camera has two modes. Master Mode, where it automatically streams and return frames and Slave Mode, where it listens to external triggers and then returns frames.

The First Modus works perfectly, however if I want to use the External Trigger Mode, I always run into a Timeout (which is natural, when no trigger is applied). I would like to disable this timeout but cannot figure out, how. This is the error message I always get is:

0:07:31.783007701] [2531]  WARN V4L2 v4l2_videodevice.cpp:2007 /dev/video0[19:cap]: Dequeue timer of 2502000.00us has expired!
[0:07:31.783119349] [2531] ERROR RPI raspberrypi.cpp:1984 Unicam has timed out!
[0:07:31.783144108] [2531] ERROR RPI raspberrypi.cpp:1985 Please check that your camera sensor connector is attached securely.
[0:07:31.783167442] [2531] ERROR RPI raspberrypi.cpp:1986 Alternatively, try another cable and/or sensor.

The connection is fine, as the camera works fine in Master Mode streaming on it's own. Is there a way to avoid this timeout?

naushir, 2023-10-05:
You cannot completely disable the timeout, but you can override the value to something very very large.  To do this, open the pipeline config files at ``/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml`` (or at ``/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml`` if you built and installed libcamera manually), and add the following to the end of the parameter list:

```json
"camera_timeout_value_ms": 999999
```

If you are using ``libcamera-apps``, this will be picked up automatically.  If not, you will need to set the environment variable:
```bash
export LIBCAMERA_RPI_CONFIG_FILE=/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml
```
before running the application.

naushir, 2023-10-11:
If there are no further problems with this, I'll close it down shortly.

naushir, 2023-10-12:
Closing now, please reopen if you are still having problems.

closed 2023-10-12 by naushir