raspberrypi/libcamera #83: The problem that status_.targetExposureValue in agc_channel.cpp is always 0

tom-ard, 2023-11-10:
I found that the data in status_ are obtained from filtered, but targetExposureValue is obtained from target_.Is this intentional?
[image]

This will cause status_.targetExposureValue to always have a value of 0.
[image]

Since status_.targetExposureValue is always 0, lockCount has never been updated.
[image]

naushir, 2023-11-10:
This seems intentional to me.  @davidplowman can you confirm?

davidplowman, 2023-11-10:
Thanks for reporting this. Indeed I think it got broken when the filtering and digital gain calculations were re-ordered.

There's a fix [here](https://github.com/raspberrypi/libcamera/commit/f32ccfc419c471e18a181f1adb8a3b61eca872f6) on the next_integrate branch. I've also sent a patch to the libcamera mailing list.

There should probably be a Picamera2 test to check that 'AeLocked' does seem to get set eventually. Obviously not until this change has got merged, though!!

tom-ard, 2023-11-20:
I tested your modification and the problem was solved. Thanks.

naushir, 2023-11-23:
Changes should be merged soon.  Resolving now.

closed 2023-11-23 by naushir