raspberrypi/libcamera #85: Raspberry Pi 5 libx265 encode: Specified pixel format -1 is invalid or not supported

popsUlfr, 2023-11-17:
Hello,

I just got my Raspberry Pi 5 and I wanted to test video capturing.

Using libav codec to encode with `libx264` works but if I try with `libx265` it fails with `Specified pixel format -1 is invalid or not supported`. Is there a way to specify the pixel format ?

```
libcamera-vid -t 10000 --width 1280 --height 720 --framerate 30 --tuning-file /usr/share/libcamera/ipa/rpi/pisp/imx708_wide.json --codec libav --libav-video-codec libx265 -o video.ts
```

```
WARNING: v3d support for hw version 71 is neither a complete nor a conformant OpenGL implementation. Testing use only.
Made X/EGL preview window
[0:57:40.305459331] [3752]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:57:40.340581155] [3759]  INFO RPI pisp.cpp:652 libpisp version v1.0.0 ce5624ed8726 23-10-2023 (07:41:40)
[0:57:40.365558789] [3759]  INFO RPI pisp.cpp:1106 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:57:40.366798315] [3752]  WARN V4L2 v4l2_pixelformat.cpp:336 Unsupported V4L2 pixel format RPBP
Mode selection for 1280:720:12:P(30)
    SRGGB10_CSI2P,1536x864/120.135 - Score: 1100
    SRGGB10_CSI2P,2304x1296/56.0255 - Score: 1400
    SRGGB10_CSI2P,4608x2592/14.3536 - Score: 33592.8
Stream configuration adjusted
[0:57:40.367086891] [3752]  INFO Camera camera.cpp:1181 configuring streams: (0) 1280x720-YUV420 (1) 1536x864-BGGR16_PISP_COMP1
[0:57:40.367162484] [3759]  INFO RPI pisp.cpp:1390 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B
[libx265 @ 0x5555cf4d40e0] Specified pixel format -1 is invalid or not supported
ERROR: *** libav: unable to open video codec: -22 ***
```

naushir, 2023-11-17:
``libx265`` is not supported through ``rpicam-vid``.  We did look at support a while back, but the encode performance requirements are way too large for this platform right now.

If you do want to experiment with it, you need to add the relevant codec target here:
https://github.com/raspberrypi/rpicam-apps/blob/dd744cb385e2b455f76f9a48106606fb9a160f21/encoder/libav_encoder.cpp#L57

popsUlfr, 2023-11-17:
> `libx265` is not supported through `rpicam-vid`. We did look at support a while back, but the encode performance requirements are way too large for this platform right now.
>
> If you do want to [...]

Thank you very much for the info!

The help for the `--libav-video-codec` option is a bit confusing in this regard:

```
  --libav-video-codec arg (=h264_v4l2m2m)
                                        Sets the libav video codec to use. To list available codecs, run  the "ffmpeg 
                                        -codecs" command.
```

It makes it sound like one can pick whatever codec ffmpeg supports. But actually there's no choice at all on the Pi 5 if I understand this right (`h264_v4l2m2m` is not available so only `libx264` is suported).

naushir, 2023-11-20:
I can add a small change to open up the other codecs, but I will not be able to provide optimised config for all possible codecs like I did for libx264.  This will (in theory) at least allow you to test alternate codecs.

popsUlfr, 2023-11-20:
Ah that's very neat thank you! I'll see if I maybe manage to add an additional command line argument to be able to specify ffmpeg codec options.

naushir, 2023-11-20:
> Ah that's very neat thank you! I'll see if I maybe manage to add an additional command line argument to be able to specify ffmpeg codec options.

That might be useful, but tricky to get right in the generic case.