raspberrypi/libcamera #87: OV5647 image is blue

tom-ard, 2023-11-20:
Run OV5647 on PI5. There is a chance that the image is blue.
[image]

There are many "nan" in LOG, and I guess this may be the reason why the image becomes blue.
```
[0:04:03.319676807] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.12962 b 0.428575: nan
[0:04:03.319689898] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.13486 b 0.43706: nan
[0:04:03.319702654] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.1401 b 0.445546: nan
[0:04:03.319715950] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.14534 b 0.454031: nan
[0:04:03.319728724] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.15057 b 0.462517: nan
[0:04:03.319742001] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.15581 b 0.471003: nan
[0:04:03.319754925] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.16105 b 0.479488: nan
[0:04:03.319767829] [2462] [1;36mDEBUG [1;37mRPiAwb [1;34mawb.cpp:605 [0mAt t 2478.6 r 1.16629 b 0.487974: nan
```

Complete log
[ov5647.log](https://github.com/raspberrypi/libcamera/files/13407407/ov5647.log)

naushir, 2023-11-20:
@davidplowman for info.

What type of lighting is illuminating the scene here?  Is the color temperature actually 2400K as reported by the AWB algorithm?  Would you be able to try this out with another Raspberry Pi camera and see if you get the same results?

davidplowman, 2023-11-20:
Indeed, the NaNs certainly look bad, and may be why the search remains stuck at the the wrong end of the CT curve. 

Having fired up an ov5647 for testing, and having a little look around the code, I'm not really seeing how this can come about. In very low light, if it finds no statistics at all that it will trust, it simply resorts to grey world, which we aren't seeing in the log. Otherwise it will have statistics regions that it trusts, at which point I don't quite get where the NaNs come from. Are you in a position to put any extra debug into the code, or could you tell us more about this scenario so that we can try again to reproduce it?

In the meantime, there are probably workarounds such as using explicit colour gains, or placing the algorithm into grey world mode, but it does seem like there's something more to look into here. Thanks.

6by9, 2023-11-20:
Stupid thought - it's not the protective blue piece of cellophane that is stuck to the lens for production?

tom-ard, 2023-11-21:
In fact, image anomalies are only accidental phenomena, and most of the time the images are normal.
[image]

Log information under normal circumstances
[ov5647_normal.log](https://github.com/raspberrypi/libcamera/files/13420727/ov5647_normal.log)

@naushir This is an indoor environment and I'm not sure about the actual color temperature. I tried running on another PI5 and replaced multiple cameras. The same problem occurred.

I traced it back to the "constructM" function by printing the log.It is found that "diagonal" may be equal to 0, resulting in a divide-by-zero exception.
[image]

```
C = 0.000000
W[0] = 0.000000 W[1] = 0.000000 W[2] = 0.000000 W[3] = 0.000000
diagonal: 0.000000
M[0] = nan M[1] = inf M[2] = nan M[3] = nan
```

Complete log
[ov5647_constructM.log](https://github.com/raspberrypi/libcamera/files/13422705/ov5647_constructM.log)

Because I don't know enough about the algorithm meaning. I can only trace it so far.

davidplowman, 2023-11-21:
Thanks for that log. That's all very interesting. To me it suggests the `calculateCrCb` is generating zero `cr` and `cb` values. Indeed this looks possible. There's a few things we could do. Are you able to find the line in there that says
```
		if (s.counted <= minCount || s.val.gSum / s.counted <= minG) {
```
and then let's change it to
```
		if (s.counted <= minCount || s.val.gSum / s.counted <= minG ||
			s.val.rSum / s.counted <= minG  || s.val.bSum / s.counted <= minG) {
```
If that works for you then I can submit that as a libcamera patch. Thanks!

tom-ard, 2023-11-22:
This problem has been solved according to your modifications, thank you.
[image]

And I printed out the data information in "awbRegion" and got this set of data:
```
rSum: 0 bSum: 0 gSum: 16704 counted: 300
```
This means that the previous judgment conditions were not sufficient enough.

davidplowman, 2023-11-22:
I've posted that fix to the libcamera mailing list so hopefully it will filter through "in due course". Thanks again for your work on this!

naushir, 2023-11-23:
Changes should be merged soon.  I'll resolve this now.

closed 2023-11-23 by naushir