raspberrypi/libcamera #88: add new Control

Lusifug, 2023-11-20:
Is there a way to add new values to ControlInfoMap for Camera?
I have added new controls to control_ids and ipa_base, but when I try to use them, I receive an error message:
_ERROR Controls controls.cpp:1069 Control 0x0000000c is not valid for /base/soc/i2c0mux/i2c@1/imx708@1a_
According to the validation method, it seems like this happens cause Camera ControlInfoMap doesn`t have such controls

naushir, 2023-11-20:
Can you expand on what you are trying to do here?  The control error above is for a V4L2 control error, not a libcamera control.

Lusifug, 2023-11-21:
I`m trying to add a possibility to pass value via controls, to change params for existing contrast algorithm on the fly, such as low level, low histogram...

naushir, 2023-11-21:
Unfortunately I am still not sure what you are trying to do. Would you be able to share snippets of source code showing you setting up this new control and how use use it please?

Lusifug, 2023-11-21:
I`ve added setters to [Contrast](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/controller/rpi/contrast.h) class:
```
	void setConfigLoHistogram(double loHistogram) override;
	void setConfigLoLevel(double loLevel) override;
	void setConfigLoMax(double loMax) override;
	void setConfigHiHistogram(double hiHistogram) override;
	void setConfigHiLevel(double hiLevel) override;
	void setConfigHiMax(double hiMax) override;
```
and interface to [ContrastAlgorithm](https://github.com/raspberrypi/libcamera/blob/release-v0.0.5%2B83-bde9b04f/src/ipa/rpi/controller/contrast_algorithm.h) class:
```
	virtual void setConfigLoHistogram(double loHistogram) = 0;
	virtual void setConfigLoLevel(double loLevel) = 0;
	virtual void setConfigLoMax(double loMax) = 0;
	virtual void setConfigHiHistogram(double hiHistogram) = 0;
	virtual void setConfigHiLevel(double hiLevel) = 0;
	virtual void setConfigHiMax(double hiMax) = 0;
```

in [IpaBase](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/common/ipa_base.cpp) method applyControls I`ve added cases for this new controls:
```
case controls::CONTRAST_LO_HISTOGRAM: {
			RPiController::ContrastAlgorithm *contrast = dynamic_cast<RPiController::ContrastAlgorithm *>(
				controller_.getAlgorithm("contrast"));
			if (!contrast) {
				LOG(IPARPI, Warning)
					<< "Could not set CONTRAST_LO_HISTOGRAM - no contrast algorithm";
				break;
			}

			contrast->setConfigLoHistogram(ctrl.second.get<float>());
			libcameraMetadata_.set(controls::ContrastLoHistogram,
					       ctrl.second.get<float>());
			break;
		}

		case controls::CONTRAST_LO_LEVEL: {
			RPiController::ContrastAlgorithm *contrast = dynamic_cast<RPiController::ContrastAlgorithm *>(
				controller_.getAlgorithm("contrast"));
			if (!contrast) {
				LOG(IPARPI, Warning)
					<< "Could not set CONTRAST_LO_LEVEL - no contrast algorithm";
				break;
			}

			contrast->setConfigLoLevel(ctrl.second.get<float>());
			libcameraMetadata_.set(controls::ContrastLoLevel,
					       ctrl.second.get<float>());
			break;
		}

		case controls::CONTRAST_LO_MAX: {
			RPiController::ContrastAlgorithm *contrast = dynamic_cast<RPiController::ContrastAlgorithm *>(
				controller_.getAlgorithm("contrast"));
			if (!contrast) {
				LOG(IPARPI, Warning)
					<< "Could not set CONTRAST_LO_MAX - no contrast algorithm";
				break;
			}

			contrast->setConfigLoMax(ctrl.second.get<float>());
			libcameraMetadata_.set(controls::ContrastLoMax,
					       ctrl.second.get<float>());
			break;
		}

		case controls::CONTRAST_HI_HISTOGRAM: {
			RPiController::ContrastAlgorithm *contrast = dynamic_cast<RPiController::ContrastAlgorithm *>(
				controller_.getAlgorithm("contrast"));
			if (!contrast) {
				LOG(IPARPI, Warning)
					<< "Could not set CONTRAST_HI_HISTOGRAM - no contrast algorithm";
				break;
			}

			contrast->setConfigHiHistogram(ctrl.second.get<float>());
			libcameraMetadata_.set(controls::ContrastHiHistogram,
					       ctrl.second.get<float>());
			break;
		}

		case controls::CONTRAST_HI_LEVEL: {
			RPiController::ContrastAlgorithm *contrast = dynamic_cast<RPiController::ContrastAlgorithm *>(
				controller_.getAlgorithm("contrast"));
			if (!contrast) {
				LOG(IPARPI, Warning)
					<< "Could not set CONTRAST_HI_LEVEL - no contrast algorithm";
				break;
			}

			contrast->setConfigHiLevel(ctrl.second.get<float>());
			libcameraMetadata_.set(controls::ContrastHiLevel,
					       ctrl.second.get<float>());
			break;
		}

		case controls::CONTRAST_HI_MAX: {
			RPiController::ContrastAlgorithm *contrast = dynamic_cast<RPiController::ContrastAlgorithm *>(
				controller_.getAlgorithm("contrast"));
			if (!contrast) {
				LOG(IPARPI, Warning)
					<< "Could not set CONTRAST_HI_MAX - no contrast algorithm";
				break;
			}

			contrast->setConfigHiMax(ctrl.second.get<float>());
			libcameraMetadata_.set(controls::ContrastHiMax,
					       ctrl.second.get<float>());
			break;
		}
```

and in [control_ids.yaml](https://github.com/raspberrypi/libcamera/blob/main/src/libcamera/control_ids.yaml) I`ve added new values for new controls:
```
  - ContrastLoHistogram:
      type: float
      description:  |
        Specify a low end higher quantile value.

  - ContrastLoLevel:
      type: float
      description:  |
        Specify a lower level in the output range.

  - ContrastLoMax:
      type: float
      description:  |
        Specify a low end desired adjustment value.

  - ContrastHiHistogram:
      type: float
      description:  |
        Specify a high end higher quantile value.

  - ContrastHiLevel:
      type: float
      description:  |
        Specify a high end desired adjustment value.

  - ContrastHiMax:
      type: float
      description:  |
        Specify a high end desired adjustment value.
```

naushir, 2023-11-21:
That all seems reasonable enough.  Have you definitely re-compiled all of libcamera and your applications to make sure the new control values are addressed correctly?

On a side note, perhaps you don't need to do this and can adjust the histogram through the camera tuning file instead?

Lusifug, 2023-11-21:
for compiling I used these commands, as described in the manual
```bash
meson setup --prefix=/usr build
sudo ninja -C build install
```

changes in the camera tuning file are not possible in real-time, cause it requires to restart camera app, which takes time. correct me if I`m wrong

naushir, 2023-11-21:
Once you rebuild libcamera, you must rebuild libcamera-apps (or whatever application you are running) otherwise the control enum values will be mismatched between libcamera and the application.

You are correct, we cannot change the tuning file parameters in real-time.

Lusifug, 2023-11-21:
I use my own simple testing app, to take pictures via libcamera, i rebuild before every use:
```c++
#include <iostream>
#include <iomanip>
#include <memory>
#include <thread>
#include <fstream>
#include <fcntl.h>
#include <sys/mman.h>
#include <libcamera/libcamera.h>

using namespace libcamera;
using namespace std::chrono_literals;

static void requestComplete(Request *request)
{
    if (request->status() == Request::RequestCancelled)
    {
        return;
    }

    const std::map<const Stream *, FrameBuffer *> &buffers = request->buffers();
    for (auto bufferPair : buffers) {
        FrameBuffer *buffer = bufferPair.second;

        const libcamera::FrameBuffer::Plane &plane = buffer->planes().front();
        void *frameData = mmap(nullptr, plane.length, PROT_READ, MAP_SHARED, plane.fd.get(), 0);

        std::string photoFilename = "captured_photo_final.ppm";
        std::ofstream photoFile(photoFilename, std::ios::out | std::ios::binary);

        photoFile << "P6\n";
        photoFile << "1920 1080" << "\n";
        photoFile << "255\n";
        photoFile.write(reinterpret_cast<const char *>(frameData), plane.length);
        photoFile.close();
    }
}

int main()
{
    std::unique_ptr<CameraManager> cm = std::make_unique<CameraManager>();
    cm->start();
    static std::shared_ptr<Camera> camera = cm->cameras()[0];
    camera->acquire();

    std::unique_ptr<CameraConfiguration> config = camera->generateConfiguration( { StreamRole::Raw } );
    StreamConfiguration &streamConfig = config->at(0);
    streamConfig.pixelFormat = libcamera::formats::BGR888;
    streamConfig.size = libcamera::Size(1920, 1080);
    config->validate();
    camera->configure(config.get());

    FrameBufferAllocator *allocator = new FrameBufferAllocator(camera);

    int ret = allocator->allocate(streamConfig.stream());
    if (ret < 0) {
        std::cerr << "Can't allocate buffers" << std::endl;
        return -ENOMEM;
    }

    Stream *stream = streamConfig.stream();
    const std::unique_ptr<FrameBuffer> &buffer = allocator->buffers(stream)[0];

    std::unique_ptr<Request> request = camera->createRequest();
    if (!request)
    {
        std::cerr << "Can't create request" << std::endl;
        return -ENOMEM;
    }

    ret = request->addBuffer(stream, buffer.get());
    if (ret < 0)
    {
        std::cerr << "Can't set buffer for request"
              << std::endl;
        return ret;
    }

    camera->requestCompleted.connect(requestComplete);

    camera->start();
    ControlList &reqcon = request.get()->controls();
    reqcon.set(controls::ContrastLoHistogram, 0.0);
	reqcon.set(controls::ContrastLoLevel, 0.1);
	reqcon.set(controls::ContrastLoMax, 20000);
	reqcon.set(controls::ContrastHiLevel, 1.0);
	reqcon.set(controls::ContrastHiHistogram, 0.9);
	reqcon.set(controls::ContrastHiMax, 10000);
    camera->queueRequest(request.get());
    std::this_thread::sleep_for(1000ms);
    camera->stop();
    allocator->free(stream);
    delete allocator;
    camera->release();
    camera.reset();
    cm->stop();

    return 0;
}
```
when I set standard controls, it works perfectly, but when I`m trying same with new controls, I receive error, and even if disable this error, nothing changes on the photo

naushir, 2023-11-21:
Are you able to run under gdb and put a breakpoint on the line that caused the error and share a callstack?

Lusifug, 2023-11-21:
Thread 2 "capture" hit Breakpoint 2, libcamera::ControlList::find (this=0x7ff65ffbd8, id=9963796) at ../src/libcamera/controls.cpp:1070
1070		if (validator_ && !validator_->validate(id)) {
(gdb) backtrace
#0  libcamera::ControlList::find (this=0x7ff65ffbd8, id=9963796) at ../src/libcamera/controls.cpp:1070
#1  0x0000007ff7d3e4c0 in libcamera::ControlList::set (this=0x7ff65ffbd8, id=9963796, value=...) at ../src/libcamera/controls.cpp:1030
#2  0x0000007ff7d22ba4 in libcamera::CameraSensor::init (this=0x7ff0003eb0) at ../src/libcamera/camera_sensor.cpp:117
#3  0x0000007ff7e25ad0 in libcamera::RPi::PipelineHandlerBase::registerCamera (this=0x7ff000a780, cameraData=std::unique_ptr<libcamera::RPi::CameraData> = {...}, frontend=0x7ff000dfd0, frontendName="unicam-image", backend=0x7ff0008e50, 
    sensorEntity=0x7ff000a930) at ../src/libcamera/pipeline/rpi/common/pipeline_base.cpp:804
#4  0x0000007ff7e3dd54 in libcamera::PipelineHandlerVc4::match (this=0x7ff000a780, enumerator=0x7ff0000dd0) at ../src/libcamera/pipeline/rpi/vc4/vc4.cpp:215
#5  0x0000007ff7d1b08c in libcamera::CameraManager::Private::createPipelineHandlers (this=0x55555b21b0) at ../src/libcamera/camera_manager.cpp:122
#6  0x0000007ff7d1af20 in libcamera::CameraManager::Private::init (this=0x55555b21b0) at ../src/libcamera/camera_manager.cpp:94
#7  0x0000007ff7d1adf0 in libcamera::CameraManager::Private::run (this=0x55555b21b0) at ../src/libcamera/camera_manager.cpp:71
#8  0x0000007ff787c07c in libcamera::Thread::startThread (this=0x55555b21c0) at ../src/libcamera/base/thread.cpp:319
#9  0x0000007ff78800b4 in std::__invoke_impl<void, void (libcamera::Thread::*)(), libcamera::Thread*> (__f=@0x55555b37a0: (void (libcamera::Thread::*)(libcamera::Thread * const)) 0x7ff787bf7c <libcamera::Thread::startThread()>, 
    __t=@0x55555b3798: 0x55555b21c0) at /usr/include/c++/10/bits/invoke.h:73
#10 0x0000007ff787ffcc in std::__invoke<void (libcamera::Thread::*)(), libcamera::Thread*> (__fn=@0x55555b37a0: (void (libcamera::Thread::*)(libcamera::Thread * const)) 0x7ff787bf7c <libcamera::Thread::startThread()>)
    at /usr/include/c++/10/bits/invoke.h:95
#11 0x0000007ff787ff30 in std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> >::_M_invoke<0ul, 1ul> (this=0x55555b3798) at /usr/include/c++/10/thread:264
#12 0x0000007ff787fee8 in std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> >::operator() (this=0x55555b3798) at /usr/include/c++/10/thread:271
#13 0x0000007ff787fec8 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> > >::_M_run (this=0x55555b3790) at /usr/include/c++/10/thread:215
#14 0x0000007ff76cdcac in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#15 0x0000007ff77e3648 in start_thread (arg=0x7ff6601a40) at pthread_create.c:477
#16 0x0000007ff753dfdc in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78

Lusifug, 2023-11-21:
not sure what do you mean by frame 1 and frame 2
this onecontrol - reqcon.set(controls::ContrastLoHistogram, 0.0);

naushir, 2023-11-21:
In gdb, when you hit the error, type `frame 1` and look at the output line, then do the same for `frame 2`.

Lusifug, 2023-11-21:
(gdb) frame 1
#1  0x0000007ff7d3e450 in libcamera::ControlList::get (this=0x7ff65ffbd8, id=9963796) at ../src/libcamera/controls.cpp:1009
1009		const ControlValue *val = find(id);
(gdb) frame 2
#2  0x0000007ff7d93b7c in libcamera::V4L2Device::updateControls (this=0x7ff000e0b0, ctrls=0x7ff65ffbd8, v4l2Ctrls=...) at ../src/libcamera/v4l2_device.cpp:708
708			ControlValue value = ctrls->get(id);

naushir, 2023-11-21:
If you have a branch that I could try out (libcamera + application) perhaps I can give it a try and see if I can help.  Otherwise it's going to be impossible for me to say what could be going wrong with your changes.

Lusifug, 2023-11-21:
I`ve created repo, with my changes, my test program and make file for it is inside "test app" dir
https://github.com/Lusifug/libcamera/tree/CE_controls

naushir, 2023-11-21:
I'm afraid I get compile errors with that tree:
```make
[40/200] Generating src/ipa-priv-key with a custom command
....+....+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*...+..+.+..+...+.......+...+.....+.+.....+....+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*...+......+......+...............+....+.........+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
..............+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*...+.......+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++*.............+...+..+...+.......+..+.............+........+................+........+.......+......+..+...+..........+..+...............+............+.+.....+....+.........+..+.............+...........+...+......+.+........+.+..+.......+..+.+...+......+.........+.........+.....+...+...+......+.............+............+..............+.+...+...+.....+.......+..+.........+.........+.....................+......+....+..+.......+..+....+...........+...+.+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
[42/200] Generating src/libcamera/ipa_pub_key_cpp with a custom command
writing RSA key
[43/200] Generating src/libcamera/control_ids_cpp with a custom command
FAILED: src/libcamera/control_ids.cpp
/home/pi/libcamera-lusifug/utils/gen-controls.py -o src/libcamera/control_ids.cpp ../src/libcamera/control_ids.yaml ../src/libcamera/control_ids.cpp.in
Traceback (most recent call last):
  File "/home/pi/libcamera-lusifug/utils/gen-controls.py", line 313, in <module>
    sys.exit(main(sys.argv))
             ^^^^^^^^^^^^^^
  File "/home/pi/libcamera-lusifug/utils/gen-controls.py", line 294, in main
    data = generate_cpp(controls)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/libcamera-lusifug/utils/gen-controls.py", line 160, in generate_cpp
    'description': format_description(ctrl.description),
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/libcamera-lusifug/utils/gen-controls.py", line 125, in format_description
    description = description.strip('\n').split('\n')
                  ^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'strip'
[48/200] Generating include/libcamera/ipa/core_ipa_serializer_h with a custom command
ninja: build stopped: subcommand failed.
```

Lusifug, 2023-11-22:
```
pi@sb-dev:~$ git clone https://github.com/Lusifug/libcamera.git
Cloning into 'libcamera'...
remote: Enumerating objects: 39341, done.
remote: Counting objects: 100% (6231/6231), done.
remote: Compressing objects: 100% (1432/1432), done.
remote: Total 39341 (delta 4971), reused 5923 (delta 4792), pack-reused 33110
Receiving objects: 100% (39341/39341), 8.11 MiB | 873.00 KiB/s, done.
Resolving deltas: 100% (30349/30349), done.
pi@sb-dev:~$ cd libcamera
pi@sb-dev:~/libcamera$ meson setup --prefix=/usr build
The Meson build system
Version: 1.0.0
Source dir: /home/pi/libcamera
Build dir: /home/pi/libcamera/build
Build type: native build
Project name: libcamera
Project version: 0.1.0
C compiler for the host machine: cc (gcc 10.2.1 "cc (Debian 10.2.1-6) 10.2.1 20210110")
C linker for the host machine: cc ld.bfd 2.35.2
C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Debian 10.2.1-6) 10.2.1 20210110")
C++ linker for the host machine: c++ ld.bfd 2.35.2
Host machine cpu family: aarch64
Host machine cpu: aarch64
Header "unistd.h" has symbol "issetugid" : NO 
Header "locale.h" has symbol "locale_t" : YES 
Header "stdlib.h" has symbol "secure_getenv" : YES 
Compiler for C supports arguments -Wno-c99-designator: NO 
Found pkg-config: /usr/bin/pkg-config (0.29.2)
Found CMake: /usr/bin/cmake (3.18.4)
Run-time dependency lttng-ust found: NO (tried pkgconfig and cmake)
Program ./parser.py found: YES (/home/pi/libcamera/utils/ipc/./parser.py)
Program ./generate.py found: YES (/home/pi/libcamera/utils/ipc/./generate.py)
Program ./extract-docs.py found: YES (/home/pi/libcamera/utils/ipc/./extract-docs.py)
Program ./gen-tp-header.py found: YES (/home/pi/libcamera/utils/tracepoints/./gen-tp-header.py)
Configuring version.h using configuration
Program openssl found: YES (/usr/bin/openssl)
Library atomic found: YES
Run-time dependency threads found: YES
Run-time dependency libdw found: YES 0.183
Run-time dependency libunwind found: YES 1.3.2
Header "execinfo.h" has symbol "backtrace" : YES 
Library rt found: YES
Run-time dependency libpisp found: NO (tried pkgconfig and cmake)
Looking for a fallback subproject for the dependency libpisp
Cloning into 'libpisp'...
remote: Enumerating objects: 82, done.
remote: Counting objects: 100% (82/82), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 82 (delta 5), reused 41 (delta 5), pack-reused 0
Receiving objects: 100% (82/82), 80.76 KiB | 1.12 MiB/s, done.
Resolving deltas: 100% (5/5), done.

Executing subproject libpisp 

libpisp| Project name: libpisp
libpisp| Project version: 1.0.1
libpisp| C compiler for the host machine: cc (gcc 10.2.1 "cc (Debian 10.2.1-6) 10.2.1 20210110")
libpisp| C linker for the host machine: cc ld.bfd 2.35.2
libpisp| C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Debian 10.2.1-6) 10.2.1 20210110")
libpisp| C++ linker for the host machine: c++ ld.bfd 2.35.2
libpisp| Configuring pisp_build_config.h using configuration
libpisp| Run-time dependency nlohmann_json found: NO (tried pkgconfig and cmake)
libpisp| Looking for a fallback subproject for the dependency nlohmann_json
libpisp| Using subprojects/libpisp/subprojects/nlohmann_json.wrap
libpisp| Downloading nlohmann_json source from https://github.com/nlohmann/json/releases/download/v3.11.2/include.zip
Download size: 293810
Downloading: ..........

Executing subproject libpisp:nlohmann_json
[... 251 lines cut ...]
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx219_noir.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx290.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx296.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx296_mono.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx378.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx477.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx477_noir.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx477_scientific.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx519.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx708.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx708_noir.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx708_wide.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/imx708_wide_noir.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/ov5647.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/ov5647_noir.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/ov9281_mono.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/se327m12.json to /usr/share/libcamera/ipa/rpi/vc4
Installing /home/pi/libcamera/src/ipa/rpi/vc4/data/uncalibrated.json to /usr/share/libcamera/ipa/rpi/vc4
Installing symlink pointing to libcamera-base.so.0.1.0 to /usr/lib/aarch64-linux-gnu/libcamera-base.so.0.1
Installing symlink pointing to libcamera-base.so.0.1 to /usr/lib/aarch64-linux-gnu/libcamera-base.so
[... cut ...] 0 to /usr/lib/aarch64-linux-gnu/libpisp.so
Installing symlink pointing to libcamera.so.0.1.0 to /usr/lib/aarch64-linux-gnu/libcamera.so.0.1
Installing symlink pointing to libcamera.so.0.1 to /usr/lib/aarch64-linux-gnu/libcamera.so
Running custom install script '/home/pi/libcamera/src/ipa/ipa-sign-install.sh /home/pi/libcamera/build/src/ipa-priv-key.pem lib/aarch64-linux-gnu/libcamera/ipa_rkisp1.so lib/aarch64-linux-gnu/libcamera/ipa_rpi_pisp.so lib/aarch64-linux-gnu/libcamera/ipa_rpi_vc4.so'
```

naushir, 2023-11-22:
Are you using the [CE_controls](https://github.com/Lusifug/libcamera/tree/CE_controls) branch or the main branch for your build above?  From what I gather, the main branch does not have your new controls.

naushir, 2023-11-30:
Any updates on this?  If not, I'll close it down shortly.

closed 2024-01-03 by naushir