raspberrypi/libcamera #90: Expose histogram statistics ( and other statistics ) to higher level apps ( i.e. rpicam-apps )

schoolpost, 2023-11-21:
https://github.com/raspberrypi/libcamera/blob/99b177b3e5a50093f664eacd6b09576d17a7e64d/src/ipa/rpi/pisp/pisp.cpp#L451C2-L451C2

Notice this comment mentioning that histogram values are not populated? How can this be captured?

Is it possible to interface with the underlying V4L2 device to obtain this histogram statistics?

I have brought this up on the forums some time ago, but given the new ISP in the Pi5 I'm very much hoping these stats will be available for end-user consumption: https://forums.raspberrypi.com/viewtopic.php?p=2071546#p2071546

naushir, 2023-11-21:
The histogram data lives within the IPA for use with the 3A algorithms, so is not exported out.

However, this is something we can possibly do in the near future.  I am working on a "vendor controls" change to add to libcamera.  This will allow us to provide custom structures out with the frame data.  One of these could be the AE histogram generated by the ISP.

naushir, 2023-11-23:
@schoolpost - just to satisfy our curiosity, can you expand on what you would use the histograms for?

schoolpost, 2023-11-23:
> @schoolpost - just to satisfy our curiosity, can you expand on what you would use the histograms for?

Absolutely, mainly as a visual monitor / assist tool for evaluating scene exposure in real time. ( video application ) 

It would for my application be of paramount importance that these statistics were gathered from the CFE from the bayer pixels before the other processing steps begin to skew / alter the pixels values. 

Two ways I would like to use the histogram data:
1. A liveview traditional histogram ( specifically a histogram that tracks per channel statistics, at a bayer level. ) 
[image]
[image]

2. An extension of the traditional histogram, using something known as a "traffic light" system that gives a very nice at a glance information about whether there is channel clipping occuring in either the top end ( highlights ) or the bottom ( shadows ) and generally where the exposure stands: 
[image]

Any permutation of this style of exposure monitoring, which many of these can be displayed to the user in real-time with some fairly rudimentary opengl shaders or other gui toolkits.

naushir, 2023-11-23:
Thanks!  One thing to note - you will only get the luminance/Y histogram from the Pi 5 output.  This is all we use for AGC.  It is possible to get R/G/B histograms, but they will have to be frame-interleaved and to be honest we don't have plans to do that right now.

schoolpost, 2023-11-23:
> Thanks!  One thing to note - you will only get the luminance/Y histogram from the Pi 5 output.  This is all we use for AGC.  It is possible to get R/G/B histograms, but they will have to be frame-interleaved and to be honest we don't have plans to do that right now.

I'd be happy with luma histogram data for the time being.

Too bad R/G/B histogram data is not available as it was in Pi4, it would be greatly appreciated if it could be added at least at some point! 🙂

kbingham, 2023-11-23:
> However, this is something we can possibly do in the near future. I am working on a "vendor controls" change to add to libcamera. This will allow us to provide custom structures out with the frame data. One of these could be the AE histogram generated by the ISP.

I'd still love for these to be generic (but a vendor control could always be a first implementation).
Are histograms really so difficult to describe generically? (I guess the bin sizes/ranges are platform specific?)

Sounds like we can expose different 'types' of histogram in the metadata - but for each one ... I would hope we could have a way to describe a histogram quite generically? Then there's just sets of histogram or optionally provided histogram for luminance/r/g/b channels?

naushir, 2023-11-23:
> I'd still love for these to be generic (but a vendor control could always be a first implementation). Are histograms really so difficult to describe generically? (I guess the bin sizes/ranges are platform specific?)

https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/controller/histogram.h

> 
> Sounds like we can expose different 'types' of histogram in the metadata - but for each one ... I would hope we could have a way to describe a histogram quite generically? Then there's just sets of histogram or optionally provided histogram for luminance/r/g/b channels?

https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/controller/statistics.h

😄 

But these are not exposed at the public api level yet.

6by9, 2023-11-23:
Memory says that our old friend the Android Camera3 HAL had metadata tags for histograms.

pinchartl, 2023-11-23:
@kbingham Have you forgotten how hardware designers can be "create" and "inventive" ? :-)

kbingham, 2023-11-23:
Sure ... But a histogram ... Is a sequence of numbers ...

I haven't yet seen why this is hard to store.
Harder to describe maybe :-)

pinchartl, 2023-11-23:
Everything in a computer is a sequence of numbers :-)

kbingham, 2023-11-23:
Exactly ;-)

naushir, 2023-12-01:
I've had a quick go at trying to implement statistics output using the newly integrated vendor control changes.  You can have a look at the change here: https://github.com/raspberrypi/libcamera/commit/88eb9123c6df03d1d82a65a0abbb3acf2b046f01.  Note this is entirely untested!

schoolpost, 2023-12-01:
> I've had a quick go at trying to implement statistics output using the newly integrated vendor control changes. You can have a look at the change here: [d758394](https://github.com/raspberrypi/libcamera/commit/d75839447c0d20daa5dc7a9178877022e0ca0149) / [2c85cc0](https://github.com/raspberrypi/libcamera/commit/2c85cc01856923ebe5f9fc043291d6d436dd7504). Note this is entirely untested!
> 
> EDIT: This is only a proof of concept, the controls will definitely change names/functionality in the future.

I will test this soon! thanks

schoolpost, 2023-12-03:
Here's a look at a quick demo / prototype of a real-time histogram overlay using the agc_statistics. 

https://github.com/raspberrypi/libcamera/assets/25234407/dd2976eb-5b14-444c-939b-f18848ed473b

I may need some clarification on how the actual histogram pixel sums relate to the image, I get 1252900 pixel sum across the entire histogram, but I'm, not getting a clear idea how this is scaled/derived from the HQ camera sensor frame dimensions, 2028x1520 == 3082560 pixels...

naushir, 2023-12-04:
You can find all details at https://datasheets.raspberrypi.com/camera/raspberry-pi-image-signal-processor-specification.pdf for the stats specification.

Regarding the pixel counts, each pixel quad gives us a single Y value for the histogram count, so you start by dividing the pixel count by 4.  Additionally, the histogram counts are weighted by 16x16 regions across the image, hence the mismatch in counts.

naushir, 2024-01-03:
This is now available on the next branch.  Eventually we will include it in the official release.