raspberrypi/libcamera #92: No cameras available! after libcamera inslatation

miguelwon, 2023-11-29:
I'm testing the Raspberry 5 with Bookworm as OS.

By default Bookworm already comes with libcamera.

The default libcamera is able to find the camera and take some frames. 

The problem is when I build libcamera from source and install it. After the instalation no cameras are found anymore:

```
pi@raspberrypi:~ $ libcamera-still --list-cameras
No cameras available!
```

I follow a simple and common instalation, just like readme instructions:

```
git clone https://git.libcamera.org/libcamera/libcamera.git
cd libcamera
meson setup build
ninja -C build install 
```

I tried both with RPI camera v2 and v3.  My system: 

```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

kbingham, 2023-11-29:
If you've cloned here:
```
git clone https://git.libcamera.org/libcamera/libcamera.git
```
Then you won't currently get Raspberry Pi 5 support. This is still being developed, and is not yet upstream in the libcamera project.

For now - please clone from this repository (https://github.com/raspberrypi/libcamera.git) - which has the Raspberry Pi 5 support.

miguelwon, 2023-11-29:
I just edited `/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml` file and replaced `bcm2835` with `pisp`, and it worked. I got some frames.

However, when tried with the `gst-launch-1.0` (my main objective) I have a new error. I'll open a new issue.

kbingham, 2023-11-29:
Just the fact it's looking at '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml' shows there's something wrong in the configuration. I had that myself earlier today. I'm not sure how I fixed it other than uninstalling the package libcamera-apps and reinstalling/rebuilding the latest rpicam-apps directly. Things seem to work for me now. (The libcamera-apps repository has now been renamed to prevent confusion between libcamera-apps and libcamera)

miguelwon, 2023-11-29:
Ok, just rebuild them (rpicam-apps) and got the same error I'm having when trying to use libcamera with gstreamer:

```
pi@raspberrypi:~/libcamera $ rpicam-hello 
[0:16:20.467019527] [6843]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:16:20.483165166] [6846]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:16:20.501891571] [6846]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 2304:1296:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 3400
    SRGGB10_CSI2P,2304x1296/0 - Score: 1000
    SRGGB10_CSI2P,4608x2592/0 - Score: 1900
Stream configuration adjusted
[0:16:20.502767869] [6843]  INFO Camera camera.cpp:1183 configuring streams: (0) 2304x1296-YUV420 (1) 2304x1296-BGGR16_PISP_COMP1
[0:16:20.502887147] [6846]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
[0:16:20.516584427] [6846] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[17:cap]: Failed to start streaming: Invalid argument
ERROR: *** failed to start camera ***
```

Should I open a new issue? Not sure where...

naushir, 2023-11-30:
> Just the fact it's looking at '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml' shows there's something wrong in the configuration. I had that myself earlier today. I'm not sure how I [...]

It sounds like you were missing https://github.com/raspberrypi/rpicam-apps/commit/c0617ebacb59c974415c80bc462fc79623bc2475 from you rpicam-apps build tree for this problem.

For your your last problem, please make sure you are building or fetching (via ``sudo rpi-update``) the latest kernel tree.

miguelwon, 2023-11-30:
Thanks but when building rpicam-apps from that commit I get the following error:

```
pi@raspberrypi:~/rpicam-apps $ meson compile -C build
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/pi/rpicam-apps/build
ninja: Entering directory `/home/pi/rpicam-apps/build'
[6/44] Compiling C++ object libcamera_app.so.1.2.2.p/core_libcamera_app.cpp.o
FAILED: libcamera_app.so.1.2.2.p/core_libcamera_app.cpp.o 
c++ -Ilibcamera_app.so.1.2.2.p -I. -I.. -Icore -I/usr/include/libcamera -I/usr/include -I/usr/include/aarch64-linux-gnu -I/usr/include/libpng16 -I/usr/include/libdrm -I/usr/include/aarch64-linux-gnu/qt5/QtCore -I/usr/include/aarch64-linux-gnu/qt5 -I/usr/include/aarch64-linux-gnu/qt5/QtWidgets -I/usr/include/aarch64-linux-gnu/qt5/QtGui -fdiagnostics-color=always -Wall -Winvalid-pch -Wextra -Wpedantic -Werror -std=c++17 -O3 -pedantic -Wno-unused-parameter -faligned-new -D_FILE_OFFSET_BITS=64 -Wno-psabi -ftree-vectorize -DLIBAV_PRESENT=1 -DLIBDRM_PRESENT=1 -DLIBEGL_PRESENT=1 -DQT_PRESENT=1 -fPIC -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -pthread -DBOOST_ALL_NO_LIB -MD -MQ libcamera_app.so.1.2.2.p/core_libcamera_app.cpp.o -MF libcamera_app.so.1.2.2.p/core_libcamera_app.cpp.o.d -o libcamera_app.so.1.2.2.p/core_libcamera_app.cpp.o -c ../core/libcamera_app.cpp
../core/libcamera_app.cpp: In member function ‘void LibcameraApp::ConfigureViewfinder()’:
../core/libcamera_app.cpp:367:25: error: ‘class libcamera::CameraConfiguration’ has no member named ‘transform’
  367 |         configuration_->transform = options_->transform;
      |                         ^~~~~~~~~
../core/libcamera_app.cpp: In member function ‘void LibcameraApp::ConfigureStill(unsigned int)’:
../core/libcamera_app.cpp:413:25: error: ‘class libcamera::CameraConfiguration’ has no member named ‘transform’
  413 |         configuration_->transform = options_->transform;
      |                         ^~~~~~~~~
../core/libcamera_app.cpp: In member function ‘void LibcameraApp::ConfigureVideo(unsigned int)’:
../core/libcamera_app.cpp:467:25: error: ‘class libcamera::CameraConfiguration’ has no member named ‘transform’
  467 |         configuration_->transform = options_->transform;
      |                         ^~~~~~~~~
../core/libcamera_app.cpp:492:25: error: ‘class libcamera::CameraConfiguration’ has no member named ‘transform’
  492 |         configuration_->transform = options_->transform;
      |                         ^~~~~~~~~
[11/44] Compiling C++ object libcamera_app.so.1.2.2.p/encoder_null_encoder.cpp.o
ninja: build stopped: subcommand failed.
```

But anyway, with the kernel update and after re-building libcamera, I get it working. Also,  I'm now able to get something from gstreamer.

naushir, 2023-11-30:
Good to hear.  In general, when you build rpicam-apps yourself, you should first rebuild libcamera to endure the API remains compatible.

closed 2023-11-30 by naushir