raspberrypi/libcamera #94: New version of python3-libcamera dependency error

tom-ard, 2023-12-01:
When I was using "picamera2", the following error occurred:
[image]

My current python3-libcamera version
[image]

You can see through the ldd command that the dynamic library depends on libpisp1.0.0
[image]

tom-ard, 2023-12-01:
Thanks for the reminder, I found the cause of the problem.
This was caused by installing libcamera under /usr/local when I compiled libcamera before.

closed 2023-12-01 by tom-ard

davidplowman, 2023-12-01:
Glad it's working now. I don't know if it helps, but I quite often "apt purge" the various system camera packages and applications, just to be sure I can never accidentally pick them up...!