raspberrypi/libcamera #96: Encoding and Decoding Example?

Type1J, 2023-12-14:
It seems that libcamera is where Raspberry Pi OS (64-bit bullseye and beyond) have landed as a replacement for OpenMAX and maybe Video4Linux2 as well. Does is replace m2m?

If it does replace m2m, I can't seem to find an example of encoding a YU12 to H.264 and H.264 to YU12.

Is there such an example?

(It would be fine if the example used YUYV or some other uncompressed format.)

naushir, 2023-12-14:
Are you talking about using cameras here?  If so, you can find some examples [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-vid).

6by9, 2023-12-14:
libcamera replaces the legacy camera stack, whether accessed through OpenMAXIL,  MMAL, or the old bcm2835-camera V4L2 driver.
It is solely dealing with processing camera images and delivering YUV or RGB pixel buffers, with the option of the raw Bayer if desired.

It does not touch the codecs, which are still exposed as V4L2 memory to memory devices on Pi0-4. (Pi5 only has hardware acceleration for HEVC decode).

closed 2023-12-14 by Type1J

Type1J, 2023-12-14:
My project uses WebRTC. I'm not sure why the Pi5 would remove support for H.264 since H.264 and VP8 are the only codecs supported in WebRTC, but even then, it's only the constrained baseline profile of H.264.

Type1J, 2023-12-14:
> Are you talking about using cameras here? If so, you can find some examples [here](https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-vid).

Yes, I'll be detecting some things in the image, so I'll need it uncompressed, but my cameras only give 10 fps for uncompressed formats. I'd like 30 fps which is offered with MJPG and H264, but that means that I need to decompress it to process it, but I can pass along the uncompressed version with the results through WebRTC and composite those results in the viewer (a browser), which won't be on the same machine.

naushir, 2023-12-14:
> My project uses WebRTC. I'm not sure why the Pi5 would remove support for H.264 since H.264 and VP8 are the only codecs supported in WebRTC, but even then, it's only the constrained baseline profile of H.264.

H.264 software encode is still supported through our libav interface.  In fact, you will find the quality is much improved over the old HW codec, particularly for low bitrate encode usecases such as WebRTC.