raspberrypi/libcamera #99: Handling of 16-bit bayer Endianness

schoolpost, 2023-12-22:
Trying to get the ClearHDR 16bit mode of an IMX585 working with Raspberry Pi 5 through libcamera. 

Here is a capture from libcamera-hello with default viewfinder mode:
```
pi@imx585:~ $ libcamera-hello --framerate 24 --analoggain 1.0 --shutter 20833 --width 3856 --height 2180 -t 0 --awbgains 1.2,3.5 
[1:27:32.570571057] [4647]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+147-a409bd53-dirty (2023-12-14T00:22:27-07:00)
[1:27:32.571924478] [4648]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 14-12-2023 (00:22:27)
[1:27:32.573327733] [4648]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'imx585'
[1:27:32.573347048] [4648]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[1:27:32.577095715] [4648]  INFO RPI pisp.cpp:1119 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx585@37 to CFE device /dev/media0 and ISP device /dev/media3 using PiSP variant BCM2712_C0
Made DRM preview window
Mode selection for 1910:1080:12:P(24)
    SRGGB12_CSI2P,1928x1090/90.1713 - Score: 7.10831
    SRGGB12_CSI2P,3856x2180/50 - Score: 761.608
    SRGGB16,1928x1090/30.0003 - Score: 2007.11
    SRGGB16,3856x2180/30.0003 - Score: 2761.61
Stream configuration adjusted
[1:27:32.635195864] [4647]  INFO Camera camera.cpp:1183 configuring streams: (0) 1910x1080-YUV420 (1) 1928x1090-RGGB16_PISP_COMP1
[1:27:32.635293328] [4648]  INFO RPI pisp.cpp:1403 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx585@37 - Selected sensor format: 1928x1090-SRGGB12_1X12 - Selected CFE format: 1928x1090-PC1R
```
[image]

Here is a capture from libcamera-hello, with a forced 16-bit unpacked viewfinder mode:
```
pi@imx585:~ $ libcamera-hello --framerate 24 --analoggain 1.0 --shutter 20833 --width 3856 --height 2180 --viewfinder-mode 3856:2180:16:U -t 0 --awbgains 1.2,3.5 
[1:28:54.404346340] [4694]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+147-a409bd53-dirty (2023-12-14T00:22:27-07:00)
[1:28:54.405863763] [4695]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 14-12-2023 (00:22:27)
[1:28:54.407285129] [4695]  WARN CameraSensorProperties camera_sensor_properties.cpp:261 No static properties available for 'imx585'
[1:28:54.407307185] [4695]  WARN CameraSensorProperties camera_sensor_properties.cpp:263 Please consider updating the camera sensor properties database
[1:28:54.411138816] [4695]  INFO RPI pisp.cpp:1119 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx585@37 to CFE device /dev/media0 and ISP device /dev/media3 using PiSP variant BCM2712_C0
Made DRM preview window
Mode selection for 3856:2180:16:U(24)
    SRGGB12_CSI2P,1928x1090/90.1713 - Score: 8036
    SRGGB12_CSI2P,3856x2180/50 - Score: 2000
    SRGGB16,1928x1090/30.0003 - Score: 6036
    SRGGB16,3856x2180/30.0003 - Score: 0
Stream configuration adjusted
[1:28:54.468039706] [4694]  INFO Camera camera.cpp:1183 configuring streams: (0) 1910x1080-YUV420 (1) 3856x2180-RGGB16_PISP_COMP1
[1:28:54.468141466] [4695]  INFO RPI pisp.cpp:1403 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx585@37 - Selected sensor format: 3856x2180-SRGGB16_1x16 - Selected CFE format: 3856x2180-PC1R
```
[image]

How can libcamera / libpisp be configured to handle this?

schoolpost, 2023-12-22:
Same issue noted here, but for the CM4/Pi4:
https://forums.raspberrypi.com/viewtopic.php?p=2123617#p2123617

6by9, 2023-12-22:
> Same issue noted here, but for the CM4/Pi4: https://forums.raspberrypi.com/viewtopic.php?p=2123617#p2123617

That was a misinterpretation of the captured data. The produced data should match https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/pixfmt-srggb16.html and has to be processed accordingly.

For this issue am I reading that when using `--viewfinder-mode 3856:2180:16:U` that the ISP is apparently incorrectly processing the data when producing the viewfinder image?
I don't believe we've had a 16bit sensor to test with. The fact that the format definition in [cfe_fmts.h](https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/media/platform/raspberrypi/rp1_cfe/cfe_fmts.h#L195) is missing a `.csi_dt` entry looks suspicious (it isn't defined in CSI2 v1.3, but is now defined as 0x2e or use [MIPI_CSI2_DT_RAW16](https://elixir.bootlin.com/linux/latest/source/include/media/mipi-csi2.h#L42)).

The FE is always producing compressed or 16bpc data, so it seems unlikely that the back end is misinterpreting the data.

6by9, 2023-12-22:
#5806 includes a patch to add the csi_dt value for 16bit formats, but I need others to confirm whether that is needed or not. I have a suspicion that all CSI2 data (including metadata) will be going through the CFE without that.

naushir, 2023-12-23:
Not sure if the csi_dt change will make a difference, but does no harm to try it out.  I'm pretty sure the ISP Backend correctly reads in 16-bit Bayer as that's one of the normal modes of operation.  Can you try asking for 16-bit compressed to 8-bit Bayer and see if that makes any difference?  You can do this by using ``--viewfinder-mode 3856:2180:16:P`` as the command line argument.

Also, have you managed to verify this 3856x2180 16-bit mode is correctly streaming from the sensor?  Possibly the easiest way to verify this is to use a CM4 and drive Unicam directly via ``v4l2-ctl`` and save the output Bayer frames to a file.

naushir, 2023-12-30:
Any update on this?

naushir, 2024-01-15:
Closing this now due to inactivity.... Feel free to reopen if this is still an issue.

schoolpost, 2024-02-29:
Hi, re-opening this issue; still struggling to get an rpicam-app ( libcamera-app ) to return a unpacked 16-bit RAW frame, no matter my mode query I always get compressed `RGGB16_PISP_COMP1` format

Info:
```
Linux raspberrypi 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```
```
rpicam-apps build: 08895fbe3c06-intree-dirty 28-02-2024 (05:34:30)
libcamera build: v0.0.0+4131-7ca67c3c
```

```
pi@raspberrypi:~/cinepi-raw $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx585 [3840x2160 16-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx585@1a)
    Modes: 'SRGGB12_CSI2P' : 1928x1090 [87.60 fps - (0, 0)/3840x2160 crop]
                             3856x2180 [43.80 fps - (0, 0)/3840x2160 crop]
           'SRGGB16' : 1928x1090 [30.00 fps - (0, 0)/3840x2160 crop]
                       3856x2180 [21.90 fps - (0, 0)/3840x2160 crop]```

schoolpost, 2024-02-29:
Ok, now having time to revisit this; I'm able to better present the issue at hand; and hopefully there is a mechanism somewhere in this software/hardware stack to handle it: 

the IMX585 has a clearHDR mode that combines pixel readings at 2 different gains within the same exposure into a single 16-bit frame. 

Now that I am able to capture uncompressed 16-bit bayer frames, I have been able to confirm the issue has to do with endianness as originally suspected. Shown from the datasheet: 
[image]

I can further confirm with a python script that will display the bayer raw as a 16-bit mono image, when I perform a byte swap the image is rendered correctly ( albeit possessing some artifacts / noise / clipping ) 

Unswapped: 
[image]

Swapped: 
[image]

Now outside of my python script, I can perform the same operation within a libcamera based app by doing:
```
		BufferWriteSync w(static_cast<RPiCamApp*>(&app), completed_request->buffers[app.RawStream()]);
		const std::vector<libcamera::Span<uint8_t>> mem = w.Get();
		uint16_t *ptr = (uint16_t *)mem[0].data();
		for (unsigned int i = 0; i < mem[0].size() / 2; ++i) {  
			uint16_t value = *ptr;  
			*ptr = (value >> 8) | (value << 8); 
			++ptr;  
		}
```

How to configure libcamera to interpret the big-endian data appropriately? can it be done via the Pi5 ISP via libpisp? Can it be done in software before it gets further down the ISP pipeline ( RGB pipe ) ? 

As it stands the ISP / preview is rendered unusable: 
[image]

njhollinghurst, 2024-02-29:
This seems to be a hardware issue, and is not something that can easily be fixed in libcamera.
- For Unicam (up to Raspberry Pi 4), RAW16 is simply not supported; it was added to CSI-2 after the chip was designed.
- For RP1 (Raspberry Pi 5) it was supposed to be supported, but an endian-swap was omitted from the hardware pipeline.

schoolpost, 2024-02-29:
If not in hardware, are there any avenues in software for accomplishing this? I noticed there are already a number of format conversions happening with CPU and not on ISP when I investigated a couple months back: https://github.com/raspberrypi/libcamera/issues/93 https://github.com/raspberrypi/libcamera/blob/075b54d5229d0894109e7cbb4bb890bc48bb37e8/src/libcamera/pipeline/rpi/pisp/pisp.cpp#L321

Understandably those happen after the fact once the image is out of the ISP, It would need to perform a byte swap on the raw image while it is in memory before it is processed by the ISP

naushir, 2024-03-01:
> If not in hardware, are there any avenues in software for accomplishing this?

This might be possible, we will explore the options.  However, you end up in a situation where you cannot rely on the statistics for the 3A algorithms.  This may not be a problem if you want to control everything (exposure/gain/white balance) manually.

schoolpost, 2024-03-01:
> This might be possible, we will explore the options.  However, you end up in a situation where you cannot rely on the statistics for the 3A algorithms.  This may not be a problem if you want to [...]

Yes,

My application is full manual control, losing 3A would not be a major concern.

Thank you for looking into this, I eagerly await a solution and hope to test soon.

schoolpost, 2024-03-06:
Hi, following up to see if you;ve had the chance to look into this further.

thanks!

naushir, 2024-03-07:
Not yet unfortunately.  I will try to get to it next week if I can.

naushir, 2024-03-07:
@schoolpost I've attempted to add the SW endian swap code here:
https://github.com/raspberrypi/libcamera/commit/919a4365ef0834ddf968a18dd386e824d7c92262

Unfortunately, I can't really try it out properly since I don't have a 16-bit sensor.  Would you be able to give it a try and let me know if it works.  Note that you must run in fully manual mode as statistics will not be functional in this mode.

schoolpost, 2024-03-08:
> @schoolpost I've attempted to add the SW endian swap code here: [9c80dba](https://github.com/raspberrypi/libcamera/commit/9c80dbae77c8480bd791d499bc121a3d6ebf5af5) (from the next tree).
> 
> Unfortunately, I can't really try it out properly since I don't have a 16-bit sensor. Would you be able to give it a try and let me know if it works. Note that you must run in fully manual mode as statistics will not be functional in this mode.

Thanks @naushir,

I have been able to test and can confirm that it does in fact seem to bring me closer to a working preview. 

I'm only now realizing how many other ISP functions depend on the 3A for their function, because even though I set the:
- red/blue gains
- analog gain
- exposure time

colors and such are way off, maybe there are some things I'll need to address via a tuning.json file?

naushir, 2024-03-08:
You are definitely going to have to switch off alsc in the config.  In the tuning config file, under the ``"rpi.alsc`` section, set ``n_iter`` to 0.

schoolpost, 2024-03-09:
Ok, so this is currently where I stand with preview:
[image]

I am simultaneously juggling whether or not certain artifacts are from the PiSP or from incorrect / malformed sensor configuration in the driver.

Getting some reasonable color is my first goal, any ideas for why I am getting such heavily magenta image? issue with the CCM?

I am using the same same `tuning.json` file and same gain, shutter speed and red/blue gains in between testing sensor modes, as demonstrated here with the 12-bit non clearHDR mode I get correct colors ( other then my white balance being set too warm ) 
[image]

schoolpost, 2024-03-09:
@naushir can you comment on where in the FE process the byte swap is happening?

[image]

Like you mention before 3A would be broken still, so I imagine the byte swap is happening at the AXI point at the end not at the input?

Is it possible the FE processing is messing with the bayer pixels before the byteswap? i.e. DPC mis-identifying pixels and correcting? I notice a ton of speckly white noise in my RAW image...

Again I can only speculate at this point... I'd love to know your thoughts

naushir, 2024-03-11:
Have you tried taking the captured 16-bit raw frame and processing it through something like ``dcraw``?  That would indicate if the RAW data from the sensor (+ the SW endian handling) is correct or not.

> can you comment on where in the FE process the byte swap is happening?
The byte swap to correct the endiness is happening entirely in software after the FE has written the buffer to memory, so not in the hardware pipeline at all.  If things are setup correctly (which I think is the case), the FE will not be doing any pixel processing at all, so the wrong endianness in the pipeline should not matter.

You could try disabling some ISP stages and see if that improves things.  To do this, remove (or simply rename) the blocks in the config file to disable the pipeline stages.  For example, rename ``rpi.sharpness`` to `xrpi.sharpness``.

naushir, 2024-03-11:
Also, what command are you using for the preview?

njhollinghurst, 2024-03-11:
Byte swap is not happening in ISP-FE -- what is happening is that big-endian RAW16 data are being passed into ISP-FE input, which assumes little-endian. This is due to an omission from the specification of the preceding (CSI2AXI) block.

In order to write out the RAW16 data unchanged, it is important that Front End Decompand, BLA, DPC, Downscale and Compress stages are all disabled (or if BLA is enabled, it must be configured as a no-op; which I think is the default setup).

schoolpost, 2024-03-11:
> Byte swap is not happening in ISP-FE -- what is happening is that big-endian RAW16 data are being passed into ISP-FE input, which assumes little-endian. This is due to an omission from the specification of the preceding (CSI2AXI) block.
> 
> In order to write out the RAW16 data unchanged, it is important that Front End Decompand, BLA, DPC, Downscale and Compress stages are all disabled (or if BLA is enabled, it must be configured as a no-op; which I think is the default setup).
> 
> Edit; If you've changed the black level, make sure BLA is disabled here.
> 
> Software byte-swap now happens after ISP-FE output.

Turns out in addition to disabling BLA/BLC on the FE, there some peculiarities with bayer order with the this specific sensor mode. I am now able to get a proper ISP preview: 
[image]

@naushir I am using a custom EGL/OpenGL app for preview.

schoolpost, 2025-03-26:
Hello,

Just want to report that this fix is not longer valid in kernel `6.6` nor `6.12` , something must have changed with CFE or another component of the pipeline between `6.1` and those later versions that introduced this regression.

naushir, 2025-03-26:
Can you expand on what errors and/or corruption you see? Unfortunately, I am unable to test 16-bit output as I don't have the correct hardware.

schoolpost, 2025-03-26:
This is what the RAW looks like: 

[image]

Feels like an issue with width/stride?, as I can notice some slanted / skewed features, but nothing resembling a the original scene, very noisy and garbled.

schoolpost, 2025-03-26:
Command:
`rpicam-raw -t 2000 --segment 1 -o test%05d.raw --mode 3856:2180:16:U --framerate 24`

Libcamera Info:
```rpicam-apps build: v1.6.0 000000000000-invalid 25-03-2025 (16:52:37)
rpicam-apps capabilites: egl:1 qt:0 drm:1 libav:1
libcamera build: v0.4.0+53-29156679-dirty (2025-03-25T00:26:46MDT)
```

( Please excuse the dark image, it's just how it is with linear space. ) 

Image taken with Kernel 6.1:
```
6.1.93-v8-16k+
```
[image]

Image taken with Kernel 6.6:
```
6.6.78-v8-16k+
```
[image]

Only difference between the two is the change to different kernel, otherwise ran the same command with the same app/version of rpicam-apps/libcamera.

naushir, 2025-03-26:
Would you be able to show all the libcamera console output logged?  This shows the format/stride information that I'm after.

schoolpost, 2025-03-26:
> Would you be able to show all the libcamera console output logged? This shows the format/stride information that I'm after. Also, would be ideal to do this on both kernel versions to ensure they show identical formats being used.

Here is `6.6`:
```
[0:01:01.580122166] [1715]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679-dirty (2025-03-25T00:26:46MDT)
[0:01:01.587767462] [1832]  INFO RPI pisp.cpp:720 libpisp version v1.2.0 50426319aa1a-dirty 24-03-2025 (15:54:44)
[0:01:01.623989499] [1832]  WARN RPiAlsc alsc.cpp:171 no luminance table - assume unity everywhere
[0:01:01.625245296] [1832]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx585@1a to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
WARNING: consider inline headers with 'pause'/split/segment/circular
Mode selection for 3856:2180:16:U(24)
    SRGGB12_CSI2P,1928x1090/120.005 - Score: 8036
    SRGGB12_CSI2P,3856x2180/60.0024 - Score: 2000
    SRGGB16,1928x1090/60.0024 - Score: 6036
    SRGGB16,3856x2180/30.0003 - Score: 0
Stream configuration adjusted
[0:01:01.627200962] [1715]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-YUV420 (1) 3856x2180-SRGGB16
[0:01:01.627382425] [1832]  WARN RPI pisp.cpp:1457 The sensor is configured for a 16-bit output, statistics  will not be correct. You must use manual camera settings.
[0:01:01.627507073] [1832]  INFO RPI pisp.cpp:1485 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx585@1a - Selected sensor format: 3856x2180-SRGGB16_1X16 - Selected CFE format: 3856x2180-RG16
Raw stream: 3856x2180 stride 7744 format SRGGB16
```

naushir, 2025-03-26:
And can you do the same with the 6.1 (working) kernel?

schoolpost, 2025-03-26:
Here is `6.1` ( sorry, takes a bit to recompile driver/switch kernel on the fly ) 

```
[0:00:35.163801615] [1584]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679-dirty (2025-03-25T00:26:46MDT)
[0:00:35.173185958] [1603]  INFO RPI pisp.cpp:720 libpisp version v1.2.0 50426319aa1a-dirty 24-03-2025 (15:54:44)
[0:00:35.223099351] [1603]  WARN RPiAlsc alsc.cpp:171 no luminance table - assume unity everywhere
[0:00:35.224286758] [1603]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx585@1a to CFE device /dev/media0 and ISP device /dev/media1 using PiSP variant BCM2712_C0
WARNING: consider inline headers with 'pause'/split/segment/circular
[0:00:35.225933542] [1584]  WARN V4L2 v4l2_pixelformat.cpp:346 Unsupported V4L2 pixel format RPBP
Mode selection for 3856:2180:16:U(24)
    SRGGB12_CSI2P,1928x1090/90.1713 - Score: 8036
    SRGGB12_CSI2P,3856x2180/50 - Score: 2000
    SRGGB16,1928x1090/30.0003 - Score: 6036
    SRGGB16,3856x2180/30.0003 - Score: 0
Stream configuration adjusted
[0:00:35.226143772] [1584]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-YUV420 (1) 3856x2180-SRGGB16
[0:00:35.226186809] [1603]  WARN RPI pisp.cpp:1457 The sensor is configured for a 16-bit output, statistics  will not be correct. You must use manual camera settings.
[0:00:35.226206983] [1603]  INFO RPI pisp.cpp:1485 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx585@1a - Selected sensor format: 3856x2180-SRGGB16_1X16 - Selected CFE format: 3856x2180-RG16
Raw stream: 3856x2180 stride 7744 format SRGGB16
```

njhollinghurst, 2025-03-26:
It might possibly have been https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43 or https://github.com/raspberrypi/linux/commit/019c78a29bba078fdfe4c3feed570af46c18d79b
which might be interacting with a DT mismatch in the CFE hardware...

naushir, 2025-03-26:
@schoolpost can you try reverting https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43 and see if that works?

schoolpost, 2025-03-26:
> [@schoolpost](https://github.com/schoolpost) can you try reverting [raspberrypi/linux@98b1f0d](https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43) and see if that works?

Appears to have done the trick.

will127534, 2025-06-26:
Hi @naushir is it still expected to require reverting  [raspberrypi/linux@98b1f0d ](https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43) for 16bit mode to work on RPI now?

naushir, 2025-06-26:
Yes, I believe it is, although @njhollinghurst will have to remind me what's going wrong.  Should we be using the 14-bit id ``MIPI_CSI2_DT_RAW14`` for the 16-bit mode?

naushir, 2025-06-26:
See https://github.com/raspberrypi/libcamera/issues/278#issuecomment-3008209784.  With this, we should be able to run with 16-bit modes without needing to revert [raspberrypi/linux@98b1f0d](https://github.com/raspberrypi/linux/commit/98b1f0dbb23faf3e96005a25f6acb95b64afce43).