raspberrypi/libcamera #101: Private headers must not be included in the libcamera API

Digital1O1, 2024-01-16:
Hello!

So I'm running Bookworm on my Raspberry PI 4 and I built/installed libcamera with the following commands as stated here : https://libcamera.org/getting-started.html

```
git clone https://git.libcamera.org/libcamera/libcamera.git
cd libcamera
meson setup build
ninja -C build install
```

Once the build was done I went to libcamera --> test --> camera --> then opened 'capture.cpp' since my endgoal was just to understand how to use libcamera to just capture a single image.

When I finally got the G++ command right to build/compile capture.cpp I got the following errors.

```
In file included from /home/pi/libcamera/include/libcamera/base/event_dispatcher.h:12,
                 from capture.cpp:12:
/home/pi/libcamera/include/libcamera/base/private.h:21:2: error: #error "Private headers must not be included in the libcamera API"
   21 | #error "Private headers must not be included in the libcamera API"
      |  ^~~~~
In file included from /home/pi/libcamera/include/libcamera/base/thread.h:14,
                 from capture.cpp:13:
/home/pi/libcamera/include/libcamera/base/private.h:21:2: error: #error "Private headers must not be included in the libcamera API"
   21 | #error "Private headers must not be included in the libcamera API"
      |  ^~~~~
In file included from /home/pi/libcamera/include/libcamera/base/message.h:12,
                 from /home/pi/libcamera/include/libcamera/base/thread.h:16:
/home/pi/libcamera/include/libcamera/base/private.h:21:2: error: #error "Private headers must not be included in the libcamera API"
   21 | #error "Private headers must not be included in the libcamera API"
      |  ^~~~~
In file included from /home/pi/libcamera/include/libcamera/base/utils.h:23,
                 from /home/pi/libcamera/include/libcamera/base/thread.h:18:
/home/pi/libcamera/include/libcamera/base/private.h:21:2: error: #error "Private headers must not be included in the libcamera API"
   21 | #error "Private headers must not be included in the libcamera API"
      |  ^~~~~
In file included from /home/pi/libcamera/include/libcamera/base/timer.h:13,
                 from capture.cpp:14:
/home/pi/libcamera/include/libcamera/base/private.h:21:2: error: #error "Private headers must not be included in the libcamera API"
   21 | #error "Private headers must not be included in the libcamera API"
      |  ^~~~~
```

The part that I'm really confused about is that I'm under the heavy assumption that the majority of the .cpp files in libcamera that I built from source are supposed to be used as frameworks for whatever project we choose to endeavor in. 

If that is the case I'm confused about why private headers were included as mentioned above if 'capture.cpp' was supposed to work 'right out of the box'

kbingham, 2024-01-16:
Getting that error means that something you have compiled is internal code that should not be compiled outside of libcamera, at least not without being sure of what you are doing.

The internal tests are only unit tests. They're not specifically expected to be used in applications.

For a sample code to get started with libcamera, look at https://git.libcamera.org/libcamera/simple-cam.git/

naushir, 2024-01-16:
I think @kbingham has pretty much answered this.  Do you have any further questions?  If not, feel free to close this down.

closed 2024-01-16 by Digital1O1