raspberrypi/libcamera #103: [BUG] ov9281 doesn't work on PI5

tom-ard, 2024-01-23:
The following error occurs when ov9281 is run on Raspberry Pi 5.
[image]

Complete debug information
[debug.log](https://github.com/raspberrypi/libcamera/files/14021794/debug.log)

Raspberry Pi system version
[image]
[image]

6by9, 2024-01-23:
This has been fixed on the next branch, but not made it to master and been packaged into apt yet
https://github.com/raspberrypi/libcamera/commit/d8c775077c33f7475c5d4828ce3e374d5769b59c

kbingham, 2024-01-23:
Especially now that libcamera has updated the kernel headers to 6.7, there's no reason not to get this merged.

I've cherry-picked d8c7750 and posted it to libcamera-devel (https://patchwork.libcamera.org/patch/19461/)

kbingham, 2024-01-23:
Merged Y16 to : https://git.libcamera.org/libcamera/libcamera.git/commit/?id=2ab8b8d48aa84f58b055ecf241d7c47b9fcb7a71
Doesn't fix this issue until it's packaged and updated into apt - but clears the technical debt.

tom-ard, 2024-01-24:
Thank you for your reply. When will it be merged into the main branch?

naushir, 2024-01-24:
We are trying to push all our Pi 5 changes to upstream libcamera.  Hopefully this should be fairly quick, and our next release will then closely align with the upstream libcamera tree.

naushir, 2024-02-26:
This should now be available and working in the latest apt release of our libcamera package.

closed 2024-02-26 by naushir

ScottGibb, 2024-04-11:
Hi there, i'm currently trying to use the ov9281 on the Raspberry Pi 5. However i'm bumping into the same issue as above, But I am using the latest version of libcamera which should support the ov9281. The only issue I see, is that my kernel is not the latest. Will this prevent me from using the ov9281?

**uname -a:** 
Linux e088407d62b1 6.1.54-v8 #1 SMP PREEMPT Wed Sep 27 16:45:31 UTC 2023 aarch64 GNU/Linux

**libcamera-hello:**
[0:34:02.842931936] [1053]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+46-075b54d5
[0:34:02.879393510] [1056]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 14-02-2024 (14:00:12)
[0:34:02.881266436] [1056]  WARN CameraSensorProperties camera_sensor_properties.cpp:274 No static properties available for 'ov9281'
[0:34:02.881286455] [1056]  WARN CameraSensorProperties camera_sensor_properties.cpp:276 Please consider updating the camera sensor properties database
[0:34:02.886738640] [1056]  WARN V4L2 v4l2_subdevice.cpp:522 'ov9281 6-0060': Unknown subdev format 0x0000, defaulting to RGB encoding
[0:34:02.886764103] [1056] ERROR V4L2 v4l2_subdevice.cpp:246 No information available for format '1280x800-0x0000'
[0:34:02.889103029] [1056]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov9281@60 to CFE device /dev/media0 and ISP device /dev/media3 using PiSP variant BCM2712_C0
[0:34:02.889167066] [1056]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 14-02-2024 (14:00:12)
[0:34:02.989424825] [1056]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media4 using PiSP variant BCM2712_C0
Preview window unavailable
[0:34:02.996792695] [1053]  WARN V4L2 v4l2_pixelformat.cpp:346 Unsupported V4L2 pixel format <INVALID>
[0:34:02.996812862] [1053]  WARN Formats formats.cpp:997 Unsupported pixel format 0x00000000
[0:34:02.996809844] [1053] ERROR RPI pisp.cpp:228 Pixel format <INVALID> unsupported
[0:34:02.996830177] [1053] FATAL default pisp.cpp:229 assertion "0" failed in toPiSPImageFormat()
Backtrace:
??? [0x0000007fa427aa88] (/usr/lib/aarch64-linux-gnu/libcamera.so.0.2.0 [0x0000007fa427aa88])
??? [0x0000007fa427ad68] (/usr/lib/aarch64-linux-gnu/libcamera.so.0.2.0 [0x0000007fa427ad68])
libcamera::PiSPCameraData::platformValidate(libcamera::RPi::RPiCameraConfiguration*) const+0x12c (/usr/lib/aarch64-linux-gnu/libcamera.so.0.2.0 [0x0000007fa427b29c])
libcamera::RPi::RPiCameraConfiguration::validate()+0x3f4 (/usr/lib/aarch64-linux-gnu/libcamera.so.0.2.0 [0x0000007fa42728e8])
libcamera::RPi::PipelineHandlerBase::generateConfiguration(libcamera::Camera*, libcamera::Span<libcamera::StreamRole const, 18446744073709551615ul>)+0x4b0 (/usr/lib/aarch64-linux-gnu/libcamera.so.0.2.0 [0x0000007fa42734d0])
libcamera::Camera::generateConfiguration(libcamera::Span<libcamera::StreamRole const, 18446744073709551615ul>)+0xdc (/usr/lib/aarch64-linux-gnu/libcamera.so.0.2.0 [0x0000007fa41f430c])
RPiCamApp::OpenCamera()+0x270 (/usr/lib/aarch64-linux-gnu/rpicam_app.so.1.4.3 [0x0000007fa47668f0])
main+0xa8 (/usr/bin/rpicam-hello [0x00000055780f1ce8])
??? [0x0000007fa4337780] (/usr/lib/aarch64-linux-gnu/libc.so.6 [0x0000007fa4337780])
__libc_start_main+0x98 (/usr/lib/aarch64-linux-gnu/libc.so.6 [0x0000007fa4337858])
??? [0x00000055780f21f0] (/usr/bin/rpicam-hello [0x00000055780f21f0])
Aborted

naushir, 2024-04-11:
You will need the latest kernel to get this working correctly.

ScottGibb, 2024-04-11:
@naushir Thank you so much for the quick reply!!!