raspberrypi/libcamera #104: libcamera-apps fail to run V2 camera on RPi5

TetsuyaNOMURA, 2024-02-01:
On RPi5 with Bookworm OS as below.
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

In case of IMX477 I can run rpicam-hello as below (camera_auto_detect=1 and dtoverlay=imx477,cam0 in config.txt.)
pi@raspberrypi:~ $ rpicam-hello
[0:06:42.300970236] [2085]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+151-9e31bb6e
[0:06:42.312874880] [2087]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 01-02-2024 (16:31:10)
[0:06:42.329705978] [2087]  INFO RPI pisp.cpp:1119 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:06:42.329897532] [2087]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 01-02-2024 (16:31:10)
[0:06:42.340172661] [2087]  INFO RPI pisp.cpp:1119 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx477@1a to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0
Made X/EGL preview window
Mode selection for 2028:1520:12:P
    SRGGB10_CSI2P,1332x990/0 - Score: 3456.22
    SRGGB12_CSI2P,2028x1080/0 - Score: 1083.84
    SRGGB12_CSI2P,2028x1520/0 - Score: 0
    SRGGB12_CSI2P,4056x3040/0 - Score: 887
Stream configuration adjusted
[0:06:43.110268054] [2085]  INFO Camera camera.cpp:1183 configuring streams: (0) 2028x1520-YUV420 (1) 2028x1520-BGGR16_PISP_COMP1
[0:06:43.110423383] [2087]  INFO RPI pisp.cpp:1403 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected CFE format: 2028x1520-PC1B

But in case of IMX219(V2 camera) I cannot.
pi@raspberrypi:~ $ libcamera-hello -t 0 --camera 0 [0:03:19.381199316] [2241]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e [0:03:19.393088444] [2244]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22) [0:03:19.415715061] [2244]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0 [0:03:19.416108545] [2244]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22) [0:03:19.430771027] [2244]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media3 and ISP device /dev/media1 using PiSP variant BCM2712_C0 Made X/EGL preview window [0:03:20.255264754] [2241]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 [0:03:20.255292365] [2241]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6 [0:03:20.255297939] [2241]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6 [0:03:20.255304476] [2241]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:03:20.255631794] [2241]  INFO Camera camera.cpp:1183 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1 [0:03:20.255745165] [2244]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B [0:03:20.268855695] [2244] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video12[16:cap]: Failed to start streaming: Broken pipe
ERROR: *** failed to start camera ***

I built the libcamera with "next" branch, the symptom is the same.
I cannot operate IMX290 neither.

camera_auto_detect=1 and dtoverlay=imx219,cam0 in config.txt.

Any suggestion, please.

TetsuyaNOMURA, 2024-02-02:
I switched to Bookworm 32Bit OS, and now I can see IMX219 image on RPi5. It might be 64bit OS related problem.

naushir, 2024-02-02:
@TetsuyaNOMURA have you tried a fresh 64-bit OS image with updates installed?

TetsuyaNOMURA, 2024-02-02:
Dear Naushir-san

 

Yes, I used the Raspberry Pi Imager and down loaded the latest 64 bit (recommended) OS.

 

And down loaded again the 32 bit OS, I can run IMX219 and also IMX327(IMX290) as well as RPi4 case.

 

Best Regards,

 

NOMURA

naushir, 2024-02-02:
On the 64-bit OS, can you try running ``sudo rpi-update stable``, reboot and see if that works?

TetsuyaNOMURA, 2024-02-02:
Dear Naushir-san

 

The result is the same. IMX219 on cam0 does not work, IMX477 on cam1 works.

 

***@***.***:~ $ libcamera-hello --camera 0

[0:01:03.194158943] [2207]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+151-9e31bb6e

[0:01:03.206528713] [2208]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 01-02-2024 (16:31:10)

[0:01:03.228929803] [2208]  INFO RPI pisp.cpp:1119 Registered camera ***@***.******@***.******@***.*** to CFE device /dev/media0 and ISP device /dev/media3 using PiSP variant BCM2712_C0

[0:01:03.229091451] [2208]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 01-02-2024 (16:31:10)

[0:01:03.241084103] [2208]  INFO RPI pisp.cpp:1119 Registered camera ***@***.******@***.******@***.*** to CFE device /dev/media2 and ISP device /dev/media4 using PiSP variant BCM2712_C0

Made X/EGL preview window

Mode selection for 1640:1232:12:P

    SRGGB10_CSI2P,640x480/0 - Score: 4504.81

    SRGGB10_CSI2P,1640x1232/0 - Score: 1000

    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48

    SRGGB10_CSI2P,3280x2464/0 - Score: 1718

    SRGGB8,640x480/0 - Score: 5504.81

    SRGGB8,1640x1232/0 - Score: 2000

    SRGGB8,1920x1080/0 - Score: 2541.48

    SRGGB8,3280x2464/0 - Score: 2718

Stream configuration adjusted

[0:01:03.592373085] [2207]  INFO Camera camera.cpp:1183 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1

[0:01:03.592531882] [2208]  INFO RPI pisp.cpp:1403 Sensor: ***@***.******@***.******@***.*** - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B

[0:01:03.613515611] [2208] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[16:cap]: Failed to start streaming: Broken pipe

TetsuyaNOMURA, 2024-02-03:
> Can you also attach the output of camera-bug-report after you run libcamera-hello and see it fail?
>
>
> --------------------------------------------------------------------------------
> rpicam-apps Bug Report
> --------------------------------------------------------------------------------
> Command: /usr/local/bin/camera-bug-report
>
> --------------------------------------------------------------------------------
> Hardware information
> --------------------------------------------------------------------------------
> ** hostname **
> raspberrypi
>
> ** cat /proc/cpuinfo **
> processor	: 0
> BogoMIPS	: 108.00
> Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
> CPU implementer	: 0x41
> CPU architecture: 8
> CPU variant	: 0x4
> CPU part	: 0xd0b
> CPU revision	: 1
>
> processor	: 1
> [... 1107 lines cut ...]
> 		<- "csi2":5 []
>
> - entity 27: rp1-cfe-csi2_ch2 (1 pad, 1 link)
>              type Node subtype V4L flags 0
>              device node name

Dear Nausir-san

 

The cam1 does not work neither.

I’m attaching the bug report.

***@***.***:~ $ libcamera-hello --camera 1

[0:00:39.964855160] [1755]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+151-9e31bb6e

[0:00:39.980847845] [1756]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 01-02-2024 (16:31:10)

[0:00:39.999321271] [1756]  INFO RPI pisp.cpp:1119 Registered camera ***@***.******@***.******@***.*** to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0

[0:00:39.999441623] [1756]  INFO RPI pisp.cpp:662 libpisp version v1.0.1 5652744a74d7 01-02-2024 (16:31:10)

[0:00:40.010009030] [1756]  INFO RPI pisp.cpp:1119 Registered camera ***@***.******@***.******@***.*** to CFE device /dev/media3 and ISP device /dev/media4 using PiSP variant BCM2712_C0

Made X/EGL preview window

Mode selection for 1640:1232:12:P

    SRGGB10_CSI2P,640x480/0 - Score: 4504.81

    SRGGB10_CSI2P,1640x1232/0 - Score: 1000

    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48

    SRGGB10_CSI2P,3280x2464/0 - Score: 1718

    SRGGB8,640x480/0 - Score: 5504.81

    SRGGB8,1640x1232/0 - Score: 2000

    SRGGB8,1920x1080/0 - Score: 2541.48

    SRGGB8,3280x2464/0 - Score: 2718

Stream configuration adjusted

[0:00:40.384314530] [1755]  INFO Camera camera.cpp:1183 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1

[0:00:40.384484363] [1756]  INFO RPI pisp.cpp:1403 Sensor: ***@***.******@***.******@***.*** - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B

[0:00:40.404920993] [1756] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video4[34:cap]: Failed to start streaming: Broken pipe

ERROR: *** failed to start camera ***

alphanumeric007, 2024-02-04:
I do believe I have the same issue. Latest Pi OS 64 bit on a Pi 5, NVMe install. 
Official camera cables, official power supply, official v2 cameras. 
One standard and one NOIR. Both show the same error.
I've done a rpi-eeprom update, and set it to use latest in raspi-config.
I've also done apt update, apt upgrade, and rpi-update, no change in the behavior.  

`
BOOTLOADER: up to date
   CURRENT: Mon Jan 22 02:44:36 PM UTC 2024 (1705934676)
    LATEST: Mon Jan 22 02:44:36 PM UTC 2024 (1705934676)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
`
`
kyrie@raspberrypi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx219 [3280x2464 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@88000/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

1 : imx219 [3280x2464 10-bit RGGB] (/base/axi/pcie@120000/rp1/i2c@80000/imx219@10)
    Modes: 'SRGGB10_CSI2P' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                             1640x1232 [41.85 fps - (0, 0)/3280x2464 crop]
                             1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                             3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]
           'SRGGB8' : 640x480 [206.65 fps - (1000, 752)/1280x960 crop]
                      1640x1232 [83.70 fps - (0, 0)/3280x2464 crop]
                      1920x1080 [47.57 fps - (680, 692)/1920x1080 crop]
                      3280x2464 [21.19 fps - (0, 0)/3280x2464 crop]

`

`
kyrie@raspberrypi:~ $ libcamera-hello --camera 0 -t 0
[0:05:43.560205697] [2896]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:05:43.573060269] [2899]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:05:43.600561393] [2899]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 to CFE device /dev/media1 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[0:05:43.600717338] [2899]  INFO RPI pisp.cpp:653 libpisp version v1.0.2 fa44a258644a 22-11-2023 (21:59:22)
[0:05:43.610991058] [2899]  INFO RPI pisp.cpp:1112 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx219@10 to CFE device /dev/media2 and ISP device /dev/media3 using PiSP variant BCM2712_C0
Made X/EGL preview window
[0:05:44.298961436] [2896]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format Y16 
[0:05:44.298993918] [2896]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format RGB6
[0:05:44.299004455] [2896]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format BGR6
[0:05:44.299017492] [2896]  WARN V4L2 v4l2_pixelformat.cpp:338 Unsupported V4L2 pixel format PC1M
Mode selection for 1640:1232:12:P
    SRGGB10_CSI2P,640x480/0 - Score: 4504.81
    SRGGB10_CSI2P,1640x1232/0 - Score: 1000
    SRGGB10_CSI2P,1920x1080/0 - Score: 1541.48
    SRGGB10_CSI2P,3280x2464/0 - Score: 1718
    SRGGB8,640x480/0 - Score: 5504.81
    SRGGB8,1640x1232/0 - Score: 2000
    SRGGB8,1920x1080/0 - Score: 2541.48
    SRGGB8,3280x2464/0 - Score: 2718
Stream configuration adjusted
[0:05:44.299402158] [2896]  INFO Camera camera.cpp:1183 configuring streams: (0) 1640x1232-YUV420 (1) 1640x1232-BGGR16_PISP_COMP1
[0:05:44.299519899] [2899]  INFO RPI pisp.cpp:1396 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected CFE format: 1640x1232-PC1B
[0:05:44.312581265] [2899] ERROR V4L2 v4l2_videodevice.cpp:1906 /dev/video12[16:cap]: Failed to start streaming: Broken pipe
ERROR: *** failed to start camera ***
kyrie@raspberrypi:~ $ 

`

sandyol55, 2024-02-04:
Seeing a similar issue with picamera2 code no longer working on a 64 bit OS and V2 cameras on Pi5.
Stepping back through the kernel releases, 6.1.65 seems to be the last that didn't show this problem.

```sudo rpi-update d16727d```

will get the 6.1.65 version.

Issue raised in the linux repository which may or not be relevant?

alphanumeric007, 2024-02-04:
That fixed it for me, both cameras now working. Thankyou =)

naushir, 2024-02-05:
I can reproduce this.  Investigating now what may be missing for this particular camera...

naushir, 2024-02-05:
Hopefully https://github.com/raspberrypi/linux/pull/5923 ought to fix this issue correcly.

sandyol55, 2024-02-05:
Will an equivalent fix filter down to 6.1.y or is  6.6 becoming the apt default soon? 
Running with 6.1.65 seems ok for the meantime :-)

naushir, 2024-02-05:
> Will an equivalent fix filter down to 6.1.y or is 6.6 becoming the apt default soon? Running with 6.1.65 seems ok for the meantime :-)

I've just pushed the commit to the 6.1.y tree now.

closed 2024-02-05 by naushir

Timst, 2024-02-06:
Had a similar issue but with a slightly different presentation (see post [here](https://forums.raspberrypi.com/viewtopic.php?t=364984)) that was fixed by the same solution, btw.

DiMTRX, 2024-02-11:
> Seeing a similar issue with picamera2 code no longer working on a 64 bit OS and V2 cameras on Pi5. Stepping back through the kernel releases, 6.1.65 seems to be the last that didn't show this problem.
> 
> `sudo rpi-update d16727d`
> 
> will get the 6.1.65 version.
> 
> Issue raised in the linux repository which may or not be relevant? Edit: link to the issue report added [raspberrypi/linux#5915](https://github.com/raspberrypi/linux/issues/5915)

- Thanks for posting this workaround for the moment, was wrecking my brain trying to fix this.

grantsingleton, 2024-02-15:
> Seeing a similar issue with picamera2 code no longer working on a 64 bit OS and V2 cameras on Pi5. Stepping back through the kernel releases, 6.1.65 seems to be the last that didn't show this [...]

Worked for me as well. Both cams now working on 6.1.65

TetsuyaNOMURA, 2024-02-16:
Now I can see the IMX219 camera working on 6.1.65. And by adding imx290.json modified from imx219.json to libcamera/src/ipa/rpi/pisp/data/ I can see the image from IMX462 as well as Bullseye.
Thank you for the advice.
[image]

shiguangxiaotou3, 2024-03-02:
# 我的Pi 4B,Pi 5和Zero 2 W上都无法开启Camera Module 2
## 我在尝试使用*sudo raspi-config* 的*Interface Options*选项中,没有找到开启摄像头
[image]
## `sudo nano /boot/config.txt` 配置如下：
[image]
## `sudo libcamera-hello -t 0` 也同样报错
[image]
# 我是用的 Raspbreey Pi OS (64-bit) 2023-12-05 操作系统，请问这是为什么

TetsuyaNOMURA, 2024-03-03:
Did you execute below?
>Seeing a similar issue with picamera2 code no longer working on a 64 bit OS and V2 cameras on Pi5. Stepping back through the kernel releases, 6.1.65 seems to be the last that didn't show this problem.

$ sudo rpi-update d16727d

will get the 6.1.65 version.

xXAzazelXx, 2024-03-08:
sorry dumb question, if I understand correctly we are rolling back the kernal with "sudo rpi-update d16727d"?
does this mean this wont be fixed with future update?

naushir, 2024-03-08:
> does this mean this wont be fixed with future update?

The latest kernels already have this fix applied.  If you are comfortable with updating the kernel to the latest release, just run ``sudo rpi-update`` and you will get the fix.

xXAzazelXx, 2024-03-08:
> > does this mean this wont be fixed with future update?
> 
> The latest kernels already have this fix applied. If you are comfortable with updating the kernel to the latest release, just run `sudo rpi-update` and you will get the fix.

thank you, I might just wait until it bleeds into normal updates.

TheoLiu31, 2024-05-14:
> Now I can see the IMX219 camera working on 6.1.65. And by adding imx290.json modified from imx219.json to libcamera/src/ipa/rpi/pisp/data/ I can see the image from IMX462 as well as Bullseye. Thank you for the advice. ![RPi5 BW64 libcamera](https://private-user-images.githubusercontent.com/46389475/305265104-44c04095-c631-40cc-b441-877e90edefc9.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2OTgzNzAsIm5iZiI6MTcxNTY5ODA3MCwicGF0aCI6Ii80NjM4OTQ3NS8zMDUyNjUxMDQtNDRjMDQwOTUtYzYzMS00MGNjLWI0NDEtODc3ZTkwZWRlZmM5LkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTE0VDE0NDc1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjYzJkMTg4NjI5OWFiNGQ4NGY4ODgyYWY4NDc3M2JkZTA0NDY5ZmUxNWVkZDkzZmZjMGFkNDgwZDk1ZWZkOTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qqzcP3w0Dc2HeItmNFX4NuhABuXXUm7FmSnEB69i9CU)

can you share you modified imx290.json from imx219?
Tkx