raspberrypi/libcamera #105: Specific reason for using shared_library over library?

callum-hopkins-dev, 2024-02-03:
Is there any particular reason for using the explicit `shared_library` meson function over the more generic `library` function?

In some cases, it can be useful to generate static libraries for simplified linking on platforms such as alpine.

If there isn't a good reason for it, would you be willing to look at a PR by me which would allow both static and dynamic libraries?

kbingham, 2024-02-03:
I believe it is already successfully packaged by Alpine:

-  https://gitlab.alpinelinux.org/alpine/aports/-/blob/master/community/libcamera/APKBUILD

Static linkage would create complicated licensing scenarios with the L-GPL which is likely not intended. Furthermore, we explicitly build IPA modules as shared objects so that they can be loaded and verified as modules on the needs of the platform and can be interchanged with closed source IPA modules in some cases.

callum-hopkins-dev, 2024-02-03:
Oh, that makes sense.

Thanks for the info :smiley:

closed 2024-02-03 by callum-hopkins-dev