raspberrypi/libcamera #106: [HOW-TO] Run AutoFocus only on a specific range of lenses

ilirosmanaj, 2024-02-06:
My end goal is simple, but to do so, I need some understanding of Picamera2 and libcamera as well. In this case I am using picamera2 as well, but I think the wrapper is not really important and most of the logic would fall into libcamera tunning.

Let me explain my problem in a couple of bullet points: 

- I want to use camera autofocus, which runs only when I trigger it
- I want to use the Fast autofocus speed
- I want to use a Macro autofocus range, limited in lens position from 13 to 18. Defaulting to 13
- When the autofocus is triggered, I want to use PDAF in a way that it acts as quickly as possible
    - Potentially, I would want to allow PDAF to do only one range of scans (as in starting from lens 13 and going up, but then stopping as quickly as it finds some suitable capture)
    - I would be cool with PDAF only testing rounded lens positions (e.g., only values 13, 14, 15, 16, 17, or 18). Currently, this spends plenty of time in fractional values (e.g. 13.55555)

Note: what I am trying to capture is my hand, which is in front of the camera (about 10-15 cm from the camera itself).

Here is my camera tunning file (adding only the autofocus part here):

```
{
            "rpi.af":
            {
                "ranges":
                {
                    "normal":
                    {
                        "min": 1.0,
                        "max": 32.0,
                        "default": 1.0
                    },
                    "macro":
                    {
                        "min": 13.0,
                        "max": 18.0,
                        "default": 13.0
                    }
                },
                "speeds":
                {
                    "normal":
                    {
                        "step_coarse": 2.0,
                        "step_fine": 0.5,
                        "contrast_ratio": 0.75,
                        "pdaf_gain": -0.03,
                        "pdaf_squelch": 0.2,
                        "max_slew": 4.0,
                        "pdaf_frames": 20,
                        "dropout_frames": 6,
                        "step_frames": 4
                    },
                    "fast":
                    {
                        "step_coarse": 1.0,
                        "step_fine": 1.0,
                        "contrast_ratio": 0.75,
                        "pdaf_gain": -0.05,
                         # modified from 0.2, which is the default
                        "pdaf_squelch": 0.4,
                        "max_slew": 5.0,
                         # modified from 16, which is the default
                        "pdaf_frames": 10,
                        "dropout_frames": 6,
                        "step_frames": 2
                    }
                },
                "conf_epsilon": 8,
                "conf_thresh": 12,
                "conf_clip": 512,
                "skip_frames": 5,
                "map": [ 0.0, 420, 35.0, 920 ]
            }
        }
[... 273 lines cut ...]
2024-02-05 10:28:37,796 - root - DEBUG: Processed frame: 61
2024-02-05 10:28:37,800 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.0
2024-02-05 10:28:37,815 - root - DEBUG: Processed frame: 62
2024-02-05 10:28:37,818 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.0
2024-02-05 10:28:37,836 - root - DEBUG: Processed frame: 63
2024-02-05 10:28:37,839 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.0
2024-02-05 10:28:37,859 - root - DEBUG: Processed frame: 64
2024-02-05 10:28:37,863 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,880 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,881 - root - DEBUG: Processed frame: 65
2024-02-05 10:28:37,899 - root - DEBUG: Processed frame: 66
2024-02-05 10:28:37,902 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,918 - root - DEBUG: Processed frame: 67
2024-02-05 10:28:37,921 - root - DEBUG: Autofocus camera state: Scanning, lens position: 13.492711067199707
2024-02-05 10:28:37,926 - root - DEBUG: Processed frame: 68
2024-02-05 10:28:37,945 - root - DEBUG: AF Success on lens position: 13.492711067199707
2024-02-05 10:28:37,963 - root - DEBUG: Processed frame: 69
2024-02-05 10:28:37,966 - root - DEBUG: Finished
```

So this took 2s already, which is way too much for what I want. I would need it at a maximum of 200ms speed.

To wrap up, I need to run the AF on lenses [13, 14, 15, 16, 17, 19] as quickly as possible in a trustable way.

I do expect this to be a combination of picamera and libcamera as well, so happy to perform tunning as much as needed. I failed to find some good explanation of the PDAF algorithm and their parameters so that I would know how to change.

naushir, 2024-02-06:
@ilirosmanaj are you using the Camera v3 Wide or Normal variant here?  You can find more details on the PDAF algorithms and tuning params in our [camera tuning guide](https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf).

@njhollinghurst can provide more guidance here as well.

ilirosmanaj, 2024-02-06:
Hi @naushir - I am using the wide variant in this case. I tried playing with most of the parameters and understanding the PDAF logic from the source code, but unfortunately, I couldn't get very far with it since I think I lack some basic understanding of this.

Changes tried:

* Increasing pdaf_squelch from 0.2 to 0.4
* Lower pdaf_frames from 16 to 10
* Tested with different values of confidence threshold and epsilon

My concern is whether the AF is trying PDAF, then failing, and then jumping to CDAF. Is there a way to enforce PDAF only and limit it to rounded numbers of lens positions?

naushir, 2024-02-06:
Closing this, please track this issue at https://github.com/raspberrypi/picamera2/issues/944.

closed 2024-02-06 by naushir