raspberrypi/libcamera #110: Missleading Build Instructions

Thenrik404, 2024-02-16:
This week I tried to build libcamera in a Docker container running ubunut focal on a raspberry pi 5 with bookworm as host. 
However, while libcamera build successfully it was not able to detect any cameras. I wasted two days on troubleshooting until I found out that the the build instructions were the cause. Instead of 
```
git clone https://git.libcamera.org/libcamera/libcamera.git
```
shouldn't it be 
```
git clone https://github.com/raspberrypi/libcamera.git 
```
???
It would be great if you would update this line not only in the Readme.md but also on https://libcamera.org/getting-started.html
as this is the link users get lead to by https://www.raspberrypi.com/documentation/computers/camera_software.html

naushir, 2024-02-16:
This repo is a fork of the upstream libcamera repo (https://git.libcamera.org/libcamera/libcamera.git).  As such, the top level README.rst file follows instructions for building off the upstream tree.

At present the two trees have diverged because we are still waiting to push Pi 5 support upstream.  So you can only build Pi 5 support with this tree until then.  The intention is that eventually (hopefully in the next few weeks), Pi 5 support will make it upstream, and you can build for Pi 5 using either tree and the instructions you followed.

I am reluctant to change the README.rst file as this is an upstream file, and we don't want to diverge  our trees any further going forward as this increases our maintainence burden.  Hope this explains the situation better.

closed 2024-02-16 by naushir

Thenrik404, 2024-02-16:
Thank you for your detailed answer, that explains a lot! :) I am looking forward for the pi5 support to finally go upstream. Until then, I hope that others find this thread useful.