raspberrypi/libcamera #112: Unable to build libcamera with Python bindings

gwvr, 2024-02-22:
I have been able to build libcamera [v0.2.0](https://github.com/raspberrypi/libcamera/tree/v0.2.0), but without Python bindings.
Perhaps there is an omission in the readme?

I have installed the dependencies as follows (on Raspberry Pi OS Bookworm)

> A C++ toolchain: [required]
> Either {g++, clang}
> Meson Build system: [required]
> meson (>= 0.60) ninja-build pkg-config
> for the libcamera core: [required]
> libyaml-dev python3-yaml python3-ply python3-jinja2
> for gstreamer: [optional]
> libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
> for Python bindings: [optional]
> libpython3-dev pybind11-dev

`meson setup build --reconfigure` output:

```
The Meson build system
Version: 1.0.1
Source dir: /home/pi/Code/libcamera
Build dir: /home/pi/Code/libcamera/build
Build type: native build
Project name: libcamera
Project version: 0.2.0
C compiler for the host machine: cc (gcc 12.2.0 "cc (Debian 12.2.0-14) 12.2.0")
C linker for the host machine: cc ld.bfd 2.40
C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Debian 12.2.0-14) 12.2.0")
C++ linker for the host machine: c++ ld.bfd 2.40
Host machine cpu family: aarch64
Host machine cpu: aarch64
Header "unistd.h" has symbol "issetugid" : NO (cached)
Header "locale.h" has symbol "locale_t" : YES (cached)
Header "stdlib.h" has symbol "secure_getenv" : YES (cached)
Compiler for C supports arguments -Wno-c99-designator: NO (cached)
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Found CMake: /usr/bin/cmake (3.25.1)
Run-time dependency lttng-ust found: NO (tried pkgconfig and cmake)
Program ./parser.py found: YES (/home/pi/Code/libcamera/utils/ipc/./parser.py)
Program ./generate.py found: YES (/home/pi/Code/libcamera/utils/ipc/./generate.py)
Program ./extract-docs.py found: YES (/home/pi/Code/libcamera/utils/ipc/./extract-docs.py)
Program ./gen-tp-header.py found: YES (/home/pi/Code/libcamera/utils/tracepoints/./gen-tp-header.py)
Configuring version.h using configuration
Program openssl found: YES (/usr/bin/openssl)
Library atomic found: YES
Dependency threads found: YES unknown (cached)
Run-time dependency libdw found: YES 0.188
Run-time dependency libunwind found: YES 1.6.2
Header "execinfo.h" has symbol "backtrace" : YES (cached)
Checking for function "dlopen" : YES (cached)
Run-time dependency libudev found: YES 252
Dependency yaml-0.1 found: YES 0.2.5 (cached)
Run-time dependency gnutls found: NO (tried pkgconfig and cmake)
Dependency libcrypto found: YES 3.0.11 (cached)
Dependency libexif skipped: feature android disabled
Dependency libjpeg skipped: feature android disabled
Run-time dependency libevent_pthreads found: NO (tried pkgconfig and cmake)
Run-time dependency libevent_pthreads found: NO (tried pkgconfig and cmake)
Dependency libtiff-4 found: YES 4.5.0 (cached)
Run-time dependency GTest found: NO (tried pkgconfig and system)
Looking for a fallback subproject for the dependency gtest

Executing subproject gtest 

gtest| Project name: gtest
gtest| Project version: 1.11.0
gtest| C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Debian 12.2.0-14) 12.2.0")
gtest| C++ linker for the host machine: c++ ld.bfd 2.40
gtest| Dependency threads found: YES unknown (cached)
gtest| Dependency threads found: YES unknown (cached)
gtest| Dependency threads found: YES unknown (cached)
gtest| Dependency threads found: YES unknown (cached)
gtest| Build targets in project: 46
gtest| Subproject gtest finished.

Dependency gtest from subproject subprojects/googletest-release-1.11.0 found: YES 1.11.0
Run-time dependency qt5 (modules: Core, Gui, Widgets) found: YES 5.15.8 (pkg-config)
Header "QOpenGLWidget" has symbol "QOpenGLWidget" with dependencies Qt5Core, Qt5Core, Qt5Gui, Qt5Widgets: YES (cached)
Detecting Qt5 tools
Run-time dependency qt5 (modules: Core) found: YES 5.15.8 (pkg-config)
Program /usr/lib/aarch64-linux-gnu/qt5/bin/moc found: YES 5.15.8 (/usr/lib/aarch64-linux-gnu/qt5/bin/moc)
Program /usr/lib/aarch64-linux-gnu/qt5/bin/uic found: YES 5.15.8 (/usr/lib/aarch64-linux-gnu/qt5/bin/uic)
Program /usr/lib/aarch64-linux-gnu/qt5/bin/rcc found: YES 5.15.8 (/usr/lib/aarch64-linux-gnu/qt5/bin/rcc)
Program /usr/lib/aarch64-linux-gnu/qt5/bin/lrelease found: NO
Program lrelease5 found: NO
Program lrelease-qt5 found: NO
Program lrelease found: NO found  but need: '== 5.15.8' (/usr/bin/lrelease)
Dependency glib-2.0 found: YES 2.74.6 (cached)
Run-time dependency gstreamer-video-1.0 found: YES 1.22.0
Run-time dependency gstreamer-allocators-1.0 found: YES 1.22.0
Dependency python3 skipped: feature pycamera disabled
Program doxygen found: NO
Program dot found: NO
Program sphinx-build-3 found: NO
Program sphinx-build found: NO
Configuring config.h using configuration
Program python3 (jinja2, ply, jinja2, yaml) found: YES (/usr/bin/python3) modules: jinja2, ply, jinja2, yaml
Build targets in project: 51

libcamera 0.2.0

  Versions
    Sources                  : 0.2.0

  Paths
    LIBCAMERA_DATA_DIR       : "/usr/local/share/libcamera"
    LIBCAMERA_SYSCONF_DIR    : "/usr/local/etc/libcamera"
    IPA_PROXY_DIR            : "/usr/local/libexec/libcamera"
    IPA_CONFIG_DIR           : "/usr/local/etc/libcamera/ipa:/usr/local/share/libcamera/ipa"
    IPA_MODULE_DIR           : "/usr/local/lib/aarch64-linux-gnu/libcamera"

  Configuration
    IPA modules signed with  : libcrypto
    Enabled pipelines        : imx8-isi
                               rkisp1
                               rpi/vc4
                               simple
                               uvcvideo
    Enabled IPA modules      : rkisp1
                               rpi/vc4
    Controls files           : control_ids_draft.yaml
                               control_ids_core.yaml
                               control_ids_rpi.yaml
    Properties files         : property_ids_draft.yaml
                               property_ids_core.yaml
    Hotplug support          : YES
    Tracing support          : NO
    Android support          : NO
    GStreamer support        : YES
    Python bindings          : NO
    V4L2 emulation support   : NO
    cam application          : NO
    qcam application         : YES
    lc-compliance application: NO
    Unit tests               : NO

  Subprojects
    gtest                    : YES

Found ninja-1.11.1 at /usr/bin/ninja
```

kbingham, 2024-02-23:
You also need to "sudo apt install pybind11-dev"

I posted a patch to libcamera (https://patchwork.libcamera.org/patch/18856/) to add documentation about this but it didn't get merged yet. It's not controversial - I'll just merge it.

gwvr, 2024-02-23:
Thanks @kbingham. I did have `pybind11-dev` installed, and it is in the README.rst for v0.2.0 (I haven't checked other branches).
I've also attempted and failed to build libcamera `v0.1.0`, `v0.1.0+rpt20231122`, `v0.2.0+rpt20240215` & `main` (075b54d5).
I don't know what I'm doing, but I'll have a dig and see if I can find clues in the codebase.

gwvr, 2024-02-23:
@kbingham thanks, I just worked that out for myself - I found a clue in one of the files in the repo, but for reasons I can't explain I can't find that file now.
I originally ran `meson setup build`, then installed additional apt packages and ran `meson setup build --reconfigure`.

I can confirm that both
`meson setup build --reconfigure -Dpycamera=auto`
&
`meson setup build --reconfigure -Dpycamera=enabled`
seem to give the desired outcome.

closed 2024-02-23 by gwvr